chan-sccp/chan-sccp #368: Ringtype enhancement for mixed SIP and SCCP environment

gplindstrom, 2017-01-30:
In a mixed environment (SIP and SCCP phones), trying to set ringtype for internal and internal calls is very difficult.  It would be helpful if chan_sccp understood the Bellcore ringtypes and choose/translated to an appropriate SCCP ringtype.

The idea is to set the ALERT_INFO to one of the Bellcore ring types, for example Bellcore-dr2 for inbound routes from the PSTN, and Bellcore-dr1 for extension to extension calls.  In a environment with mixed phone types (SIP and SCCP), chan_sip has no problems understanding what is needed.  But in current chan_sccp driver, it does not know what to do with it.  Determining what type of phone the call is going to, in all the correct places, is very difficult in the dialplan, so I decided the translation was best done in the driver, in this case chan_sccp.

Its been 25+ years since I did any real programming... back in those days it was Cobol, Fortran, Pascal, and assembler.  So I know this is not the "correct" way to do this, but it works for me.  And I know sccp_enum.c gets over written, but I was not sure where to set this up so the code would be generated correctly...  I will leave that to someone with more current experience.  I would like to see some form of this implemented though.  Perhaps it could be set in the config files, or maybe it is a ./configure option, or maybe something else.  Here is what I did that works for me:

$ diff -u sccp_enum.c.orig sccp_enum.c
--- sccp_enum.c.orig	2017-01-29 17:45:50.843998771 -0700
+++ sccp_enum.c	2017-01-29 18:02:21.115042020 -0700
@@ -2322,6 +2322,16 @@
 	[SKINNY_RINGTYPE_BELLCORE_5] = "Bellcore5",
 	[SKINNY_RINGTYPE_SENTINEL] = "LOOKUPERROR"
 };
+static const char *const skinny_ringtype_altmap[] = {
+        [SKINNY_RINGTYPE_OFF] = "Ring Off",
+        [SKINNY_RINGTYPE_INSIDE] = "Bellcore-dr1",
+        [SKINNY_RINGTYPE_OUTSIDE] = "Bellcore-dr2",
+        [SKINNY_RINGTYPE_FEATURE] = "Bellcore-dr3",
+        [SKINNY_RINGTYPE_SILENT] = "Silent",
+        [SKINNY_RINGTYPE_URGENT] = "Bellcore-dr5",
+        [SKINNY_RINGTYPE_SENTINEL] = "LOOKUPERROR"
+};
+
 
 int skinny_ringtype_exists(uint skinny_ringtype_int_value) {
 	if ((SKINNY_RINGTYPE_OFF <=skinny_ringtype_int_value) && (skinny_ringtype_int_value < SKINNY_RINGTYPE_SENTINEL )) {
@@ -2345,6 +2355,11 @@
 			return (skinny_ringtype_t) idx;
 		}
 	}
+        for (idx = 0; idx < ARRAY_LEN(skinny_ringtype_altmap); idx++) {
+                if (sccp_strcaseequals(skinny_ringtype_altmap[idx], lookup_str)) {
+                        return (skinny_ringtype_t) idx;
+                }
+        }
 	pbx_log(LOG_ERROR, "%s %s_str2val('%s') not found\n", LOOKUPERROR_STR, __skinny_ringtype_str, lookup_str);
 	return SKINNY_RINGTYPE_SENTINEL;
 }

For my asterisk implementation (incrediblepbx), I simply set the Alertinfo type to Bellcore-dr2 for inbound routes, and in extensions_custom.conf I added:

[from-internal-custom]  
;Set default to single - SCCP seems to want to do default to outside
exten => _XXXX,1,ExecIf($["${ALERT_INFO}" = ""]?Set(__ALERT_INFO=Bellcore-dr1))

This is working for me.

Also, it might be nice if there was an option to set the chan_sccp default ring.  It looks like it is set to 'outside' currently, but I haven't found correct place in the code to change it.  Perhaps a config option could be made to set it to something else ('inside' for instance).

Thanks for the consideration!

Gary

dkgroot, 2017-01-30:
Hi Gary,

I think this is a good idea / solution. 

Your patch is not completely useable in it's current version, mostly because sccp_enum.c is a generated piece of source code :-).

I created an initial (untested) version you in branch 'issues/368', can you give it a try:
```
git clone https://github.com/chan-sccp/chan-sccp.git chan-sccp_issues368
cd chan-sccp_issues368
git checkout issues/368
./configure [....]
make && make reload
```

We already had the bellcore ringtypes, just the wrong mapping for the way asterisk called these ringtype. I am not sure if this is what you wanted. If we actually need a substitution mapping (the way you did), then i will add that later on.

Also set the new default to SKINNY_RINGTYPE_INSIDE for now, not sure if it is a good idea to change this behaviour mid version. Don't really want to add yet another config paremeter at this moment in time to behonest. I think setting INSIDE as the default does make sense. I guess we have to talk about this a little :-)

gplindstrom, 2017-01-31:
Hi.  Thank you for looking at this.  I saw those bellcore types in the code too, and tried changing them as well, hoping for the easy fix, but it did not work.  Nevertheless, I tried what you sent in case you put something in that I missed :)  So that you know what I did:  I removed the set for __ALERT_INFO in the [from-internal-custom] context, I changed the ALERT_INFO setting on inbound calls from the PSTN to Bellcore-dr1, and installed your custom version.  Based on your changes, I should get single rings on both extension-to-extension calls, and inbound calls from the PSTN.   The results:

SIP phone to SCCP phone - Bellcore-dr2 ring
SCCP phone to SCCP phone - Bellcore-dr2 ring
PSTN to SCCP phone - No ringer sound at all

I then re-added the set for __ALERT_INFO in the [from-internal-custom] context, and the result of both SIP-to-SCCP and SCCP-to-SCCP was there was no ringer sound at all.

So, my guess is the default ringer is still being set to 'Outside', and while the lookup worked to find the alert_info 'Bellcore-dr1' for the PSTN to SCCP phone call:q, the index number being returned and getting passed to the phone API interface (is that the correct terminology) is a value that the phone does not recognize as being valid, so the phone defaults to ring off/silent.  But I am guessing here.  Note that I also changed the line:

channel->ringermode = SKINNY_RINGTYPE_OUTSIDE;
to
channel->ringermode = SKINNY_RINGTYPE_INSIDE;

in sccp_channel.c and nothing changed.

These results are basically the same when I was trying to figure this out.  That is why I created the "substitution mapping" as you called it, because it maintained compatibility with the original function, but I also noticed there was some specific code (in sccp_indicate.c and sccp_pbx.c) that dealt with SKINNY_RINGTYPE_URGENT (and maybe some other code elsewhere) and my patch maintained compatibility with that code.

Hope this helps.  I could give you a rather long debug output if you need, but I didn't see anything I would consider extremely usefull other than I found this buried in the call:

[2017-01-30 20:15:25] NOTICE[30449][C-00000008]: sccp_pbx.c:871 sccp_pbx_channel_allocate:     -- 0005: Allocated asterisk channel SCCP/3699-00000005
 sccp_pbx.c     :0695 (sccp_pbx_channel_allocate          ) <-----                 refcount decreased 05  <- 06 for       line: 3699 (0x7f6288007cd0)
 sccp_pbx.c     :0682 (sccp_pbx_channel_allocate          ) <----------            refcount decreased 10  <- 11 for     device: SEP0012DADBD573 (0x7f6288001df0)
 sccp_pbx.c     :0681 (sccp_pbx_channel_allocate          ) <---                   refcount decreased 03  <- 04 for    channel: SCCP/3699-00000005 (0x7f6240015620)
[2017-01-30 20:15:25] NOTICE[30449][C-00000008]: sccp_callinfo.c:265 callinfo_Setter:     -- 0x7f6240004f90: (sccp_callinfo_setter) changes:2
 ast113.c       :1531 (sccp_wrapper_asterisk113_request   ) <--                    refcount decreased 02  <- 03 for    channel: SCCP/3699-00000005 (0x7f6240015620)
[2017-01-30 20:15:25] NOTICE[30449][C-00000008]: ast113.c:1790 sccp_wrapper_asterisk113_get_rtp_info:  SCCP: (asterisk113_get_rtp_info) Asterisk requested EarlyRTP peer for channel SCCP/3699-00000005
[2017-01-30 20:15:25] NOTICE[30449][C-00000008]: ast113.c:1837 sccp_wrapper_asterisk113_get_vrtp_info:  SCCP: (asterisk113_get_vrtp_info) Asterisk requested EarlyRTP peer for channel SCCP/3699-00000005
 ast113.c       :1631 (sccp_wrapper_asterisk113_call      ) -->                    refcount increased 02  +> 03 for    channel: SCCP/3699-00000005 (0x7f6240015620)
 sccp_pbx.c     :0212 (sccp_pbx_call                      ) ----->                 refcount increased 05  +> 06 for       line: 3699 (0x7f6288007cd0)
[2017-01-30 20:15:25] NOTICE[30449][C-00000008]: sccp_callinfo.c:515 callinfo_Getter:     -- 0x7f6240004f90: (sccp_callinfo_getter) entries:2
[2017-01-30 20:15:25] NOTICE[30449][C-00000008]: sccp_pbx.c:243 sccp_pbx_call:     -- SCCP: (sccp_pbx_call) asterisk callerid='Gary Lindstrom (V) <3685>'
[2017-01-30 20:15:25] NOTICE[30449][C-00000008]: sccp_device.c:1890 sccp_device_getActiveChannel:     -- SEP0012DADBD573: Getting the active channel on device.
[2017-01-30 20:15:25] NOTICE[30449][C-00000008]: sccp_device.c:1898 sccp_device_getActiveChannel:     -- SEP0012DADBD573: No active channel on device.
[2017-01-30 20:15:25] NOTICE[30449][C-00000008]: sccp_pbx.c:360 sccp_pbx_call:     -- SEP0012DADBD573: Ringing Line: 3699 on device:SEP0012DADBD573 using channel:SCCP/3699-00000005, ringermode:Outside
 sccp_indicate.c:0051 (__sccp_indicate                    ) ---------->            refcount increased 10  +> 11 for     device: SEP0012DADBD573 (0x7f6288001df0)
 sccp_indicate.c:0058 (__sccp_indicate                    ) ------>                refcount increased 06  +> 07 for       line: 3699 (0x7f6288007cd0)
[2017-01-30 20:15:25] NOTICE[30449][C-00000008]: sccp_indicate.c:67 __sccp_indicate:     -- SEP0012DADBD573: Indicate SCCP new state:RINGING, current channel state:DOWN on call:SCCP/3699-00000005, lineInstance:1 (previous channelstate:DOWN)
[2017-01-30 20:15:25] NOTICE[30449][C-00000008]: sccp_session.c:984 sccp_session_send2: SEP0012DADBD573: Send Message: Clear Notify Message(0x0115) 4 bytes length
[2017-01-30 20:15:25] NOTICE[30449][C-00000008]: sccp_device.c:1702 sccp_dev_cleardisplaynotify:     -- SEP0012DADBD573: Clear the display notify message
[... cut ...]
[2017-01-30 20:15:25] NOTICE[30449][C-00000008]: sccp_softkeys.c:1097 sccp_softkey_setSoftkeyState:     -- SEP0012DADBD573: softkey 'Confrn' on RINGIN to on

Notice the two lines:

[2017-01-30 20:15:25] NOTICE[30449][C-00000008]: sccp_pbx.c:360 sccp_pbx_call:     -- SEP0012DADBD573: Ringing Line: 3699 on device:SEP0012DADBD573 using channel:SCCP/3699-00000005, ringermode:Outside

and

[2017-01-30 20:15:25] NOTICE[30449][C-00000008]: sccp_device.c:1383 sccp_dev_set_ringer:     -- SEP0012DADBD573: Send ringer mode Outside(3) on device

Anyway, if you need anything else, let me know.

Thanks!

Gary

gplindstrom, 2017-01-31:
FYI, think I found line that is setting default to 'Outside'. It is the line:
skinny_ringtype_t ringermode = SKINNY_RINGTYPE_OUTSIDE;
in the file /chan-sccp_issues368/src/pbx_impl/ast113/ast113.c
Not sure what the interaction is/should be with the line you changed.

dkgroot, 2017-01-31:
Hi Gary,
Thanks you for the detailed reply. We tried a quick and dirty solution (which did not work). I think your substitution map is ok to implement, which i will do. I will revert my previous patch and create a working solution, this time.

dkgroot, 2017-01-31:
Pushed a new commit to the 'develop' branch, which handles alertinfo2ringtype conversion nicely and correctly this time.

Also added a way to change the default ringtype
```
[general]
...
ringtype=inside                               ; default value is "outside"
...
```

closed 2017-01-31 by dkgroot

dkgroot, 2017-01-31:
Added a new wiki page, see:
[Ringer-RingType-Alert_Info](https://github.com/chan-sccp/chan-sccp/wiki/Ringer-RingType-Alert_Info)

gplindstrom, 2017-02-01:
Diederik,

Outstanding!!!  I can confirm it works ( bellcore interpretation and the 
ringtype setting in [general] ) .  Thank you very much! This SCCP driver 
has come a long way over the years and is very good.

I should have put the details of my test environment when I submitted 
the enhancement request, in case you are keeping track of what works:

Asterisk 13.13.1 (IncrediblePBX)
Phones: Cisco 7936, 7941, 7961, and7970 (All use SCCP 9-4-2SR1-1S except 
7936 which is 3.3(21.00.0003) )
                 SCCPPhone app on android (SCCP)
                 Linphone app on android (SIP)
                 Aastra 6751i, 6755i (SIP)
                 SNOM 360 (SIP)

Thanks again!

Gary

dkgroot, 2017-02-01:
Hi Gary,

Thanks for reporting back, always appreciated. If/when you have more good ideas that are relatively simple to add, keep them coming :-)

Hope you checked the wiki page as well. If there are any wiki pages that need improvement, feel welcome to help out !

Regards,

Diederik

gplindstrom, 2017-02-06:
Hi Diederik,

Quick question.  I was trying to add the Cisco SIP patch (see link) so 
that I could integrate some of the newer Cisco SIP-Only phones.  I got 
the SIP patch to work showing presence status, but it is a real bear to 
get working compared to chan_sccp and my version of SCCP_Manager.

http://usecallmanager.nz/patching-asterisk.html

Anyway, from the very beginning after applying the patch and 
re-compiling (both asterisk 13.13.1 and chan_sccp), SCCP functionality 
broke.  Calls from any type phone (SIP or SCCP) to a SCCP phone result 
in the call not going through, and when doing a "sccp show channels", 
there seem to be calls "stuck".  SIP-to-SIP (Cisco, Aastra, SNOM, and 
Android App) all work ok.

Anyway, the question is, have you or anyone else ever tried mixing 
chan_sccp with this patch?  Or am I doing something wrong, and I just 
need to work through it?

Thanks!

Gary

dkgroot, 2017-02-06:
Hi Gary,

Note: Please don't mix issues, especially not after the issue is already closed. 
Note: This question is not directly related to a problem in chan-sccp, so it's more something for our mailing-list then a direct issue report.

I have seen these chan_sip patches but not really looked at them or tried them out. The guy who created these chan_sip/asterisk patches has chosen a very hard route. It will be hard to keep up, but never the less he has been for a while now, and his patches look pretty clean. It might have been easier if he had created this as a separate add-on, instead of directly changing asterisk and chan_sip sources. I don't think digum/asterisk will (ever) accept the patches in their current form.

It's difficult to tell what might be wrong without running these patches on my local asterisk and debugging the issue for you. And now the annoying part of this reply: If this patch breaks compatibility, it should be fixed in the patch, not in chan-sccp (right?). 

Note:His patch is not only changing chan_sip, but also stuff in asterisk/res/ and asterisk/main/, we depend on these files being clean and immaculate. Out configure scripts has to go through great length to figure out exactly which version of asterisk we are dealing with and might just be thrown off by these sort of changes.

Sorry to leave you hanging in the middle this way, but you either have to debug this issue yourself or ask http://usecallmanager.nz to have a look. I am pretty sure they are aware of chan-sccp, as they are implementing similar features on top of the sip protocol.

gplindstrom, 2017-02-06:
Diederik,

Thank you for the reply.  That basically gives me the info I need.  I 
think most installs are small enough and most people don't mix 
technologies, so it has not really been tried and/or worked on much.  If 
I think I can narrow it down a bit and it looks like something should be 
changed for chan_sccp (which I am doubting),  I will open a new issue 
with you.  Chan_sccp works so well for me, I figure it probably is 
really something with the SIP patch.  The fact that it breaks a 
stand-alone module (chan_sccp), implies it should be fixed in his 
patch.  I just wanted to get your opinion, and see if you or anyone had 
tried making these patches and chan_sccp work together.

Thanks again!

Gary

dkgroot, 2017-02-06:
Hi Gary,

The dialplan changes required for this project, might be causing some issues (don't know). Have you tried increasing the debug logging on the sip and sccp sides, to see if you can find/see anything particular. I don't know if you have taken any measures to integrate both project, or are just expecting everything to work out of the box. I don't know exactly what you mean by calls not going through, not how far they get, etc...

For example, if you are using the [call-extension] dialplan context from the usecallmanager website, then you will never be dialing any SCCP extensions ever (sorry for sounding a little pedantic). But i don't know what you have done and what you haven't.

gplindstrom, 2017-02-06:
Hi Diederik,

Do you want me to open a different case for this?  I can do that....  
But I have sent an email to the USECALLMANAGER patch folks to see what 
they say.

Basically, I take my "working" freepbx/asterisk install, with chan_sccp, 
install their patch, recompile asterisk and chan_sccp, install, and 
restart.  No config changes anywhere at this point... that's important 
because it should be taking any dialplan stuff out of the picture.  I 
didn't use the [call-extension] stuff the way they did, but this is a 
non-issue since the problem occurs with only the patch applied and no 
config changes (my opinion anyway).  I can place sip-to-sip calls 
(including to/from a sip cisco phone), but all sccp-to-sip, 
sccp-to-sccp, and sip-to-sccp which use to work, no longer work, 
sometimes causing asterisk to crash.

I did turn full debugging (sip and sccp) on, verbosity to 10, and 
scanned  quickly, but it was 2:00am for me, and had been a long day, so 
I was not 100%.  I believe I saw an error about writing data to the sccp 
channel, but I will need to look again.  I am by no means an expert, but 
the channel may have already been closed by that point.  I will take a 
look at it later again and see if I can figure more out.  I should point 
out again that the Cisco "SIP" phone works fine in the un-patched system 
(with sccp), except for the presence/hints stuff which is the reason I 
installed the patch.

I will look at it some more and let you know what I find, but lets see 
if we can get the usecallmanager folks involved.

Your awesome.  I didn't expect a reply yet :)

Thanks and have a great day!

Gary

dkgroot, 2017-02-06:
Adding freeobx into the mix is making everything even more difficult. Freepbx has one of the most complex and confusing dialplan ever. I am pretty sure the usecallmanager guys are going to tell you that as well. I don't see a chance of getting the nice features from the usecallmanager project to I regrate nicely into freepbx without some major work on your part. I am pretty sure if you take freepbx out of the mix will make you live a lot easier. You will have to create your own dialplan.though, and the usecallmanager project is already expecting you to.

You don't have to open a new issue here, based on the mix you are trying to make work, I am pulling out.

dkgroot, 2017-09-12:
@Gary
Hope you were able to figure things out in the meantime ! Did you ?

commit ce8bce82 (2017-01-31): Enh: Added sccp_parse_alertinfo function which converts ALERT_INFO Bellcore types to skinny_ringtypes
Enh: Added [general]ringtype to be able to define global default ringtype
Enh: Updated cli and appfunctions output to include global ringtype

Fixes: #368