chan-sccp/chan-sccp #369: Channel called with no recorded File Descriptor

miguelsanzpardo, 2017-01-31:
**Description:**

[2017-01-31 07:02:52] ERROR[31652][C-000002d5] channel.c: ast_read() on chan 'SIP/elastix-cc-00000603' called with no recorded file descriptor.
[2017-01-31 08:33:55] ERROR[2877][C-00000320] channel.c: ast_read() on chan 'SIP/elastix-cc-0000067a' called with no recorded file descriptor.
[2017-01-31 08:35:36] ERROR[2993][C-0000032a] channel.c: ast_read() on chan 'SIP/18605-00000686' called with no recorded file descriptor.
[2017-01-31 08:58:51] ERROR[4167][C-00000313] channel.c: ast_read() on chan 'SCCP/12005-00000F45' called with no recorded file descriptor.

**chan-sccp version:**  Release: 4.3.0 develop - f08ac44M 

**asterisk version:** 11.21.2

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: No
- Problem can be reliably reproduced, doesn't happen randomly: Im not sure
- Platform:
  - [64 Bit]
  - [Debian 8.4]

dkgroot, 2017-01-31:
Notice: Low Priority

This debug output is only generated during a transfer (I also noticed it while working on 'features/transfer; branch). The good thing is, that it is preventing further issues down the line. 

It would be nice if we can find a way, to prevent the issue from happening in the first place. ast_quiet_channel(xxx) might be of use in this case. Or switching of RTP traffic on select channels before attempting the transfer. Will have to experiment a little.

Related: #308, #299
Related Branch: features/transfer

miguelsanzpardo, 2017-02-06:
Sharing: ast_read() on chan 'SIP...' called with no recorded file descriptor.txt
[ast_read() on chan 'SIPprimario-fijos1-00000032' called with no recorded file descriptor.txt](https://github.com/chan-sccp/chan-sccp/files/755150/ast_read.on.chan.SIPprimario-fijos1-00000032.called.with.no.recorded.file.descriptor.txt)

dkgroot, 2019-01-09:
This one should be resolved in current develop branch.

closed 2019-01-09 by dkgroot