chan-sccp/chan-sccp #370: OpenReceiveChannelAck No channel with this PassThruPartyId

miguelsanzpardo, 2017-01-31:
**Description**
[2017-01-31 09:01:54] ERROR[3266][C-00000355] sccp_actions.c: SEP7CAD7443156C: Device sent a Keypress, but there is no (active) channel! Exiting
[2017-01-31 09:01:54] ERROR[3266][C-00000355] sccp_actions.c: SEP7CAD7443156C: Device sent a Keypress, but there is no (active) channel! Exiting

Device:
devicetype=7925

**chan-sccp version:** Release: 4.3.0 develop - f08ac44M

**asterisk version:** 11.21.2

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Not sure]
- Problem can be reliably reproduced, doesn't happen randomly: [Not sure]
- Platform:
  - [32/64 Bit]
  - [Operating System]

dkgroot, 2017-01-31:
Pushed a new branch 'issues/370' to correct the OpenReceiveChannelAck debug output.

It might just be that the user already ended the call, but we were still processing the RTP start request. 

If that is the cause (and we should see that when we increase the debug level), then we only have to reduce the debug output from ERROR to standard.

Please set "sccp debug core, channel, pbx" during testing of the patch, so that we can check if the call was already 'ended' when the error message was produced.

miguelsanzpardo, 2017-02-06:
Sharing: (OpenReceiveChannelAck) No channel with this PassThruPartyId_Error
[Uploading (OpenReceiveChannelAck) No channel with this PassThruPartyId_Error.txt…]()

commit 3f5edb32 (2017-02-21): Fix: reduce log level for '(OpenReceiveChannelAck) No channel with this PassThruPartyId' when channel has already been hungup.

Fixes #370
closed 2017-02-21 by dkgroot