chan-sccp/chan-sccp #371: Device sent a Keypress, but there is no (active) channel

miguelsanzpardo, 2017-01-31:
**[Description]**
[2017-01-31 09:01:54] ERROR[3266][C-00000355] sccp_actions.c: SEP7CAD7443156C: Device sent a Keypress, but there is no (active) channel! Exiting
[2017-01-31 09:01:54] ERROR[3266][C-00000355] sccp_actions.c: SEP7CAD7443156C: Device sent a Keypress, but there is no (active) channel! Exiting

devicetype=7925

**chan-sccp version:** 4.3.0 develop - f08ac44M

**asterisk version:** 11.21.2

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Yes/No]
- Problem can be reliably reproduced, doesn't happen randomly: [Yes/No]
- Platform:
  - [64 Bit]
  - [Debian 8.4]

dkgroot, 2017-01-31:
Note: It might be cause by wireless phone leaving the AP zone.
Note: This is a older type device, which does not send complete keypad button information :-(

Would need to see some 'sccp debug core, channel, action, pbx, message', to see what went on before this issue arose. Activating 'message' debug will generate a log out debug output, so please be prepared for this.

Maybe you can simulate this issue in a Lab environment by entering keypad buttons, while leaving the AP zone.

miguelsanzpardo, 2017-02-01:
The problem is that I dont have 7925 phones on my laboratory, I only have 7942/7962 phones. I can try to obtain one of these phones on the hospital where I have all the phones.

dkgroot, 2017-02-01:
Aah ok, understood. Maybe we should ask someone on the mailinglist to perform this test, that might be easier.
Or you can just set the debug level at your customer site for 1 day, and check the results when the issue occurs.

dkgroot, 2017-02-06:
Hi Miguel,
I created a separate branch 'issues/374' to investigate the keypress issue. When you have access to a 7925 in your Lab, you might be able to help out by checking out this branch and sending back some logging as well.

miguelsanzpardo, 2017-02-06:
Sharing: Device sent a Keypress, but there is no (active) channel_Error.txt
[Device sent a Keypress, but there is no (active) channel_Error.txt](https://github.com/chan-sccp/chan-sccp/files/755122/Device.sent.a.Keypress.but.there.is.no.active.channel_Error.txt)

dkgroot, 2017-02-15:
Hi Miguel:

Please do set core set debug 0 to reduce the amount of DialPlan Logging, the FreePBX Dialplan is extensive enough as it is.

For this KeyPress Message, the most important part of the debug log is missing, which is the output from: 'packet hex dump:', which is what i was looking for.

```
[2017-02-03 09:25:15] VERBOSE[661][C-00000012] :  SCCP: packet hex dump:
... MISSING...
[2017-02-03 09:25:15] VERBOSE[661][C-00000012] :     -- SEPA45630406BDC: SCCP (handle_keypad) digit:00000005, callid:958, lineInstance:1
[2017-02-03 09:25:15] VERBOSE[661][C-00000012] :     -- SEPA45630406BDC: Could not find channel for lineInstance:1 and callid:958 on device
[2017-02-03 09:25:15] ERROR[661][C-00000012] sccp_actions.c: SEPA45630406BDC: Device sent a Keypress, but there is no (active) channel! Exiting
```
With the MISSING information, i would be able to determine if the dissection of this message is working correctly.

Can you find this debug log and lookup these couple of lines ?

dkgroot, 2017-02-15:
BTW: Just before this KeyPress was sent, the call was already terminated/hungup. 
See:
`[2017-02-03 09:25:15] VERBOSE[661][C-00000012] :   == SEPA45630406BDC: Ending call SCCP/17350-000003BE (state:DIGITSFOLL)`

In this case it would be good to stop processing the KeyPress request.

dkgroot, 2017-02-15:
Changed the priority level, as this might have been a false positive / not really an issue.

miguelsanzpardo, 2017-02-16:
I share with you the info. related with the packet hex dump. You can see the same ERROR on 2 different devices on the same day:

- SEPA45630406BDC (09:25:15)
- SEPACA016FC338A (14:48:19)

[packet hex dump.txt](https://github.com/chan-sccp/chan-sccp/files/779915/packet.hex.dump.txt)

dkgroot, 2017-02-16:
Looked at the debug files and it in both cases the call was hungup just before the keypress was received. Logically both parties are trying to hangup, the remote party succeeded and the person with the 7925 pressed the wrong button by accident (trying to hangup but pressing a keypad button instead).

The error handling in this case is correct. And the debug output is as well. Maybe producing an asterisk ERROR is not what you want. I have pushed another commit to check if the device is already being hungup, in which case it will produce a lower level error debug.

Closing this issue. Thanks for reporting it though.

closed 2017-02-16 by dkgroot