chan-sccp/chan-sccp #372: MySQL to XML

darkzbaron, 2017-01-31:
[Short description of problem here]

**Reproduction Steps:**

1. Installation as per https://github.com/chan-sccp/chan-sccp/wiki/Getting-phone-book-from-mysql-database
Companysearch.php
$xmldir = "http://192.168.0.200/directory";   # don't add a trailing slash!
$dialprefix = ""; #prefix for dialing out (if there is any)
$searchumlaut = "yes"; #cisco phone cannot display umlauts, if this option is enabled then you will be able to search for stored umlauts in mysql db
$mysqltable = "phonebook";
$argnumber = "number";
$argname = "name";
$argcontact = "contact";
$argmobile = "mobile";
$arghomenumber = "homenumber";
$argintern = "intern"; # use value of 1 in record if internal extension

Search Directory.xml
<CiscoIPPhoneInput>
	<Prompt>Enter first letters to search</Prompt>
	<URL>http://192.168.0.200/directory/companyDirectory.php</URL>

Directory.xml

<CiscoIPPhoneMenu>
	<Title>Company Services</Title>
	<Prompt>Please select one</Prompt>
	<MenuItem>
		<Name>Search Company Directory</Name>
		<URL>http://192.168.0.200/directory/searchDirectory.xml</URL>
	</MenuItem>
	<MenuItem>
		<Name>View Company Directory</Name>
		<URL>http://192.168.0.200/directory/companyDirectory.php</URL>
	</MenuItem>
	<MenuItem>
		<Name>View Directory Backwards</Name>
		<URL>http://192.168.0.200/directory/companyDirectory.php?desc=true</URL>
	</MenuItem>
</CiscoIPPhoneMenu>

Database:
Please see screenshot.
 

**Observed behavior:**

The phone displays XML Parse Error

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Yes/No]
- Problem can be reliably reproduced, doesn't happen randomly: [Yes/No]
- Platform:
Running XAMPP for Windows 5.6.28

On 

  -OS Name	Microsoft Windows 10 Pro
Version	10.0.14393 Build 14393

- Direct Contact information:  darkzbaron@hotmail.com

I would appreciate some lights on that? Thanks so much for the great piece of work, can't wait to get it to work.

Thanks,Jonathan

[image]

dkgroot, 2017-01-31:
@darkzbaron : 
Search Directory.xml is missing the "</CiscoIPPhoneInput>" entry at the end. I am guessing this was a copy paste issue.

Could it be that you have more then 32 entries coming out of the question, when you perform a search ? There might be an issue in the paging method. Cisco devices only allow a maximum of 32 entries per page.

You can request the same result page using your webbrowser, most webbrowser will show invalid XML and point out the failure.

Note this wiki page was a user contribution by :@miguelsanzpardo
@miguelsanzpardo: Can you help out here?

darkzbaron, 2017-01-31:
just uploaded the whole project on a repo https://github.com/darkzbaron/CiscoDirectory

darkzbaron, 2017-01-31:
will upload the web debug shortly

darkzbaron, 2017-01-31:
thanks so much!

darkzbaron, 2017-01-31:
http://ec29c2ca.ngrok.io/CiscoDirectory/Directory.xml

darkzbaron, 2017-01-31:
just opened a tunnel

dkgroot, 2017-01-31:
@darkzbaron : Can you join our chat on [gitter](https://gitter.im/chan-sccp/chan-sccp) please

dkgroot, 2017-01-31:
Closing this issue report, because it is not directly related on chan-sccp.

closed 2017-01-31 by dkgroot

dkgroot, 2017-02-01:
When looking at the repository i noticed that you completely changed the original wiki script. So it was completely unrelated. 

Note: This issue list is only for and about the chan-sccp module.