chan-sccp/chan-sccp #373: DND label not being removed after turning DND off

mscdex, 2017-02-01:
With ce8bce8, the DND label added to a speed dial entry is not removed when that extension turns DND off. The light goes off, but the label is not removed. If it matters, this is on a 7914 sidecar attached to a 7961 phone running firmware SCCP41.9-4-2SR1-1S. I have not tested with speed dial entries on-device.

In the console I correctly see it get enabled:

```
SEPxxxxxxxxxxxx: change dnd to reject
```

and then when they turn DND off:

```
SEPxxxxxxxxxxxx: change dnd to silent
SEPxxxxxxxxxxxx: change dnd to off
```

**Reproduction Steps:**

1. Turn on DND (set to reject)
2. Turn off DND

**Expected behavior:**

DND label should be removed from speed dial label.

**Observed behavior:**

DND label is not removed from speed dial label.

**Screenshots**

*<none>*

**chan-sccp version:** ce8bce8

**asterisk version:** 13.9.1

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: Yes? (Previous version didn't add DND label at all IIRC)
- Problem can be reliably reproduced, doesn't happen randomly: Yes
- Platform:
  - CPU: Cortex A5
  - 32/64 Bit: 32
  - Big/Small Endian: little endian
  - Operating System: Ubuntu
  - Operating System Version: 16.04.1
- Direct Contact information: (github issue)
- Log Files (attach .txt / [pastbin](http://pastebin.com/) / [gist](http://gist.github.com/)):
- Comments:

dkgroot, 2017-02-03:
What do you mean by a Label in this case ? Are you referring to the StatusBar Message (just above the Softkeys) on the phone where you are pressing the DND softkey, or are you referring to DND indication on remote BLF speeddials ?

mscdex, 2017-02-03:
I mean on the line/button/speeddial labels.

dkgroot, 2017-02-03:
OK. Can you send a bit of debug log, with 'sccp debug core, hint, indicate, softkey' set. That should show what is going on.

dkgroot, 2017-02-06:
So the BLF Speeddials is on the addon (7914) attached to a 7961 (java phone). DND is activated and disabled on remote phone being monitored by this BLF (correct?).

I think this might be a 'remnant' left on the display. We are sending newer phone messages to these types of (java) phone (protocol level > 15), but the addon is pretty old and might not be able to handle these kinds of messages. Not sure how to easily rectify this in our current source though. We never really made provisions for dealing with older addons on newer phones. It would/might require quite a bit of work, to add exceptions for all combinations.

mscdex, 2017-02-06:
> So the BLF Speeddials is on the addon (7914) attached to a 7961 (java phone). DND is activated and disabled on remote phone being monitored by this BLF (correct?).

Yes, this is basically what is happening. Unfortunately I do not have debug output yet to give (if you still need it) ...

dkgroot, 2017-02-06:
Might be helpful to Marcello, if he is able to pick this issue up. He does have a couple of addons, which would be helpful to debug this issue. He does not have a lot of spare time at the moment, so it might take a while for him to be able to look at the issue.

I can think of a quick fix, which would be removing the Label/Text component to the remote DND indicate, which might be benifitial, if you are interested.
Remarking out line 1152 in sccp_hint.c should do the trick.

drtmk, 2017-03-21:
This is how it behaves.

Label "**Reception**":
[image]

Label "**abcdefghijklmnopr**":
[image]

Label "**iiiiiiiiiiiiiiiiii**":
[image]

Label "**aaaaaaaaaaaaaaaaaaa**":
[image]

In CUCM 11, the BLF status on 7914 extension module is not updated at all, when monitored extension goes DND. Regardless of whether the DND is set to reject the call or mute the ringer.

dkgroot, 2017-03-21:
@drtmk / @mscdex 
I just received two 7915 as a donation from Warren, so I can finally have a look at how to fix this. From the example above it looks like the issue is similar to 7960 speeddial behaviour. These old speeddial buttons cannot update their label either. The problem with addons is that we don't keep track of the type of the add-on attached to a device (at this moment), so we just send the message in device protocol level, without taking into account that the particular speeddial will be displayed on an add-on instead.

It will require quite a bit of work to change this behaviour. 

Maybe we should just issue a small patch to remedy the side effect (disable/remove the label update for DND, and just rely on the icon instead). This should be a simple patch to create.

drtmk, 2017-03-22:
I think the icon and red button is enough.

mscdex, 2017-03-22:
I cannot recall offhand, but does the icon actually differentiate between DND and offhook/busy?

drtmk, 2017-03-22:
Yes. From what I remember the busy/off hook icon shows a lifted handset.

dkgroot, 2017-03-22:
Yes the icon is different (Red Cross Icon = DND and Offhook Icon for non-idle/busy).

AccessDG, 2017-06-15:
I'm seeing this too, however, I just compiled the most current source tonight so I'll see how it works in the morning.

AccessDG, 2017-06-16:
Ok.. installed latest build and it seems to have taken care of this problem for me.

dkgroot, 2017-09-07:
If everybody is fine having only the DND icon and not the text update (which would work on any of the phones and expansions, then i could take out the (DND) text fragment, to solve the issue in a simple way. Differentiating the update based on the expansion used (7914 or 7915) in the current source tree is not really an option. If you guys can live with this, i will make the change today.

mscdex, 2017-09-07:
I think the icon and LED are *probably* enough for right now.

dkgroot, 2017-09-07:
In the meantime i might have found a way to include the addon type information, making it possible to be more flexible. Trying to make this work at the moment. If not i will fall back to removing the "(DND)" string as an 'ugly' solution.
If i do get it to work, it will be posted in a seperate branch called 'issues/373' so you guys can check it out and test/check it.

dkgroot, 2017-09-07:
The commit above is only step one. Little more work needed.

dkgroot, 2017-09-08:
@drtmk / @mscdex / @AccessDG 
Suppressing all label updates on 7914 expansion units (So remote caller info is not displayed on these, but icon/lamp updates do still get processed)
Behaviour on 7915/7916/SPA500/SPA932 should be unchanged

Please test the 'issues/373' branch and report back
If you have 7914 and 7915 please check that the results are correct on both.

When the results are are ok, we can close this issue report.

dkgroot, 2019-01-09:
@marcelloceschia @davidded 
Can you guys reproduce this issue ? Do you have a suggestion about how to resolve the issue.

dkgroot, 2019-08-23:
Merged into develop

commit 288e513b (2017-09-08): Fix: Do not update the featureLabel when the hint is on a 7914 addon

Fixes #373
commit 5cb59cf9 (2019-08-23): Fix: Getting the btnlist to include the devicetype, so we can use this information in make_button_template
Upd: switch 7915 back to using MULTI button
Fix: Do not update the featureLabel when the hint is on a 7914 addon

Fixes #373
closed 2019-08-23 by dkgroot