chan-sccp/chan-sccp #374: Network Busy and No Mic

Giotino, 2017-02-02:
I've 2 Cisco 7920 and some other SIP phones
When i try to call any number from the Cisco 7920 screen says "Network Busy"
When i call the Cisco 7920 from a SIP phone it ring, but when i answer i can't ear the Cisco 7920 on the SIP phone, also when i press a button it give me this error:

[Feb  2 21:31:28] ERROR[20283]: sccp_device.c:3028 sccp_device_buttonIndex2lineInstance: SEP0015FAF37311: buttonIndex2lineInstance for buttonIndex:192 failed!
[Feb  2 21:31:28] NOTICE[20283]: sccp_actions.c:2879 handle_keypad_button: SEP0015FAF37311: Device sent a Keypress, but there is no (active) channel! Exiting

Full SCCP log of the call: http://pastebin.com/98p02s2V
Call channels info: http://pastebin.com/9AQFHurL
My Configuration: http://pastebin.com/x5JX2E34

chan-sccp version: 4.3.0

asterisk version: Asterisk 13.10.0

Addition / Optional Information:
- Platform:
  - 64 Bit
  - Ubuntu 16.10

dkgroot, 2017-02-03:
Hi Giotino,
Which chan-sccp branch and revision are you using ? Did you download a tgz or get a clone from git ?

Can you try:
```
git clone https://github.com/chan-sccp/chan-sccp.git chan-sccp
cd chan-sccp
git checkout develop
./configure [.... configure flags if any...]
make && make reload
```
Just to make sure you are using the latest version.

Giotino, 2017-02-03:
I'm on the latest of master branch from github:

Skinny Client Control Protocol (SCCP). Release: 4.3.0 develop - ce8bce8M (built by 'root' on '2017-02-03 14:23:53 UTC')

dkgroot, 2017-02-03:
We will have to do a little digging to figure out what's going on.

* Which firmware version are you running on your 7920 ?
* Can you attach the output for 'sccp show device SEP001BD51354FB'
* Can you set 'sccp debug core, action, channel, pbx, message' and perform another dial.

I will look at the results next monday.

Giotino, 2017-02-03:
Firmware: 7920.4.0-03-02
The output for 'sccp show device SEP001BD51354FB': http://pastebin.com/G4Cvxfbe
The log from the call: http://pastebin.com/V3VQjPeP

Now it logs the DTMF Digit but the sound from the mic still missing

dkgroot, 2017-02-03:
I had a look at the debug log. I notice two things
You incoming call is chwnging it's RTP source ( maybe caused by a sip reinvite) which we do not expect. It sends a ChangeSrc after opening the RTP stream. Can you check you sip.conf file to deny reinvite just to see if that makes a difference.
I also noticed that the 7920 sends a very short keypad message (without a lineinstance number). When I get back on Monday I will see needs to change, to handle this case.

Giotino, 2017-02-03:
With 'canreinvite = no' as global rule in sip.conf nothing changes

dkgroot, 2017-02-03:
OK. Not sure yet what is causing the repeated source changes, but we will figure it out. Not going to be available during the weekend though.

dkgroot, 2017-02-04:
I looked at the source code and your debug report and I think I know how to fix the keypad issue. These older type phones send some of their messages in a different format that needed ones. I thought I fixed this a while ago, but apparently not completely.

The sourcechange causing one way audio  is not clear yet, and I am not 100% what causes this, will require a bit more debugging I guess. Fixing these issues one at a time

Giotino, 2017-02-04:
With the lastest commit the keypad works, i wrote it two comment above.

dkgroot, 2017-02-06:
Hi Giovani,
Create a seperate branch to fix this issue, called 'issues/374'

Can you do: 
```
git checkout issues/374
make && make reload
```
in your current chan-sccp directory, 

Please set 'sccp debug core channel action pbx softkey codec rtp' and send me a report for these types of calls:
- Calling from 7920 to another SCCP/SIP phone
- Receiving a call on the 7920
- Sending some DTMF keys while in the call (even if it is with one way audio)

Giotino, 2017-02-06:
Skinny Client Control Protocol (SCCP). Release: 4.3.0 issues/374 - 644a4f6M (built by 'root' on '2017-02-06 18:23:17 UTC')

There's no log for "calling from 7920 to another SCCP/SIP phone" and it says "Network Busy"
The log for the call from 7920: http://pastebin.com/xtF9uy5s

Unfortunately i made all the test before i realized i was on the develop branch so i post them below(they might help)
Skinny Client Control Protocol (SCCP). Release: 4.3.0 develop - ce8bce8M (built by 'root' on '2017-02-03 14:23:53 UTC')

There's no log for "calling from 7920 to another SCCP/SIP phone"
The log for the call from 7920: http://pastebin.com/ubAuRkLY

dkgroot, 2017-02-07:
I don't completely understand. When you start a new outgoing call on the 7920, what says "Network Busy", the phone / asterisk CLI / something else ? If it is the phone, i think you might have a major configuration issue in the phone itself.

You will need to create a new configuration using:
http://www.cisco.com/en/US/docs/voice_ip_comm/cuipph/7920/4_0/english/administration/guide/7920cfgu.html

Also checkout out:
https://sourceforge.net/p/chan-sccp-b/mailman/search/?q=7920&mail_list=chan-sccp-b-discussion
About 7920 issues discussed on our mailinglist in the past.

dkgroot, 2017-02-21:
Any updates / progress ? Did you send a mail to the mailinglist, that it was caused by setting incominglimit=1, or was that someone else.
I have merged issues/374 into the develop brach, so it does no longer exist, you can move back to the develop branch, to get the same changes.

Giotino, 2017-02-24:
I still have the same problem, but now i think it's my bad because i can't use the configuration utility and i think that my xml config is wrong

Giotino, 2017-02-24:
I can't figure out why the audio is one-way

dkgroot, 2017-02-24:
Please do a cold restart of the phone. Sometimes when something goes wrong, the devices get stuck with an open RTP channel, and i think these 7920's only have one of those.

You need a windows machine to use the 7920 configuration utility :-(. You should be able to run this in a VM if required (and it might even run under wine (don't know)).

Giotino, 2017-02-24:
I can't run configuration utility because cisco's download is not public, so i can't download it.

But i found an interesting thing on the net: the one-way audio (and also the network busy one) issue of the 7920 exists because of the Wi-Fi power of the router.
If you search "Cisco 7920 one-way audio" on google you'll find a lot of articles.
I'm setting up a new wifi network for testing.

dkgroot, 2017-02-24:
Not sure how wifi could cause this in a 'reliable' manner, but ok.

I would replace the phone instead. The 7920 is pretty annoying to setup and use (wifi coverage is limited as well). You can get 7925/7926's or even other wifi VoIP phones for good prices. But the voice is up to you 😄

closed 2020-03-01 by dkgroot