chan-sccp/chan-sccp #375: Compilation crashes with option --with-hash-size

jozzus, 2017-02-08:
Packege can't be compiled with option --with-hash-size

**Reproduction Steps:**

1. Download from Development brance (git clone https://github.com/chan-sccp/chan-sccp chan-sccp_develop)
2. Configure with options (./configure --enable-conference --enable-distributed-devicestate --with-astmoddir=/usr/lib64/asterisk/modules/ --enable-advanced-functions --with-hash-size)
3. make

**Expected behavior:**

The packege shuld be compiled

**Observed behavior:**

Compilation exitedwith Error

**Screenshots**

In file included from sccp_refcount.c:40:0:
config.h:689:25: error: ‘yes’ undeclared here (not in a function)
 #define SCCP_HASH_PRIME yes
                         ^
sccp_refcount.c:111:12: note: in expansion of macro ‘SCCP_HASH_PRIME’
 } *objects[SCCP_HASH_PRIME];           //!< objects hash table
            ^
In file included from common.h:46:0,
                 from sccp_refcount.c:41:
sccp_refcount.c: In function ‘sccp_refcount_destroy’:
sccp_dllists.h:97:20: error: invalid type argument of ‘->’ (have ‘int’)
  typeof(__list_head->first) __list_next;        \
                    ^
sccp_dllists.h:107:41: note: in expansion of macro ‘SCCP_LIST_TRAVERSE_SAFE_BEGIN’
 #define SCCP_RWLIST_TRAVERSE_SAFE_BEGIN SCCP_LIST_TRAVERSE_SAFE_BEGIN
                                         ^
sccp_refcount.c:146:4: note: in expansion of macro ‘SCCP_RWLIST_TRAVERSE_SAFE_BEGIN’
    SCCP_RWLIST_TRAVERSE_SAFE_BEGIN(&(objects[hash]->refCountedObjects), obj, list) {
    ^
sccp_dllists.h:98:20: error: invalid type argument of ‘->’ (have ‘int’)
  typeof(__list_head->first) __list_prev = NULL;       \
                    ^
sccp_dllists.h:107:41: note: in expansion of macro ‘SCCP_LIST_TRAVERSE_SAFE_BEGIN’
 #define SCCP_RWLIST_TRAVERSE_SAFE_BEGIN SCCP_LIST_TRAVERSE_SAFE_BEGIN
                                         ^
sccp_refcount.c:146:4: note: in expansion of macro ‘SCCP_RWLIST_TRAVERSE_SAFE_BEGIN’
    SCCP_RWLIST_TRAVERSE_SAFE_BEGIN(&(objects[hash]->refCountedObjects), obj, list) {
    ^
sccp_dllists.h:99:20: error: invalid type argument of ‘->’ (have ‘int’)
  typeof(__list_head->first) __new_prev = NULL;       \
                    ^
sccp_dllists.h:107:41: note: in expansion of macro ‘SCCP_LIST_TRAVERSE_SAFE_BEGIN’
 #define SCCP_RWLIST_TRAVERSE_SAFE_BEGIN SCCP_LIST_TRAVERSE_SAFE_BEGIN
                                         ^
sccp_refcount.c:146:4: note: in expansion of macro ‘SCCP_RWLIST_TRAVERSE_SAFE_BEGIN’
    SCCP_RWLIST_TRAVERSE_SAFE_BEGIN(&(objects[hash]->refCountedObjects), obj, list) {
    ^
sccp_dllists.h:100:26: error: invalid type argument of ‘->’ (have ‘int’)
  for ((var) = __list_head->first, __new_prev = (var),      \
                          ^
sccp_dllists.h:107:41: note: in expansion of macro ‘SCCP_LIST_TRAVERSE_SAFE_BEGIN’
 #define SCCP_RWLIST_TRAVERSE_SAFE_BEGIN SCCP_LIST_TRAVERSE_SAFE_BEGIN
                                         ^
sccp_refcount.c:146:4: note: in expansion of macro ‘SCCP_RWLIST_TRAVERSE_SAFE_BEGIN’
    SCCP_RWLIST_TRAVERSE_SAFE_BEGIN(&(objects[hash]->refCountedObjects), obj, list) {
    ^
sccp_dllists.h:100:46: warning: assignment makes integer from pointer without a cast [enabled by default]
  for ((var) = __list_head->first, __new_prev = (var),      \
                                              ^
sccp_dllists.h:107:41: note: in expansion of macro ‘SCCP_LIST_TRAVERSE_SAFE_BEGIN’
 #define SCCP_RWLIST_TRAVERSE_SAFE_BEGIN SCCP_LIST_TRAVERSE_SAFE_BEGIN
                                         ^
[... 324 lines cut ...]
make[3]: Leaving directory `/usr/src/chan-sccp_develop/src'
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory `/usr/src/chan-sccp_develop/src'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/usr/src/chan-sccp_develop/src'
make: *** [all-recursive] Error 1

**chan-sccp version:** 4.3.0_develop_r1812b49

**asterisk version:** 13.13.1

**Addition / Optional Information:**

- Platform:
  Operating System : Linux
          Kernel : 3.10.0-514.6.1.el7.x86_64
    Architecture : x86_64
Asterisk Version : 11300 / 113
  Asterisk Repos : TRUNK
      Build user : root
         Version : 4.3.0_develop_r1812b49
Compiler Version : 4.8.5
              CC : gcc
             CPP : gcc -E
         HOST_CC : gcc

dkgroot, 2017-02-08:
@jozzus 

You need to specify a size for the hash table, so --with-hash-size=xxxx. You only need to change this value if you have a high(er) number (100+) of phones connecting to the server. And if you do want ot change the size, the value needs to be a prime number.

BTW: You should not have to set the --with-astmoddir in most cases. Normally it is automatically detected and/or retrieved from /etc/asterisk/asterisk.conf. Note: some older asterisk binary packages were delivered with a broken asterisk.conf file (They used '=' instead of '=>' for the paths), the configure script will detect and report this. You need to fix this file manually before recompiling. Once the file is repaired you can re-run configure. The --with-astmoddir parameter was mostly added for package maintainers.

Can you replace
./configure --enable-conference --enable-distributed-devicestate --with-astmoddir=/usr/lib64/asterisk/modules/ --enable-advanced-functions --with-hash-size

With
./configure --enable-conference --enable-distributed-devicestate --enable-advanced-functions --with-hash-size=3001

dkgroot, 2017-02-08:
Committed Revision: aed782f
Which will check the value supplied. Will generate error and stop configure if false.

closed 2017-02-08 by dkgroot