chan-sccp/chan-sccp #376: Channel OffHook when not

pnlarsson, 2017-02-08:
So this something that has happend a couple of time now and then:
`SCCP/wid7663-0000011 from-company_204     wid7663             1 OffHook (None)       (Empty)                   244             02:34:00`
Today I had sccp set debug core,message active and got this in the log (this is a 7960):
```
[2017-02-08 13:18:00] VERBOSE[18515][C-00000005] sccp_actions.c:     -- SEP000000000000: >> Got message Accessory Status Message (0x49)
[2017-02-08 13:18:00] NOTICE[18515][C-00000005] sccp_actions.c: SEP000000000000: SCCP Handle Message: Accessory Status Message(0x0049) 12 bytes length
[2017-02-08 13:18:00] VERBOSE[18515][C-00000005] sccp_utils.c:  SCCP: packet hex dump:
 00000000 - 0C 00 00 00 00 00 00 00  49 00 00 00 02 00 00 00  - ........I.......
 00000010 - 01 00 00 00                                       - ....
[2017-02-08 13:18:00] VERBOSE[18515][C-00000005] sccp_actions.c:     -- SEP000000000000: >> Got message Off-Hook Message (0x6)
[2017-02-08 13:18:00] NOTICE[18515][C-00000005] sccp_actions.c: SEP000000000000: SCCP Handle Message: Off-Hook Message(0x0006) 12 bytes length
[2017-02-08 13:18:00] VERBOSE[18515][C-00000005] sccp_utils.c:  SCCP: packet hex dump:
 00000000 - 0C 00 00 00 00 00 00 00  06 00 00 00 00 00 00 00  - ................
 00000010 - 00 00 00 00                                       - ....
[2017-02-08 13:18:00] VERBOSE[18515][C-00000005] sccp_actions.c:     -- SEP000000000000: Taken Offhook
[2017-02-08 13:18:00] VERBOSE[18515][C-00000005] sccp_actions.c:     -- SEP000000000000: Using line wid11275
[2017-02-08 13:18:00] VERBOSE[18515][C-00000005] sccp_channel.c: SEP000000000000: (getEmptyChannel) on line:wid11275, maybe_c:
[2017-02-08 13:18:00] VERBOSE[18515][C-00000005] sccp_channel.c: SEP000000000000: (getEmptyChannel) getActiveChannel
[2017-02-08 13:18:00] NOTICE[18515][C-00000005] ast113.c: SCCP: (allocPBXChannel) Create New Channel with name: SCCP/wid11275-0000011B
```
And then the device restarted (no more messages between 13:18:00 -> 13:23:24):
```
[2017-02-08 13:23:24] VERBOSE[16718] sccp_session.c:     -- SCCP: Accepted Client Connection from C.C.C.C:51464
[2017-02-08 13:23:24] VERBOSE[16718] sccp_session.c:     -- SCCP: Connected on server via S.S.S.S:35772
[2017-02-08 13:23:24] VERBOSE[23525] sccp_actions.c:     -- S.S.S.S:35772: >> Got message Alarm Message (0x20)
[2017-02-08 13:23:24] NOTICE[23525] sccp_actions.c: S.S.S.S:35772: SCCP Handle Message: Alarm Message(0x0020) 96 bytes length
[2017-02-08 13:23:24] VERBOSE[23525] sccp_utils.c:  SCCP: packet hex dump:
 00000000 - 60 00 00 00 00 00 00 00  20 00 00 00 02 00 00 00  - `....... .......
 00000010 - 31 37 3A 20 4E 61 6D 65  3D 53 45 50 30 30 31 42  - 17: Name=SEP0000
 00000020 - 44 34 37 44 41 33 46 42  20 4C 6F 61 64 3D 38 2E  - 00000000 Load=8.
 00000030 - 31 28 53 52 2E 32 29 20  53 65 6E 61 73 74 65 3D  - 1(SR.2) Senaste=
 00000040 - 4B 65 65 70 61 6C 69 76  65 54 4F 00 00 00 00 00  - KeepaliveTO.....
 00000050 - 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  - ................
 00000060 - 00 08 01 00 0A 01 01 2C                           - .......,
[2017-02-08 13:23:24] VERBOSE[23525] sccp_actions.c:     -- SCCP: Alarm Message: Severity: Informational (2), 17: Name=SEP000000000000 Load=8.1(SR.2) Senaste=KeepaliveTO [67584/738263306]
[2017-02-08 13:23:24] VERBOSE[23525] sccp_actions.c:     -- S.S.S.S:35772: >> Got message Register Message (0x1)
[2017-02-08 13:23:24] NOTICE[23525] sccp_actions.c: S.S.S.S:35772: SCCP Handle Message: Register Message(0x0001) 68 bytes length
[2017-02-08 13:23:24] VERBOSE[23525] sccp_utils.c:  SCCP: packet hex dump:
 00000000 - 44 00 00 00 00 00 00 00  01 00 00 00 53 45 50 30  - D...........SEP0
 00000010 - 30 31 42 44 34 37 44 41  33 46 42 00 00 00 00 00  - 0000000000.....
 00000020 - 01 00 00 00 0A 01 01 2C  07 00 00 00 00 00 00 00  - .......,........
 00000030 - 00 00 00 00 0B 00 60 85  00 00 00 00 00 00 00 00  - ......`.........
 00000040 - 00 00 00 00 00 00 00 00  00 00 00 00              - ............
[2017-02-08 13:23:24] VERBOSE[23525] sccp_actions.c:  SEP000000000000: is registering, Instance: 1, UserId: 0, Type: Cisco 7960 (7), Version: 11 (loadinfo '')
[2017-02-08 13:23:24] NOTICE[23525] sccp_actions.c: SEP000000000000: Cleaning previous session, come back later, state:OK
[2017-02-08 13:23:24] NOTICE[23525] sccp_session.c: SCCP: Send Message: Register Reject Message(0x009D) 36 bytes length
[2017-02-08 13:23:24] VERBOSE[23525] sccp_utils.c:  SCCP: packet hex dump:
 00000000 - 24 00 00 00 00 00 00 00  9D 00 00 00 43 6C 65 61  - $...........Clea
 00000010 - 6E 69 6E 67 20 70 72 65  76 69 6F 75 73 20 73 65  - ning previous se
 00000020 - 73 73 69 6F 6E 2C 20 63  6F 6D 65 00              - ssion, come.
[2017-02-08 13:23:24] VERBOSE[23525] sccp_device.c:  SCCP: Clean Device SEP000000000000
[2017-02-08 13:23:24] VERBOSE[23525] sccp_device.c:     -- Removed device 'SEP000000000000' from Glob(devices)
[2017-02-08 13:23:24] VERBOSE[23525] sccp_device.c:   == SCCP: Hangup open channel on line wid11275 device SEP000000000000
[2017-02-08 13:23:24] VERBOSE[23525] sccp_channel.c:   == SEP000000000000: Ending call SCCP/wid11275-0000011B (state:DOWN)
[2017-02-08 13:23:24] VERBOSE[23525] sccp_device.c:   == SCCP: Remove Line wid11275 from device SEP000000000000
[2017-02-08 13:23:24] VERBOSE[23525] sccp_device.c:   == SCCP: Unregister Device SEP000000000000
[2017-02-08 13:23:24] NOTICE[23525] sccp_session.c: SEP000000000000: Send Message: Reset(0x009F) 8 bytes length
```
But the channel is not hungup for real.

The device can make calls from here on.

Got more logs if needed as well as a backtrace from before restarting the asterisk.

Asterisk 13.13.0
chan_sccp - latest from revert_session_handler branch (and I thought I was running latest develep...)
Phone: 7960 with 8.1(2) firmware

dkgroot, 2017-02-10:
We need to figure out what happened exactly:
 - Why did the device go offhook (user initiated ? / hook craddle issue / hookswitch issue):
    - the rubbers in the hook cradle, expand over time on these phone, which will slowly push the receiver outward, making the phone go offhook. A simple solution is removing / or cutting down these small rubbers.
    - It's also possible that the hookswitch get's dirty and needs a bit of cleaning.
 - When the phone was offhook, why were no further messages send or recieved. And who initiated the restart (device or server). Both parties have a timeout counter. If it was the server, that means we do get stuck somewhere in the code. I have posted a number of fixes in the last weeks, to remedy re-register issues. You should try the 'develop' branch instead.
 - During 'device_cleaning' any open channels should be hungup. Can you tell me a little more about your comment: 'But the channel is not hungup for real.'. Is there still a chan-sccp channel remaining, or is there an asterisk channel still open ?

Because of the difficulty reproducing these issues, it's hard to work on a solution for any of these. We need lot's of debug logging, to figure out what is going on, in any of these events (sccp debug core, device, action, session, refcount).

Thank you for reporting the issue. Please do switch to the latest 'develop' revision, before reporting any more.

pnlarsson, 2017-02-16:
* Don't know the reason for offhook, not a bad hookswitch atleast (we have had a bunch of thoose) - I presume user tried to call
* My guess is that it was a network related issue - the phones are in malaysia
* The channel is not hangup: core show channels list the channels as active, not sure about the sccp show channels though

I have been trying to replicate the issue with a 7960 and are failing - I always get more messages from the phone after "Got message Off-Hook Message". And this issue does not happen often in production, mayby ones a month.

I just hoped there was a simple fix now when the messages where in the debug.

dkgroot, 2017-02-16:
It might be message keepalive/timeout related. I created a new issues/keepalive branch, which might help with this issue. My previous timeout calculation was wrong, and needed revising. If you have a test system, then please give it a try.

The asterisk channel remaining after the call has terminated is wrong and should be corrected. How to replicate the issue, to making fixing it a little easier is not clear though.

I have tried several methods to timeout calls in the past, but never really succeeded in getting a reliable way to do it. I tried pulling cables mid call. Even slowing down tcp traffic, etc... but has not really worked far.

I did make a couple of changes to the sccp_dev_clean function, but there might still be more work there, to check open channels etc.

dkgroot, 2019-02-22:
No reply since issue opening.
If the issue still persists, please reopen.
Closing the issue.

closed 2019-02-22 by dkgroot