chan-sccp/chan-sccp #377: Wrong URL sent when define <directoryURL> in SEP....cnf.xml

Tenere750, 2017-02-09:
Good evening Everyone,

I don't know if I'm right here with my topic but as you are the cracks in chan_sccp and strange behaviour on Cisco IP phones I might be right here...

I followed the instructions on your [page ](https://github.com/chan-sccp/chan-sccp/wiki/Getting-phone-book-from-mysql-database) to setup a phone book search (installed and configured database, ...).
When entering the URL 
> http://192.168.178.207/cisco/menu.xml

I noticed an "HTTP error" on the phone and nothing else happened.

A tcpdump (tcpdump -A -s 0 'tcp port 80 and (((ip[2:2] - ((ip[0]&0xf)<<2)) - ((tcp[12]&0xf0)>>2)) != 0)' -vv) to monitor port 80 on my webserver showd that the URL was altered to  
http://192.168.178.207/cisco/menu.xml?locale=germany&name=SEP002155D64F77

Here's the link to the [dump](http://pastebin.com/pgrQ41jF).

I highly appreciate any help in this case.

Best regards,

Joerg

P.S.: Hi Marcello, it has been a while since we dropped some lines...

dkgroot, 2017-02-10:
Hi Joerg,

It is normal behaviour for the phone to add the name and locale to the initial web request, and there is not way to suppress that. Your webserver however, should be perfectly happy to parse those extra strings (i would assume). What happens when you request "http://192.168.178.207/cisco/menu.xml?locale=germany&name=SEP002155D64F77" using your webbrowser ?

Do note that your reply to this request, should contain a "Content-type: text/xml" header followed by the CiscoIPPhone XML.

Tenere750, 2017-02-10:
> menu.xml?locale=germany&name=SEP002155D64F77" using your webbrowser ?

Hi Diederik,

thanks for your answer,

The reply is

Jörg Bucholzki
Pfarrer-Linden-Strasse 32
53909 Zülpich

Tenere750, 2017-02-10:
Sorry, I was too trigger happy...

This is the reply

<?xml version="1.0"?>
-<CiscoIPPhoneMenu>
<Title>Asterisk Directory</Title>
-<MenuItem>
<Name>Personensuche</Name>
<URL>http://192.168.178.207/cisco/menu2.xml</URL>
</MenuItem>
</CiscoIPPhoneMenu>

Jörg Bucholzki
Pfarrer-Linden-Strasse 32
53909 Zülpich

2017-02-10 14:13 GMT+01:00 Joerg Bucholzki <bucholzki@googlemail.com>:

dkgroot, 2017-02-10:
Hi Jörg,

Have you checked that the server is setting 'Content-type: text/xml' header when sending this reply ? 

The phone generating "HTTP error" either indicates:
 - that it cannot request the URL provided (not the case, because the webserver logs the request)
 - that it does not receive anything back from the webserver (Check tcpdump)
 - that the content it got back does not have the correct header. BTW: it does not like a 'application/xml' header when receiving xml data !.
 - that the content it received does not follow the correct CiscoIPPhone structure. BTW: You are missing "<Prompt>Select One</Prompt>" for example (don't think this is required though).

Note: The debug output generated by the phone is annoyingly sparse, and the XML parser in the phone is very finicky.

Tenere750, 2017-02-11:
>    CiscoIPPhone structure. BTW: You are missing "Select One" for example

Thanks a lot for your answer.

And yes, I love to tinker with SCCP and a lot of new phones. My spouse
would like to have a proper PBX so she'll get one ;)

The first one pointed to the solution. As I installed apache2 out of the
box there was no text/xml content header defined.
I added this to the modules pat of spsche2 and it worked immediately.
Regarding MWI: added the context an voilá, success!

Thanks again!

But you're not done with helping me yet and I just run into the next
problem.
I have a SPA525G2 running in SPCP mode. After a restart of Asterisk (and
chan_sccp) the phone registers one time. After a reboot of the SPA525G2 it
can't re-register (on phone display: SPCP-Registrierungstoken abgewiesen)

This is what I see in Asterisk
[Feb 11 16:46:06] NOTICE[18095]: sccp_actions.c:699 handle_SPCPTokenReq:
SCCP: Cleaning previous session, come back later, state:Cleaning
[Feb 11 16:46:10] NOTICE[18056]: sccp_session.c:638
sccp_netsock_device_thread: SCCP: Closing session because connection timed
out after 63 seconds (ip-address: 192.168.178.207:2000).
[Feb 11 16:47:06] NOTICE[18095]: sccp_actions.c:699 handle_SPCPTokenReq:
SCCP: Cleaning previous session, come back later, state:Cleaning
[Feb 11 16:48:06] NOTICE[18095]: sccp_actions.c:699 handle_SPCPTokenReq:
SCCP: Cleaning previous session, come back later, state:Cleaning

Any ideas?

Best regards,

Joerg

Jörg Bucholzki
Pfarrer-Linden-Strasse 32
53909 Zülpich

dkgroot, 2017-02-11:
Congrats on getting the first two going
Mixing separate issues/questions is not desirable, so please don't do that in the future 😂

I will check with my spa525 to see why this is happening. Please create a new issue for the problem. I am closing this issue report.

For future reference: questions shoukd go to the mailinglist, problems, segfault etc should be reported in an issue on github. One problem/issue per report.

closed 2017-02-11 by dkgroot