chan-sccp/chan-sccp #378: SPA525G2 won't register (SPCP token reject)

Tenere750, 2017-02-11:
Good Evening Everyone,

after an Asterisk restart my SPS525G2 is able to register normally with Asterisk.
When rebooting the device without restarting Asterisk the following message is shown on the display "SPCP-Registrierungstoken abgewiesen"

In Asterisk CLI this is shown
```
Feb 11 16:46:06] NOTICE[18095]: sccp_actions.c:699 handle_SPCPTokenReq: SCCP: Cleaning previous session, come back later, state:Cleaning
[Feb 11 16:46:10] NOTICE[18056]: sccp_session.c:638 sccp_netsock_device_thread: SCCP: Closing session because connection timed out after 63 seconds (ip-address: 192.168.178.207:2000).
[Feb 11 16:47:06] NOTICE[18095]: sccp_actions.c:699 handle_SPCPTokenReq: SCCP: Cleaning previous session, come back later, state:Cleaning
[Feb 11 16:48:06] NOTICE[18095]: sccp_actions.c:699 handle_SPCPTokenReq: SCCP: Cleaning previous session, come back later, state:Cleaning
```

**Reproduction Steps:**
1. Reboot SPA525G2

**Expected behavior:**
Registering with Asterisk.

**Observed behavior:**
SPCP token rejected; no registering with Asterisk

**chan-sccp version:** (include revision and/or branch if relevant)
Skinny Client Control Protocol (SCCP). Release: 4.3.0 develop - ce8bce8M

**asterisk version:** (include revision and/or branch if relevant)
Asterisk 14.2.1 built by root @ AsteriskPi on a armv7l running Linux on 2017-02-04 20:23:31 UTC

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Yes/No]
Can't state

- Problem can be reliably reproduced, doesn't happen randomly: [Yes/No]
Yes

- Platform:
  - Raspberry Pi 2
  - Distributor ID: Raspbian 8.0 (jessie)
  - Linux AsteriskPi 4.4.38-v7+ #938 SMP Thu Dec 15 15:22:21 GMT 2016 armv7l GNU/Linux

- Direct Contact information: bucholzki@googlemail.com

[SPA525G2.zip](https://github.com/chan-sccp/chan-sccp/files/768821/SPA525G2.zip)

dkgroot, 2017-02-14:
Found the cause and submitted a patch
Thanks for reporting the issue.
Using a SPA525 finally made this reproducable :-)

Tenere750, 2017-02-14:
You're welcome!

Jörg Bucholzki
Pfarrer-Linden-Strasse 32
53909 Zülpich

Tenere750, 2017-02-15:
Unfortunately it happened again... 
And additionally there's some weird behaviour with the buttons under the display.
[image]

dkgroot, 2017-02-16:
Hi Jörg,
That doesn't look good. Was the device able to register afterwards ? The screen has multiple artifacts. I think the phone crashed, which does not have to do with the reregistration issue I would think.
And this happened with the latest revision ?

dkgroot, 2017-02-16:
Nur Für deine information, Mann kann Umlaute benutzen für Button Namen, wenn du möchtest 😊

Tenere750, 2017-02-16:
Guten Morgen,

wenn ich Umlaute benutze, sieht das so aus...

[image: Inline-Bild 1]

Statt G0 sollte da "Gästezimmer" stehen...

But nevertheless: this first picture was immediately after a manual reboot
of Asterisk and the phone because of the SPCP token rejected message.
It is registered but in addition to the funny looking display the behaviour
is strange as well. When I dial an internal extension e.g 11 an "Dial"
button appears in the display when pushed nothing happens and again the
input mask appears with a "Wählen" button and after pushing the button the
phone doesn't user the internal context as it was supposed to do (and all
other phones do immediatelay...).

Tonight I'll try to user the phone in SIP mode or do a factory reset or...?

This strange behaviour happend after the SPCP error messages...

Cheers,

Joerg

dkgroot, 2017-02-16:
I am not 100% sure if the SPA phones can do accented characters as well, i did add support for them on the Cisco phones, so i assumed it should have worked :-)

A factory reset of the phone might be in order, to get it back into a working state again. I don't think it is related to the re-registration issue though. I did not make any changes that would be able to generate something like what you saw.

Tenere750, 2017-02-16:
Thanks for your answer.

As this happened immediately after the SPCP token error I assumed it was
related to.

I'll keep you informed what's up with the phone.

Thanks again! I highly appreciate the work you guys do and the support you
give.

dkgroot, 2017-02-16:
Can't be 100% sure what happened of course, but something is off :-)
Thanks and keep us posted !

Tenere750, 2017-02-16:
Good evening,

in my opinion there's two problems which appear.

1. the articats
2. the strange behaviour when dialing

The reason for the artefacts is the
 <userLocale>
  <name>germany</name>
  <langCode>de</langCode>
  <winCharSet>iso-8859-1</winCharSet>
</userLocale>
in the SEP<MAC>.cnf.xml

If  <winCharSet> ist set to utf-8 instead of iso-8859-1 the artifacts will appear but to display Umlaute is not possible any more.

The strange dial behaviour is still as follows:
1. dial any number (internal or external) via the keypad and press the "Wählen" button (first button just below the display) the phone short beeps one time for every pressed number and the input mask appears again and the menu is now in English (dial). If you do exactly nothing and wait apprx. 8secs the call will be executed.

2. If you press the "Anruf" butten (second button below the display), enter the number and press "dial" (there's no "wählen" there, just the English name), the call will be executed immediately.

I hope this adds more to the confusion :)

Best regards,

Joerg

dkgroot, 2017-02-16:
I cannot think of a solution for the ISO/Utf issue. I guess you have to stick to the iso setting 😥

Not 100% what is going during dialing. It almost sounds like it is using sip like dialmatching behaviour, which is strange. Or maybe it's doing overlap dialing ? I don't have the german locale files for this phone on my system, but I can give them a try if need be. Can you see any debug logging when it is performing wählen (sccp debug code, action, device, channel, pbx).

Tenere750, 2017-02-18:
Good afternoon,

regarding the ISO/UTF problem: than I have to deal with it. Not nice but
could be worse.

Attached you'll find the debug outpu. Hopefully you'll find something
helpful.

If you need more testing just drop me a line.

Best regards,

Joerg

    -- SEP503DE50F0300: Send date/time
    -- SEP503DE50F0300: Getting the active channel on device.
    -- SEP503DE50F0300: No active channel on device.
    -- SEP503DE50F0300: Taken Offhook
    -- SEP503DE50F0300: Device State is 'Off Hook'
    -- SCCP: Looking for channel by state '21' on device: SEP503DE50F0300
    -- SEP503DE50F0300: line: '30'
    -- SEP503DE50F0300: Could not find active channel with state RINGING(21) on device
    -- SEP503DE50F0300: Looking for line with instance 1.
    -- SCCP: Found line 30
    -- SEP503DE50F0300: Using line 30
SEP503DE50F0300: (getEmptyChannel) on line:30, maybe_c:
SEP503DE50F0300: (getEmptyChannel) getActiveChannel
    -- SEP503DE50F0300: Getting the active channel on device.
    -- SEP503DE50F0300: No active channel on device.
    -- SEP503DE50F0300: Set the active channel 149 on device
    -- SEP503DE50F0300: Set the active line 30
    -- 0017: New channel number: 149 on line 30
    -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 30,
    -- SCCP:              cid_num: "30"
    -- SCCP:             cid_name: "G�stezimmer"
    -- SCCP:          accountcode: "11"
    -- SCCP:                exten: ""
    -- SCCP:              context: "Telekom_01_out"
    -- SCCP:             amaflags: "0"
    -- SCCP:            chan/call: "SCCP/30-00000095"
    -- SCCP:combined capabilities: "[g722/64k, alaw/64k, ulaw/64k, g723, g729, g729a, g729b, g729ab, g729/annex/b, ilbc, g726/32k, g726/24k, g726/16k]"
    -- SCCP:  reduced preferences: "[g722/64k, alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab, g729/annex/b]"
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/30-00000095
    -- SEP503DE50F0300: recalculateReadformat
    -- preference: 6(g722/64k)
    -- found first firstJointCapability 6(g722/64k)
    -- Empty remote Capabilities, using bestCodec from firstJointCapability 6(g722/64k)
    -- SCCP/30-00000095: (recalculateReadformat)
        capabilities: [g722/64k, alaw/64k, ulaw/64k, g723, g729, g729a, g729b, g729ab, g729/annex/b, ilbc, g726/32k, g726/24k, g726/16k]
        preferences: [g722/64k, alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab, g729/annex/b]
        remote caps: []
        READ:[g722/64k]
    -- SEP503DE50F0300: recalculateWriteformat
    -- preference: 6(g722/64k)
    -- found first firstJointCapability 6(g722/64k)
    -- Empty remote Capabilities, using bestCodec from firstJointCapability 6(g722/64k)
    -- SCCP/30-00000095: (recalculateWriteformat)
        capabilities: [g722/64k, alaw/64k, ulaw/64k, g723, g729, g729a, g729b, g729ab, g729/annex/b, ilbc, g726/32k, g726/24k, g726/16k]
        preferences: [g722/64k, alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab, g729/annex/b]
        remote caps: []
        Write:[g722/64k]
    -- 0017: Allocated asterisk channel SCCP/30-00000095
    -- SEP503DE50F0300: Set asterisk state OffHook (2) for call 149
    -- SEP503DE50F0300: Indicate SCCP new state:OFFHOOK, current channel state:DOWN on call:SCCP/30-00000095, lineInstance:1 (previous channelstate:DOWN)
    -- SEP503DE50F0300: Send speaker mode 'on'
    -- SEP503DE50F0300: Send and Set the call state offhook(1) on call 149 (visibility:default)
    -- SEP503DE50F0300: Send activate call plane on line 1
    -- SEP503DE50F0300: ( sccp_indicate.c:84:__sccp_indicate ) sccp_dev_displayprompt '� ' for line 1 (8)
    -- SEP503DE50F0300: Display prompt on line 1, callid 149, timeout 8
    -- SEP503DE50F0300: Sending tone Inside Dial Tone (33) on line 1 with callid 149 (direction: User)
[... 132 lines cut ...]
 - calledParty: Arbeitszimmer J�rg <11>, valid
 - callingParty: G�stezimmer <30>, valid
 - originalCallingParty: G�stezimmer <30>, valid
 - lastRedirectingParty: G�stezimmer <30>, valid, reason: 0
 - presentation: CallerId Presentation Allowed

    -- SEP503DE50F0300: send callInfo on SCCP/30-00000095 with lineInstance: 1
  == SEP503DE50F0300: (pbx_indicate) finish: send indication 'outofband signaling' (0)
    -- SCCP/11-00000096 is ringing
    -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_RINGING: Remote end is ringing'
    -- SEP503DE50F0300: (pbx_indicate) start indicate 'AST_CONTROL_RINGING: Remote end is ringing' (3) condition on channel SCCP/30-00000095 (readStat:0, receiveChannelState:0, rtp:no, default res:inband signaling (-1))
    -- SEP503DE50F0300: Indicate SCCP new state:RINGOUT, current channel state:DIALING on call:SCCP/30-00000095, lineInstance:1 (previous channelstate:DIGITSFOLL)
    -- SEP503DE50F0300: Send and Set the call state ring-out(3) on call 149 (visibility:default)
    -- SEP503DE50F0300: ( sccp_indicate.c:165:__sccp_indicate ) sccp_dev_displayprompt '�' for line 1 (8)
    -- SEP503DE50F0300: Display prompt on line 1, callid 149, timeout 8
    -- SEP503DE50F0300: Stop tone on line 1 with callid 149
    -- SEP503DE50F0300: Sending tone Alerting Tone (36) on line 1 with callid 149 (direction: User)
    -- SEP503DE50F0300: Set softkeyset to RINGOUT(8) on line 1  and call 149
    -- SEP503DE50F0300: validKeyMask 4294967291
    -- SEP503DE50F0300: Finish to indicate channel state:RINGOUT on call:SCCP/30-00000095, lineInstance:1. New channel state:RINGOUT
    -- SEP503DE50F0300: Set asterisk state Ring (4) for call 149
       > remote preferences: [g722/64k, g722/56k, g722/48k, alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab]
       > remote caps: [g722/64k, g722/56k, g722/48k, alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab]
  == SEP503DE50F0300: (pbx_indicate) finish: send indication 'inband signaling' (-1)
    -- SEP503DE50F0300: Asterisk prodded channel SCCP/30-00000095.

dkgroot, 2017-02-18:
The spa525 should be capable of doing enbloc dialing (can be switched on/off on the settings page). In this case it is switched off, and the phone calls back to the old (7960 like) mode which sends each digit separately. We try to detect automated dialing using (enbloc emulation) but with only 2 digits that's not going to succeed. Because we do not know if the complete number has been dialed yet (dialplan returns 'matches more'), we need to wait for more digits until timeout is reached. All of this is visible in the log, and does make sense.

- you should enable enbloc dialing in the phone settings. That will fix the whole issue for you.

Older phones (7960 etc, not capable of enbloc dialing):
- if you live in a country with fixed length extensions, you can make you dialplan fully qualified, which would always match completely on an extension, so that we do not have to wait until timeout. Switzerland has fixed 10 length numbers😄, Germany does not 😥.
- for emulated enbloc to work dialing the extension dialed needs to be at least 4-5 long
- if you normally enter digits before pressing the dial button, you can shorten the digittimout.

So the problem is not really an issue, it's just how it works. You can improve your experience though, using the comments above.

dkgroot, 2017-02-21:
Changed the prio. Waiting for reply.

Tenere750, 2017-02-23:
Set in SEP...cnf.xml the parameter (<useEnblocDialing>1</useEnblocDialing>) but the behaviour is just the same. This is the only parameter I found regarding enbloc dialing.

dkgroot, 2017-02-23:
Do you still see "SCCP: ENBLOC_EMU " debugging and digits being dialed separately, after setting this flag in the cnf.xml ?

Tenere750, 2017-02-23:
I do:
```
SCCP: ENBLOC_EMU digittimeout '8' ms, sched_wait '15804' ms
...
SCCP: ENBLOC_EMU digittimeout '8' ms, sched_wait '7880' ms
...
SCCP: ENBLOC_EMU digit entry time '120' ms, total dial time '120' ms, number of digits: 1
```

dkgroot, 2017-02-24:
Just to make sure we understand one one another:
Is you enter the number first and the press 'dial' the number is dialed immediately and you are happy. But if you go offhook first, and then enter the digits one by one, it takes a longer time before the number is dialed and you have to wait 'digittimeout' before the call is completed. correct ? 

If this is what you are seeing, than that is perfectly normal and how it should be. You can take steps to make this faster (for example by making your dialplan complely fixed length) or just shorten the digittimeout, or even dial a '#' after the number (which tells asterisk than dialing is complete.

I think i misunderstood your previous report. I though it was not dialing enbloc when you entered the number before pressing dial, but it does seem to do that.

Tenere750, 2017-02-24:
I'm sorry that I didn't express myself clear...

It doesn't matter if I enter the digits first and press "Dial" afterwards
or if I press "Dial" first and enter the digits afterwards.
In any case there appears another entry field and another dial button. If I
wait approx. 8 secs (7.88 secs regading to the debug output) the number is
dialed by the phone.

The only exception is pushing the speeddial button preconfigured with an
internal number. This dialing takes place immediately.

dkgroot, 2017-02-24:
But you are doing both of these 'dials' in offhook mode (so you already hear the dialtone), right ? What happens if you go onhook (hangup), then enter digits and then lift the receiver (go offhook), does it dial immediatly in that case (enbloc fial) ?

If all of the above is true, then it is executing exactly the way i would expect (ie: normal behaviour). There are however methods to shorten the time you have to wait for timeout (possibly bring it to 0 seconds).

Dialing via speeddial uses a completely other method and is not related :-).

Tenere750, 2017-02-24:
No, I do them onhook!

dkgroot, 2017-02-24:
Sorry for re-asking the same question again. I just want to make sure i am getting it right. I just hooked up my SPA 504G to confirm (can only have so many phones on my desk at any time). I think that you are correct, and that it does not do onhook dialing. I never notice, because i have fixed length dialplan, so don't have to wait for timeout, when dialing a correct number. Sorry for all the back and forth.

And it also looks like simulated enbloc dialing doesn't react well to the way the SPA sends it's digits (Non uniform pattern), so no luck there.

Options that remain:
- Enter '#' after the number (ugly but simple)
- Shorten the secondary digit timeout to 4 (Which seems fine for most people)
- Change your dialplan so that most/all extensions are finite. That way asterisk does not return 'match more' but 'Matches Exactly' when the number is complete and dial immediately.

Sorry this took longer than you and i hoped.

In the meantime i will try to figure out why this device does not support enbloc dialing to begin with, from the protocol level you would expect that.

dkgroot, 2017-02-24:
I am looking into the simulated enbloc dialing for SPA phones. The calculation of the standard deviation might have to be altered a little. 

simulated enbloc dialing, tries to capture the rythm of the digits being entered. Then they come in rapidly and with equal spacing, we assume it has to be a machine and not a person. Which would indicate you entered the digits before pressing dial. The SPA phone seems to wait a little before the second digit (maybe because it needs to change the dialtone), so simulated enbloc dial calculation may have to skip the first 1-2 digits before trying to establish the rythm.

Sorry for waisting a bit of your time on this issue. I just did not understand and assumed the SPA phones would do / be able to do enbloc dialing.

dkgroot, 2017-02-24:
Hi Joerg,
Have updated/fixed the emulated/simulated enbloc dialing code. Can you test this for me ?

git checkout issues/378
make && make reload

Minimum number of digits before emulated enbloc dialing will kick in is '4'

You can set 'sccp debug core, channel, action, pbx' to show what it is doing.

On the fourth digit it should do something like this:
```
 00000000 - 18 00 00 00 00 00 00 00  03 00 00 00 01 00 00 00  - ................
 00000010 - 01 00 00 00 04 00 00 00  00 00 00 00 00 00 00 00  - ................
    -- SEP381C1AB6B976: SCCP (handle_keypad) digit:00000001, callid:4, buttonIndex:1 => lineInstance:1
    -- SEP381C1AB6B976: SCCP (handle_keypad) digit:00000001, callid:4, lineInstance:1
    -- SEP381C1AB6B976: SCCP (handle_keypad) BOTH callid and lineinstance
    -- SEP381C1AB6B976: Looking for line with instance 1.
    -- SCCP: Found line 98031
    -- SEP381C1AB6B976: SCCP Digit: 00000001 (1) on line 98031, channel 4 with state: 12 (Using: SKINNY)
 SCCP: ENBLOC_EMU digittimeout '4' s, sched_wait '3839' ms
 SCCP: ENBLOC_EMU digit entry time '161' ms, total dial time '577' ms, number of digits: 4
 SCCP: ENBLOC EMU sqrt((85411 - (4 * (577/4)^2)/(4-1))='26.95'
 SCCP: ENBLOC EMU totaldigittimesquared '85411', totaldigittime '577', number_of_digits '4', std_deviation '26.95', variance '726.25'
 SCCP: ENBLOC EMU FAST DIAL (new timeout=2 sec)
    -- SCCP/98031-00000004: schedule digittimeout 2
    -- +- pbx extension matcher (98031          ): ---+
    -- |ignore     |exists     |can match  |match more|
    -- | no        |yes        |yes        | no       |
    -- +----------------------------------------------+
```

Don't worry about the calculation. What it tries to do is figure out if the digits all arrive with the same amount of space in between the digits. When the did, it sets the new digit timeout to 2 seconds (we still need to check if more are coming in).

Tenere750, 2017-02-27:
root@AsteriskPi:/usr/src/chan-sccp_develop# git checkout issues/378
error: pathspec 'issues/378' did not match any file(s) known to git.

...

dkgroot, 2017-02-27:
Hi Joerg,
You need to get the new branches from the server first, using
```
git fetch --all
git pull
git checkout issues/378
make && make reload
```
That should work.

dkgroot, 2017-03-25:
@Tenere750 Any update ?

dkgroot, 2017-09-08:
No further updates from issue owner

commit ea797e96 (2017-02-14): Fix: Device Cleaning causing Registration issues

Note: This does not fix the cause for the re-registration. I have not found out why devices sometimes:
- loose there connection
- do not respond until there session times out
- go into a device restart

Fixes #378
Fixes #333
commit e621a040 (2017-09-08): Merge branch 'issues/378' into develop

Changes Simulated Enbloc Dialing Calculation

Partially Fixes #378
closed 2017-09-08 by dkgroot