chan-sccp/chan-sccp #379: Asterisk CLI: sccp show mwi not accessible

Tenere750, 2017-02-15:
In Asterisk CLI the 'sccp show mwi' command is not accessible
```
AsteriskPi*CLI> sccp show
channels      device        devices       globals       hint          line          lines         mwi           refcount      sessions      softkeyssets  version
AsteriskPi*CLI> sccp show mwi
No such command 'sccp show mwi' (type 'core show help sccp show mwi' for other possible commands)
```

**Reproduction Steps:**

1. Just type 'sccp show mwi' in Asterisk CLI

**Expected behavior:**

Command should be accessible

No such command 'sccp show mwi' 

[Describe observed behavior here]

**Screenshots**

As mentioned above

**chan-sccp version:** (include revision and/or branch if relevant)

Skinny Client Control Protocol (SCCP). Release: 4.3.0 develop - f4024a6M 

**asterisk version:** (include revision and/or branch if relevant)

Asterisk 14.3.0

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: No
- Problem can be reliably reproduced, doesn't happen randomly: Yes
- Platform:
  - Raspberry Pi 2
  - [32/64 Bit]
  - [Big/Small Endian]
  - Distributor ID: Raspbian 8.0 (jessie)
  - Linux AsteriskPi 4.4.38-v7+ #938 SMP Thu Dec 15 15:22:21 GMT 2016 armv7l GNU/Linux
- Direct Contact information: bucholzki@googlemail.com

dkgroot, 2017-02-15:
Hi Tenere,

The actual command is 'sccp show mwi subscriptions', only full commands will generate output. 

Using the tab key in this case, would have helped you out though..

And you can issues this CLI command, to get a list of all matches
*CLI> `core show help sccp show`

Which would have generated:
```
            sccp show channels Show all SCCP channels.
         sccp show conferences Show running SCCP Conferences.
          sccp show conference Show SCCP Conference Info.
             sccp show devices Show all SCCP Devices.
              sccp show device Show an SCCP Device
             sccp show globals Show SCCP global settings.
     sccp show hint lineStates Show all hint lineStates
  sccp show hint subscriptions Show all hint subscriptions
               sccp show lines Show All SCCP Lines.
                sccp show line Show an SCCP Line.
   sccp show mwi subscriptions Show all mwi subscriptions
            sccp show refcount Test message.
            sccp show sessions Show All SCCP Sessions.
        sccp show softkeyssets Show all mwi configured SoftKeySets
             sccp show version SCCP show version.
```

closed 2017-02-15 by dkgroot

Tenere750, 2017-02-15:
Sorry, my fault...

dkgroot, 2017-02-15:
No need to be sorry :-)
BTW we do have a mailinglist, which might be better suited for these kind of little issues :-) Everybody would be able to help out, answering them.