chan-sccp/chan-sccp #380: Device Lines not Unloading on reload when line config changes

ijh-uk, 2017-02-22:
When we reconfigure lines on a device and reload it via sccp device reload, the old lines (and in some cases Blank button entries) persist and share ids with new versions. Only 'sccp reload force' will revert the configured device back to the configured status.

**Reproduction Steps:**

1. Configure device, buttons and line in sccp.conf and load (start or 'sccp reload force')
2. Change the line configuration (add a line, remove a line etc).
3. 'sccp reload device [MAC]' - the configured lines will be added to the end of the listing duplicating ids where it is the same line.

**Expected behavior:**

'sccp reload device [MAC]' should clear the device back to an 'empty' line button configuration with the lines (now) listed in the configuration file added to id's 1,2,3 etc. Each line/speeddial should have a unique instance ref (they seem to revert to 0) and the sub-tables should list each unique entity once (they duplicate). 

**Observed behavior:**

With debug turned on, messages are emitted claiming that the lines have been removed, eg:
  == SCCP: Remove Line 98021 from device SEP002545XXXXE7
  == SCCP: Remove Line 98014 from device SEP002545XXXXE7
  == SCCP: Remove Line 98021 from device SEP00254XXXXXE7
  == SCCP: Remove Line 98016 from device SEP00254XXXXXE7
  == SCCP: Remove Line 98021 from device SEP00254XXXXXE7
(note the lines are duplicated further illustrating that it has lost track of things)
**Screenshots**
Output of 'sccp show device SEP00254XXXXXXE7':
+--- Buttons ----------------------------------------------------------------------------------------------------------+
| Id   Inst TypeStr                                  Type                                  pendUpdt pendDel  Default   |
+ ==== ==== ======================================== ===================================== ======== ======== ========= +
| 1    0    Speeddial                                1                                     No       No       No        |
| 2    0    Line                                     0                                     No       No       No        |
| 3    0    Feature                                  3                                     No       No       No        |
| 1    0    Empty                                    4                                     No       No       No        |
| 1    0    Line                                     0                                     No       No       No        |
| 1    0    Speeddial                                1                                     No       No       No        |
| 2    0    Line                                     0                                     No       No       No        |
| 3    0    Feature                                  3                                     No       No       No        |
| 256  0    Line                                     0                                     No       No       No        |
| 1    0    Speeddial                                1                                     No       No       No        |
| 2    0    Line                                     0                                     No       No       No        |
| 3    0    Feature                                  3                                     No       No       No        |
+----------------------------------------------------------------------------------------------------------------------+

+--- LineButtons ------------------------------------------------------------------------------------------------------+
| Id   Name                    SubCid                Label                                 CfwdType   CfwdNumber       |
+ ==== ======================= ===================== ===================================== ========== ================ +
| 2    98021                                         Phone 2 Line 1                        None                        |
| 1    98014                                         FSC Line 4                            None                        |
| 2    98021                                         Phone 2 Line 1                        None                        |
| 256  98016                                         FSC Line 6                            None                        |
| 2    98021                                         Phone 2 Line 1                        None                        |
+----------------------------------------------------------------------------------------------------------------------+

+--- SpeeddialButtons -------------------------------------------------------------------------------------------------+
| Id   Name                    Number                 Hint                                                             |
+ ==== ======================= ====================== ================================================================ +
| 1    Home                    98016                                                                                   |
| 1    Home                    98016                                                                                   |
| 1    Home                    98016                                                                                   |
+----------------------------------------------------------------------------------------------------------------------+

+--- FeatureButtons ---------------------------------------------------------------------------------------------------+
| Id   Name                    Status                 Options                                                          |
+ ==== ======================= ====================== ================================================================ +
| 3    Private Call            0                      callpresent                                                      |
| 3    Private Call            0                      callpresent                                                      |
| 3    Private Call            0                      callpresent                                                      |
+----------------------------------------------------------------------------------------------------------------------+

Also note once in this state, reload device (with no changes) just exacerbates effects, generally adding the configured lines again to the end of the list. Unregister doesn't seem to have any effect on the phone, reset causes a full reset, but still loads the wrong configuration. Only sccp reload force clears the issue.

Issue observed with our own sccp configuration and something simple based on the sample config.

**chan-sccp version:** (include revision and/or branch if relevant)
First observed in 4.1 Stable (I think), also (definitely) an issue in 4.2 and 4.2.3 stable releases (all compiled). Now running git cloned current dev branch.

**asterisk version:** (include revision and/or branch if relevant)
Currently 13.13LTS (compiled) on CentOS 6 (other 12 & 13 branches observed to have same issue)

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: Unsure
- Problem can be reliably reproduced, doesn't happen randomly: Yes, seems to be every time
- Platform:
  - Xeon X3220
  - 64 Bit
  - Big Endian
  - CentOS 6.7

- Direct Contact information: ian [at] friendly service centre [dot] com
- Log Files (see above, let me know what else will help)
- Comments: [Other Information]

ijh-uk, 2017-02-22:
Further (minor) note: disregard the line with id 256 - this was manually added (after it had gone astray) with sccp add line to see if it still worked or would solve the issue. Behaviour has been observed with or without any use of sccp add line...

dkgroot, 2017-02-22:
Question: have you checked the config parsing, and the decision it comes too ? You can do this with debug=core,config,high and doing a reload. It will show you all the lines encountered and how they were interpreted. 

The easiest way for me to test /check and hopefully fix this is if you provide two sccp.conf files (before and   after), so that I can do a 'sccp reload sccp.conf.after', to switch versions. Can you provide these two sample files.

Above it is not clear which branch you are on, 'git branch' should tell you. Also include the revision you are using. Making it easier for me to check.

Thanks for reporting the issue btw.

ijh-uk, 2017-02-22:
Configuration as below (with comments stripped out via grep -v ‘^;’) – I add only the line ‘button = line, 98014’ to the end / the last phone block and sccp reload that device, and I end up with seven lines listed (original 3 x 2 + the new one) (see show device extract).

 

SCCP Reload log is attached (with ‘sccp debug core config high’ run first – I’m not sure that I can see it processing the lines etc).

 

Release is: Skinny Client Control Protocol (SCCP). Release: 4.3.0 develop - 78fb33dM (built by 'root' on '2017-02-22 17:44:48 UTC'), branch shows as develop.

 

Sincere thanks for all your efforts, let me know what else I can do! (will have to get my backside in gear and do some documentation work once I get it working properly!)

 

Thanks

 

Ian

 

***sccp show device SEP… ****

 

+--- Buttons ----------------------------------------------------------------------------------------------------------+

| Id   Inst TypeStr                                  Type                                  pendUpdt pendDel  Default   |

+ ==== ==== ======================================== ===================================== ======== ======== ========= +

| 1    0    Speeddial                                1                                     No       No       No        |

| 2    0    Line                                     0                                     No       No       No        |

| 3    0    Feature                                  3                                     No       No       No        |

| 1    0    Speeddial                                1                                     No       No       No        |

| 2    0    Line                                     0                                     No       No       No        |

| 3    0    Feature                                  3                                     No       No       No        |

| 4    0    Line                                     0                                     No       No       No        |

+----------------------------------------------------------------------------------------------------------------------+

 

+--- LineButtons ------------------------------------------------------------------------------------------------------+

| Id   Name                    SubCid                Label                                 CfwdType   CfwdNumber       |

+ ==== ======================= ===================== ===================================== ========== ================ +

| 2    98021                                         Phone 2 Line 1                        None                        |

| 2    98021                                         Phone 2 Line 1                        None                        |

| 4    98014                                         FSC Line 4                            None                        |

+----------------------------------------------------------------------------------------------------------------------+

 

***sccp.conf****

[general]

servername = Asterisk!

keepalive = 60

context = default

dateformat = D.M.Y

bindaddr = 0.0.0.0

port = 2000

disallow=all

allow=ulaw

allow=g729

firstdigittimeout = 16

digittimeout = 8

deny=0.0.0.0/0.0.0.0

permit=192.168.203.0/255.255.255.0

 

[Standard_Phone](!)

description = Phone Number One

addon = 7914

devicetype = 7960

park = off

type = device

 

[98011]

id = 1000

type = line

label = Phone 1 Line 1

description = Line 98011

cid_name = MY CID

cid_num = 98011

context = default

 

[98012]

id = 1001

type = line

label = Phone 1 Line 2

description = Line 98012

cid_name = MY LINE 2

cid_num = 98012

context = default

 

[98013]

id = 1003

type = line

label = FSC Line 3

description = Line 3

[98014]

id = 1004

type = line

label = FSC Line 4

description = Line 4

[98015]

id = 1005

type = line

label = FSC Line 5

description = Line 5

[98016]

id = 1006

type = line

label = FSC Line 6

description = Line 6

 

[SEP001DA23F4A4A](Standard_Phone)

description = Phone Number Two

devicetype = 7960

button = speeddial,Helpdesk, 98112                              ; Add SpeedDial to Helpdesk (without hint)

button = line, 98021

button= feature,Private Call,privacy,callpresent                ;set channel variable SKINNY_PRIVATE to 1 if feature is enabled

button= feature,DND Busy,DND,busy                               ;set dnd status to busy

button= feature,DND Silent,DND,silent                           ;set dnd status to silent

button= feature,Record calls,monitor                            ;record calls using automon (asterisk >= 1.6 only)

button= feature,call forward to *54,cfwdAll,*54                 ;forward all calls to *54

button = speeddial,Phone 1 Line 1, 98011, 98011@hints

button = speeddial,Phone 1 Line 2, 98012, 98012@hints

type = device

keepalive = 60

 

[98021]

id = 1000

type = line

pin = 1234

label = Phone 2 Line 1

description = Line 98021

cid_name = MY CID

cid_num = 98021

context = default

 

[SEP00254594F5E7](Standard_Phone)

description = TestPhone

devicetype = 7960

park = off

button = speeddial,Home,98016                                          ; Add SpeedDial to Helpdesk

button = line, 98021                                          ; Assign Line 98011 to Device and use this as default line

button = feature,Private Call,privacy,callpresent                                          ; Assign an Empty Button

type = device

ijh-uk, 2017-02-22:
[reload_debug.log.txt](https://github.com/chan-sccp/chan-sccp/files/795125/reload_debug.log.txt)

(Debug log which I think was stripped out of the last post due to me using a proper extension ;-))

dkgroot, 2017-02-22:
I guess I should have warned you. Getting debugging from config parsing is an exception to the rule. It does not help setting the debug level on the commandline in this case, because the first thing the config parser will do is read and interpret the debug= entry in the sccp.conf file 😛

Maybe you can send me the two test config files per mail (dkgroot at talon.nl). I want to make sure I get all the detail, instead of cutting and pasting.

ijh-uk, 2017-02-23:
Thanks and sorry for the etiquette violations / laziness!

 

I have direct emailed you the files.

 

As a side note: it looks like the issue applies irrespective of whether or not the phone is registered / active (I have just tested with it totally offline).

dkgroot, 2017-02-23:
Hi Ian,
There where no etiquette infractions what so ever. By having your before and after files, i can run the test and see what needs to be fixed. I have had problems reproducing issues in the past, so better to get the "correct" files and test with them.

dkgroot, 2017-02-24:
new debug logging for buttonconfig parse, to show what is going on:
```
...
     -- SCCP: parsing device parameter: button = 'speeddial,Home,98016'  (line: 55)
        > Token button/button
        > Create new variable set (button=speeddial,Home,98016)
        > Add to variable set (button=line, 98021)
        > Add to variable set (button=feature,Private Call,privacy,callpresent)
        > Add to variable set (button=line, 98014)
   == buttonconfig status before check
     -- index:0, type:Speeddial  (1), pendingDelete:True, pendingUpdate:False
     -- index:1, type:Line       (0), pendingDelete:True, pendingUpdate:False
     -- index:2, type:Feature    (3), pendingDelete:True, pendingUpdate:False
     -- SCCP: Checking Button Config
     -- Checking button: speeddial,Home,98016
        > Found Button index at 0:0
        > SCCP: Speeddial Button Definition remained the same
        > SCCP: ButtonTemplate remained the same
     -- Checking button: line, 98021
        > Found Button index at 1:1
        > SCCP: ComposedId extension: 98021, subscriptionId[number:, name:, label:, aux:]
        > SCCP: Line Button Definition remained the same
        > SCCP: ButtonTemplate remained the same
     -- Checking button: feature,Private Call,privacy,callpresent
        > Found Button index at 2:2
        > SCCP: Feature Button Definition remained the same
        > SCCP: Feature Button Definition changed
        > SCCP: ButtonTemplate remained the same
     -- Checking button: line, 98014
        > SCCP: ButtonTemplate has changed
     -- Button: line, 98014 changed. Giving up on checking buttonchanges, reloading all of them.
   == buttonconfig status after check
     -- index:0, type:Speeddial  (1), pendingDelete:True, pendingUpdate:False
     -- index:1, type:Line       (0), pendingDelete:True, pendingUpdate:False
     -- index:2, type:Feature    (3), pendingDelete:True, pendingUpdate:False
     -- Any Previous ButtonConfig will be discared during post-process                   <!-- replacement for the 'little lie' -->
        > SCCP: Loading New Button Config
        > New Speeddial Button 'Home' at : 0:0
        > SCCP: Speeddial Button Definition
        > Added button: speeddial,Home,98016
        > SCCP: Loading New Button Config
        > New Line Button '98021' at : 1:1
        > SCCP: Line Button Definition
        > SCCP: ComposedId extension: 98021, subscriptionId[number:, name:, label:, aux:]
        > Added button: line, 98021
        > SCCP: Loading New Button Config
        > New Feature Button 'Private Call' at : 2:2
        > SCCP: Feature Button Definition
        > Added button: feature,Private Call,privacy,callpresent
        > SCCP: Loading New Button Config
        > New Line Button '98014' at : 3:3
        > SCCP: Line Button Definition
        > SCCP: ComposedId extension: 98014, subscriptionId[number:, name:, label:, aux:]
        > Added button: line, 98014
   == buttonconfig status after adding new buttons                                       <!-- here you see what it actually did, namely old and new buttons have both been added to the list of buttons -->
     -- index:0, type:Speeddial  (1), pendingDelete:True, pendingUpdate:False
     -- index:1, type:Line       (0), pendingDelete:True, pendingUpdate:False
     -- index:2, type:Feature    (3), pendingDelete:True, pendingUpdate:False
     -- index:0, type:Speeddial  (1), pendingDelete:False, pendingUpdate:False
     -- index:1, type:Line       (0), pendingDelete:False, pendingUpdate:False
     -- index:2, type:Feature    (3), pendingDelete:False, pendingUpdate:False
     -- index:3, type:Line       (0), pendingDelete:False, pendingUpdate:False
     -- buttonconfig: changed
   == SCCP: config parameter button='speeddial,Home,98016' in line 55 changed. (causes device reset)
....
  Checking Reading Type
   == Line Post Reload
   == Device Post Reload
  SCCP: (post_reload)                                                                        <!-- post process is going to fix everything up -->
   == SEP001B54CA499B (check_update) pendingUpdate: TRUE, pendingDelete: FALSE
[... 25 lines cut ...]
     -- SEP001B54CA499B: (sccp_devstate_deviceRegisterListener) device unregistered
  SEP001B54CA499B: LineDevice FREE 0x7faa00008e80
   == Softkey Post Reload
     -- Re-attaching softkeyset: default to device d: SEP001B54CA499B
     -- SEP001B54CA499B: Unregister request Received (Reason: Powersave)
     -- SEP001B54CA499B: Registration State is 'None'
  SCCP: Clean Device SEP001B54CA499B                                                                        <!-- called by device unregister -->
                                                                                              <!-- this should have not been necessary, but doesn't really do any harm either -->
     -- SEP001B54CA499B: Registration State is 'Cleaning'
   == SEP001B54CA499B: checking buttonconfig index:1, type:Line (0) to see if there are any connected lines/channels
   == SCCP: Remove Line 98021 from device SEP001B54CA499B
     -- SEP001B54CA499B: remove device from line 98021
   == SEP001B54CA499B: checking buttonconfig index:3, type:Line (0) to see if there are any connected lines/channels
   == SCCP: Remove Line 98014 from device SEP001B54CA499B
     -- SEP001B54CA499B: remove device from line 98014
   == SEP001B54CA499B: checking buttonconfig for pendingDelete (index:0, type:Speeddial (1), pendingDelete:False, pendingUpdate:False)
[... cut ...]
   == SEP001B54CA499B: checking buttonconfig for pendingDelete (index:2, type:Feature (3), pendingDelete:False, pendingUpdate:False)
   == SEP001B54CA499B: checking buttonconfig for pendingDelete (index:3, type:Line (0), pendingDelete:False, pendingUpdate:False)
   == SCCP: Unregister Device SEP001B54CA499B
     -- SEP001B54CA499B: (sccp_devstate_deviceRegisterListener) device unregistered
     -- SEP001B54CA499B: Registration State is 'None'
...
```
Hopefully this is informative about the button parse process

dkgroot, 2017-02-24:
We can use [gitter chat](https://gitter.im/chan-sccp/chan-sccp) instead of mail if you like.

ijh-uk, 2017-02-24:
Having re-run my test files using the issue branch with the debugging turned on (and the device debug flag additionally specified): the behaviour does not change, but the log reveals that delete flags are never being correctly set at sccp reload device on our system.

 

I will test the additional change proposed later today and see if that solves the problem.

 

Thanks,

 

Ian

==== SCCP Startup from Reload Force ======
2017-02-24 10:24:11.1] VERBOSE[23359] :   == buttonconfig status before check
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- index:0, type:Speeddial  (1), pendingDelete:True, pendingUpdate:False
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- index:1, type:Line       (0), pendingDelete:True, pendingUpdate:False
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- index:2, type:Feature    (3), pendingDelete:True, pendingUpdate:False
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- index:0, type:Speeddial  (1), pendingDelete:True, pendingUpdate:False
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- index:1, type:Line       (0), pendingDelete:True, pendingUpdate:False
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- index:2, type:Feature    (3), pendingDelete:True, pendingUpdate:False
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- index:3, type:Line       (0), pendingDelete:True, pendingUpdate:False
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- SCCP: Checking Button Config
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- Checking button: speeddial,Home,98016
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > Found Button index at 0:0
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > SCCP: Speeddial Button Definition remained the same
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > SCCP: ButtonTemplate remained the same
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- Checking button: line, 98021
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > Found Button index at 1:1
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > SCCP: ComposedId extension: 98021, subscriptionId[number:, name:, label:, aux:]
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > SCCP: Line Button Definition remained the same
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > SCCP: ButtonTemplate remained the same
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- Checking button: feature,Private Call,privacy,callpresent
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > Found Button index at 2:2
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > SCCP: Feature Button Definition remained the same
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > SCCP: Feature Button Definition changed
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > SCCP: ButtonTemplate remained the same
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- Number of Buttons changed (7 != 3). Reloading all of them.
[2017-02-24 10:24:11.1] VERBOSE[23359] :   == buttonconfig status after check
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- index:0, type:Speeddial  (1), pendingDelete:True, pendingUpdate:False
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- index:1, type:Line       (0), pendingDelete:True, pendingUpdate:False
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- index:2, type:Feature    (3), pendingDelete:True, pendingUpdate:False
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- index:0, type:Speeddial  (1), pendingDelete:True, pendingUpdate:False
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- index:1, type:Line       (0), pendingDelete:True, pendingUpdate:False
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- index:2, type:Feature    (3), pendingDelete:True, pendingUpdate:False
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- index:3, type:Line       (0), pendingDelete:True, pendingUpdate:False
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- Any Previous ButtonConfig will be discared during post-process
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > SCCP: Loading New Button Config
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > New Speeddial Button 'Home' at : 0:0
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > SCCP: Speeddial Button Definition
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > Added button: speeddial,Home,98016
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > SCCP: Loading New Button Config
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > New Line Button '98021' at : 1:1
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > SCCP: Line Button Definition
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > SCCP: ComposedId extension: 98021, subscriptionId[number:, name:, label:, aux:]
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > Added button: line, 98021
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > SCCP: Loading New Button Config
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > New Feature Button 'Private Call' at : 2:2
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > SCCP: Feature Button Definition
[2017-02-24 10:24:11.1] VERBOSE[23359] :        > Added button: feature,Private Call,privacy,callpresent
[2017-02-24 10:24:11.1] VERBOSE[23359] :   == buttonconfig status after adding new buttons
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- index:0, type:Speeddial  (1), pendingDelete:True, pendingUpdate:False
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- index:1, type:Line       (0), pendingDelete:True, pendingUpdate:False
[2017-02-24 10:24:11.1] VERBOSE[23359] :     -- index:2, type:Feature    (3), pendingDelete:True, pendingUpdate:False
[... cut ...]
[2017-02-24 10:24:32.5] VERBOSE[23367] :   == SCCP: Unregister Device SEP00254594F5E7
[2017-02-24 10:24:32.5] VERBOSE[23367] :     -- SEP00254594F5E7: Registration State is 'None'
[2017-02-24 10:24:32.5] VERBOSE[23367] asterisk.c: Remote UNIX connection disconnected
[2017-02-24 10:24:32.5] VERBOSE[23206] :     -- SEP00254594F5E7: (sccp_devstate_deviceRegisterListener) device unregistered
[2017-02-24 10:24:40.9] VERBOSE[23138] asterisk.c: Remote UNIX connection
[2017-02-24 10:24:40.9] VERBOSE[23374] asterisk.c: Remote UNIX connection disconnected

dkgroot, 2017-02-24:
I see where it is going wrong. In the pre-process before reload, the buttons are not marked as pendingDelete. That clarifies things. WIll try to find the cause of this.

No need to test the change in sccp_actions.c, will not help (will not harm either).

dkgroot, 2017-02-24:
Have not been able to find the cause yet. Can you do another pull of issues/380. And set 
debug=core,config,device
in your sccp.conf and doing another one of these reloads (like in the last version).

dkgroot, 2017-02-24:
There was a little mis-communcation between using 'sccp reload' and 'sccp reload device SEPxxxxx', which caused the issue to drag on a little :-)
Issue is fixed and branch has been merged back into develop.

commit 7ad412a2 (2017-02-24): Fix: Cleanup sccp reload device/line

Closed #380
closed 2017-02-24 by dkgroot