chan-sccp/chan-sccp #381: Transfer on 6901 (one way audio with transfer recipient)

dkgroot, 2017-02-27:
Diogo reported issues when trying to transfer a call using a 6901

First issue, not being able to dial the transfer recipient
Analysis: was caused by focus shifting back to transfered
Fixed in source. Was being cause by a selectsoftkeys to the wrong callid.

Next issue:  one way audio with the transfer recipient
Analysis: the start media transmission ack message is missing the passThruPartId is missing on the second call.

```
[2017-02-27 19:42:02] NOTICE[29647]: sccp_session.c:989 sccp_session_send2: SEP3C5EC30C4C4B: Send Message: Start Media Transmission(0x008A) 136 bytes length
 SCCP: packet hex dump:
 00000000 - 88 00 00 00 14 00 00 00  8A 00 00 00 15 00 00 00  - ................
 00000010 - EA FF FF FF 00 00 00 00  0A 01 02 23 00 00 00 00  - ...........#....
 00000020 - 00 00 00 00 00 00 00 00  B4 28 00 00 14 00 00 00  - .........(......
 00000030 - 02 00 00 00 B8 00 00 00  00 00 00 00 00 00 00 00  - ................
 00000040 - 00 00 00 00 15 00 00 00  00 00 00 00 00 00 00 00  - ................
 00000050 - 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  - ................
 00000060 - 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  - ................
 00000070 - 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  - ................
 00000080 - 65 00 00 00 0A 00 00 00  00 00 00 00 00 00 00 00  - e...............
    -- SEP3C5EC30C4C4B: send callInfo on SCCP/22-00000015 with lineInstance: 1
    -- SEP3C5EC30C4C4B: >> Send message Stop Tone Message
[2017-02-27 19:42:02] NOTICE[29647]: sccp_session.c:989 sccp_session_send2: SEP3C5EC30C4C4B: Send Message: Stop Tone Message(0x0083) 16 bytes length
 SCCP: packet hex dump:
 00000000 - 10 00 00 00 14 00 00 00  83 00 00 00 01 00 00 00  - ................
 00000010 - 15 00 00 00 00 00 00 00                           - ........
    -- SEP3C5EC30C4C4B: >> Got message Start Media Transmission Acknowledge (0x154)
[2017-02-27 19:42:02] NOTICE[29647]: sccp_actions.c:121 check_session_message_device: SEP3C5EC30C4C4B: SCCP Handle Message: Start Media Transmission Acknowledge(0x0154) 52 bytes length
 SCCP: packet hex dump:
 00000000 - 34 00 00 00 14 00 00 00  54 01 00 00 00 00 00 00  - 4.......T.......
 00000010 - 00 00 00 00 15 00 00 00  00 00 00 00 0A 01 02 2C  - ...............,
 00000020 - 00 00 00 00 00 00 00 00  00 00 00 00 B4 28 00 00  - .............(..
 00000030 - 00 00 00 00 00 00 00 00  00 00 00 00              - ............
    -- SEP3C5EC30C4C4B: Getting the active channel on device.
[2017-02-27 19:42:02] WARNING[29647]: sccp_actions.c:3511 handle_startmediatransmission_ack: SEP3C5EC30C4C4B: Channel with passthrupartyid 0 / callid 0 / callid1 21 not found, please report this to developer
```

Phone debug log:
```
ACOS debuger --> 19:41:47:078 % [cent] FSM event(RA_TIMEOUT) / state(ACTIVE_CALL) / cause(0)
19:41:47:079 w [cent] Unexpected event(RA_TIMEOUT) / cause(0) / at state(ACTIVE_CALL)
19:41:48:967   [MMI] <API>: Rpt Hook-Key, ID(2)
19:41:48:967   [MMI] <VC>: [CONN]->holdKey(18)
19:41:48:968 i [call] MMI key 17
19:41:48:968   [MMI] <AUD>: set AT (3)->(3) enable=1
19:41:48:969   [MMI] <VC>: procRet viewIdx(2)-[CONN]=> retAct(1) retView(5)
19:41:48:969   [call] sapputil_setLastSkEvent
19:41:48:993 i [call] [SAPPMAIN] sappmain_gapi_event_process () msgid=81
19:41:48:993   [call] Stop xmit rtp stream now. callref=20
19:41:48:993 i [call] sccp in 008b
19:41:49:005 i [call] [SAPPMAIN] sappmain_gapi_event_process () msgid=79
19:41:49:006   [call] Keep receive channel
19:41:49:006 i [call] sccp in 0106
19:41:49:017 i [call] [SAPPMAIN] sappmain_gapi_event_process () msgid=104
19:41:49:017   [call] *Synced TimeData is (2017-2-27) (day of week:1) (19:41:49)
19:41:49:018   [call] [callclient_startTimeDateTimer][3973]stationSeconds=70909
19:41:49:018   [call] after 26291 seconds to send timeDate req
19:41:49:018 i [call] sccp in 0094
19:41:49:060 i [call] [SAPPMAIN] sappmain_gapi_event_process () msgid=100
19:41:49:060 i [call] setLamp stimulus:9, mode:3
19:41:49:060   [MMI] <API>: Set Led Blink, Led(0)
19:41:49:061 i [call] sccp in 0086
19:41:49:076 i [call] [SAPPMAIN] sappmain_gapi_event_process () msgid=69
19:41:49:076 % [call] line 4726: Focus changed from 01:0004 --> 01:0004
19:41:49:077 i [call] update call: line=1, conn=4, state=8
19:41:49:077   [MMI] <API>: Upd Call, Line(1), Call(4), State(8),
19:41:49:078   [MMI] <VC>: [CONN]->updCall(4) mask(2)
19:41:49:078   [MMI] <AUD>: set AT (3)->(0) enable=1
19:41:49:078   [MMI] <AUD>: In mmiAudio_setAudioType line 1060 gDspLoad=1 bDspLoad=0 blQueueCfgMedia=0!
[... 114 lines cut ...]
19:41:51:222   [MMI] <VC>: [CONN]->setTone(8) event(2) dir(0) side(5041036)
19:41:51:223   [MMI] <AUD>: unloadDsp: Can not unload DSP (3, 0, 0, 2, 1, 1)!
19:41:51:223   [MMI] <VC>: procRet viewIdx(2)-[CONN]=> retAct(1) retView(5)
19:41:51:224 i [call] sccp in 0083
19:41:51:247 i [call] [SAPPMAIN] sappmain_gapi_event_process () msgid=119
19:41:51:247   [call]  sapp main receive GAPI_MSG_SELECTSOFTKEYS msgdata->conn_id=5
19:41:51:247 % [call] line 6155: Focus changed from 01:0005 --> 01:0005
19:41:51:248 i [call] sccp in 0110
19:41:51:283   [MMI] <API>: Rpt Key-Up, ID(6)
19:41:51:284   [MMI] <VC>: [CONN]->releaseKey(6)
19:41:51:284   [MCU]  mcu_specificToneStop(), tone(DT)
19:41:51:284   [MMI] <VC>: procRet viewIdx(2)-[CONN]=> retAct(1) retView(5)
19:41:51:718 % [cent] FSM event(RA_TIMEOUT) / state(ACTIVE_CALL) / cause(0)
19:41:51:719 w [cent] Unexpected event(RA_TIMEOUT) / cause(0) / at state(ACTIVE_CALL)
19:41:53:719 % [cent] FSM event(RA_TIMEOUT) / state(ACTIVE_CALL) / cause(0)
19:41:53:719 w [cent] Unexpected event(RA_TIMEOUT) / cause(0) / at state(ACTIVE_CALL)
19:41:55:552 i [call] sccp in 0100
19:41:55:719 % [cent] FSM event(RA_TIMEOUT) / state(ACTIVE_CALL) / cause(0)
19:41:55:719 w [cent] Unexpected event(RA_TIMEOUT) / cause(0) / at state(ACTIVE_CALL)
19:41:57:719 % [cent] FSM event(RA_TIMEOUT)
```

Note: 6901 phones do not have a transfer key, they use hookflash to instantiate the transfer process

Call above used transfer_on_hangup to complete the second stage.

dkgroot, 2017-02-27:
Related #335

dkgroot, 2017-02-27:
Changes:
- Added exception in startMediaTransmissionAck Handling for 6901
- Close Port during HOLD
- Added commented section about protocol 20 startmediatransmission and 69XX phones.

dkgroot, 2017-02-28:
Previous changes works. Transfer is now working

There was still an issue, namely cancelling a transfer in progress. According to the cisco manual the user should press the Line Button (strange but ok). I though it would make more sense to have the user press the hold key instead. But maybe that last change is not what the users wants. Maybe they want to consult back an forth between both transfer parties (need to test/check/not even sure if that would work on a 6901).

Changes:
-Fix: allow single line device (like 6901) to cancel transfer by pressing the line key (see cisco manual for 6901)
-Enh: allow single line device (like 6901) to cancel transfer by pressing the hold key on the transferer channel

FilipeDiogo, 2017-02-28:
With revision 7725952 it appears that the transfer is working and the resume for the 1st call on hold is working to! no more problems with no audio.... 

👍

dkgroot, 2017-02-28:
Thanks for the update. Will merge this issues/381 branch back into develop tomorrow, after reviewing the changes and maybe clean up.

Thanks for helping out with all the tests.

@voipasteriskguy can you verify that these changes also work for issue #335 ?

dkgroot, 2017-02-28:
@FilipeDiogo :
The issues/381 branch has been closed and merged into the develop branch. Please do:
```
git checkout develop
git pull
make && make reload
```
To get back to the 'develop' branch

commit d106aeb0 (2017-02-28): Enh: Remove commented out lines (regarding 6901)

Fixes #381
Fixes #335
closed 2017-02-28 by dkgroot