chan-sccp/chan-sccp #384: chan_sccp.c:123 load_config: Failed to bind to 0.0.0.0:3100: Address already in use!

pnlarsson, 2017-03-03:
With develop branch we are sometimes getting failure when doing an unload / load of chan_sccp, waiting a little bit and trying again loads chan_sccp as it should. netstat -lpn does not show that the port is in use.

**Reproduction Steps:**

1. Have asterisk running
2. module unload chan_sccp.so
3. module load chan_sccp.so

**chan-sccp version:** (include revision and/or branch if relevant)
Skinny Client Control Protocol (SCCP). Release: 4.3.0 develop - e6fd157M (pulled before building - 5c9f2faaabd5c0849feec83e33bbbc1228459375)

**asterisk version:** (include revision and/or branch if relevant)
Asterisk 13.14.0 built by root @ uc02 on a x86_64 running Linux on 2017-02-27 22:16:36 UTC

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Yes/No]
Yes
- Problem can be reliably reproduced, doesn't happen randomly: [Yes/No]
No
```
[2017-03-03 12:17:04] NOTICE[12462]: sccp_config.c:2583 sccp_config_readDevicesLines:  Checking Reading Type
[2017-03-03 12:17:04] ERROR[12462]: chan_sccp.c:123 load_config: Failed to bind to 0.0.0.0:3100: Address already in use!
[2017-03-03 12:17:04] ERROR[12462]: ast113.c:3505 load_module: SCCP: config file could not be parsed
[2017-03-03 12:17:04] ERROR[12462]: ast113.c:3550 load_module: SCCP: Module Load Failed, unloading...
[2017-03-03 12:17:04] NOTICE[12462]: ast113.c:3424 unload_module: SCCP: Module Unload
```
Log with debug all: https://gist.github.com/pnlarsson/53ee5cc86af93c78965b03bd958287a4

dkgroot, 2017-03-05:
Hi Niklas,

I noticed this issue as well, just had not had time to look into it. 
I have started a new branch 'issues/384' to work on the issue. To prevent future issues and or regressions, i am going to make these functions easier to test, while fixing the issue as well.

Maybe you can check if this first iteration already improves the issue you were seeing.
```
git fetch --all
git pull
git checkout issues/384
make && make install
```
To make testing a little easier, you can call 'make reload' after the first installation (when you are running the new version).

Thanks for reporting the issue !

pnlarsson, 2017-03-06:
Will test this on a couple of boxes - will know tomorrow if it helped

pnlarsson, 2017-03-06:
Well, that failed badly: https://gist.github.com/pnlarsson/4a548c36b7479c12010f0880fd8ca56b - starting asterisk with: asterisk -cvvdd and with noload => chan_sccp.so - and then loading chan_sccp gives that gist. 

Starting in a normal way made the astierk quit after a few minutes.

dkgroot, 2017-03-06:
Hi Niklas,

I am not sure what happened in the logging, to be honest. 
The "sccp_cli.c:3468 sccp_register_cli: == Cli registered action (null)" message is a indication of something really bad happening, but not sure what exactly.

Posted a change, to potentially fix this debug output.

The "Asterisk cleanly ending(0);" doesn't look good either.

Could you use gdb to do some digging, to see what's going on here ? I think the cli output is just a side effect, but the actual cause of the issue happens before this point.

I was expecting that you would run this in a test environment first (issues/384 branch is not complete nor fully tested yet).

dkgroot, 2017-03-25:
@pnlarsson 
Can you try making this change to develop branch, to see if this fixes the issues you are having.
```
diff --git a/src/sccp_session.c b/src/sccp_session.c
index 08dbbc5..ea276ad 100644
--- a/src/sccp_session.c
+++ b/src/sccp_session.c
@@ -763,7 +763,7 @@ static void sccp_accept_connection(void)
                sccp_free(s);
                return;
        }
-       sccp_netsock_setoptions(new_socket, /*reuse*/ 0, /*linger*/ 0, GLOB(keepalive) * 2);
+       sccp_netsock_setoptions(new_socket, /*reuse*/ 0, /*linger*/ 1, GLOB(keepalive) * 2);
        
        memcpy(&s->sin, &incoming, sizeof(s->sin));
        sccp_mutex_init(&s->lock);
```

dkgroot, 2017-08-30:
References  issues/384_1 branch

dkgroot, 2017-09-13:
@pnlarsson / @voipAsteriskGuy
Were you guys able to check/test issues/384_1 and did it resolve the issue. Would like to merge it into develop :-)

pnlarsson, 2017-09-13:
Will try it out now

pnlarsson, 2017-09-14:
Hi,

did an unload of old version and loaded the new (w/o restarting asterisk) and it failed a bunch of times before loading:

    -- Starting bind and listen!
[2017-09-14 13:05:20] ERROR[4316]: sccp_session.c:901 sccp_session_bind_and_listen: Failed to bind to 0.0.0.0:3100: Address already in use!
[2017-09-14 13:05:20] ERROR[4316]: ast113.c:3538 load_module: SCCP: postPBXLoad Failed
[2017-09-14 13:05:20] ERROR[4316]: ast113.c:3545 load_module: SCCP: Module Load Failed, unloading...
[2017-09-14 13:05:20] NOTICE[4316]: ast113.c:3419 unload_module: SCCP: Module Unload
 SCCP: Unloading Module

Should it have worked going develope -> issues/384_1 or is it issues/384_1 -> issues/384_1 that should be working

dkgroot, 2017-09-14:
Looks like you are on the right version, it's a shame it's not working though. I did test it multiple times. I don't know why you are getting different results.

Do not understand what you mean with the last sentence. When just do `git checkout issues/384_1` it should be ok. What you the run ` git branch` it should show the * in front of the issues/384_1 line.

The only good thing it that it does refuse to load, when it could not bind. This is a change from previous behaviour. Not sure how to proceed from this point. How many devices do you have connected to the server, and what does `netstat -atn | grep 2000` generate as output when the port is locked ?

dkgroot, 2017-09-15:
Only unloading/loading the new issues/386_1 should work. Going from develop over to 384_1 does not help, because the develop version is setting the wrong flags, which the new one cannot fix on the fly.  Took me a little while to understand your last sentence. Hope you also tested the 384_1->384_1 way.

pnlarsson, 2017-09-20:
I tried again going from 348_1 -> 348_1 and it worked without any issue. So I would say that it's fixed. Normally I have to wait for some use (hours) before it happens, this time it has been up for a day or two. So I am happy

pnlarsson, 2017-09-20:
Arg wrong button

dkgroot, 2017-09-21:
@pnlarsson 
Thank you so much for test / re-checking this one !

The issue/384_1 branch has been merged into develop.

commit 9e0d7f43 (2017-09-21): Merge branch 'issues/384_1' into develop

Fixes #384
closed 2017-09-21 by dkgroot