chan-sccp/chan-sccp #385: DTMF issues on active call upon another incoming call ringing in

abradfordcms, 2017-03-03:
DTMF issues on active call upon another incoming call ringing in

**Reproduction Steps:**

1. Have phones in a ring group in Asterisk
2. Have a user on a SCCP phone be on an outbound call that involves an IVR
3. Incoming call rings party that is on the phone with IVR that is trying to input digits
4. User cannot proceed in having the channel they're on receive any DTMF during call notification or once it has cleared from their endpoint's display 

**Expected behavior:**

DTMF should proceed to be passed through the active channel without interruption or cease

**Observed behavior:**

Upon an incoming call coming into the ring group, the party that is trying to send DTMF to an IVR that they're currently on the phone with loses DTMF ability 

**Screenshots**

N/A

**chan-sccp version:**

4.3.0 - rev 48021edM

**asterisk version:** 

11.10.0

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Both]

This occurred in previous versions, was corrected temporarily by Marcello, and now appears to be back. 

- Problem can be reliably reproduced, doesn't happen randomly: [Yes]

Can usually be reproduced. Seems to also happen after a long amount of Asterisk uptime, as of this last report. 

- Platform:

64 bit
CentOS 6.5

- Direct Contact information: andy.bradford.cms@gmail.com
- Log Files (attached) 
[call_1.txt](https://github.com/chan-sccp/chan-sccp/files/817611/call_1.txt)
[call_2.txt](https://github.com/chan-sccp/chan-sccp/files/817610/call_2.txt)

dkgroot, 2017-03-04:
Hi Andy/Everybody,

Thanks for bringing this to my attention. The solution used to fix issues #330 does work, but only newer (java based) phones, because it uses a message that does not exist on older devices. The older devices use a different (ugly) method to suppress phonebook entries. The fix caused other issues with older type phones, so i reverted it. Asumming that newer devices would be able to use the older (uglier) method as well.

To fix the issue (once and for all :-)) I created a single function to be called for all types of phone, which does suppress calls answered elsewhere) correctly.

To make sure the new version works correctly, would you mind testing it for me, using 'issues/385' branch ? Please do:
```
git fetch --all
git pull
git checkout issues/385
make && make reload
```
Once you have check/tested, we can merge this change into develop.

dkgroot, 2017-03-04:
One more note about this: when using a sharedline and the seconds call gets answered on an other device, the call is also displayed on your device. This would shift the focus to the second call, which would also shift the destination of the keypad buttons to that second call. 

Solutions:
- You (as enduser) can either manually shift the focus back. 
- Or you (as administator), can remove the 'Intercpt' softkey from the onhookstealable softkeyset-emtry, which would prevent the second call from being displayed (once it is remotely answered).

abradfordcms, 2017-03-09:
Thanks, Diederik. I gave this a try on my test system and everything works
as expected. Ran for a few days without issues after retries.

Installed on the PBX of the folks I had that were complaining and will see
how they make out for the rest of the week.

Andy

dkgroot, 2017-03-09:
Hi Andy,
Thanks for testing this branch.
The issues/385 branch has been merged back into the develop branch (and it does not exist any more from this moment). You can switch back to the 'develop' branch using:
```
git checkout develop
git pull
make && make install
```

commit 915ef325 (2017-03-09): Enh: Add sccp_device_indicate_suppress_phonebook_entry,
which uses either callhistorydisposition or sendcallstate(visibility_hidden) depending on devicetype
Fix: Answered elsewhere by calling d->indicate->suppress_phonebook_entry function

References #330
Fixes #385
closed 2017-03-09 by dkgroot