chan-sccp/chan-sccp #386: cfwdall = off in sccp.conf but still able to cfwdall

jamesmz, 2017-03-04:
Hi all. I'm using the latest develop commit at the time of writing (01e4148)

I want to disable call forwarding for particular handsets. This has come about after someone managed to accidentally set up cfwdall to an external number they were trying to dial, and that person got spammed with calls being diverted to them!

In sccp.conf, under [SEPMACMACMAC] for the devices in question, I have cfwdall = off but it is still possible to set up cfwdall by pressing the softkey and entering a number. 

[2017-03-04 21:24:33] NOTICE[11969]: sccp_line.c:461 sccp_line_cfwd:     -- SEPMACMACMAC: Call Forward All enabled on line 203 to number 12345

What I am expecting to happen is for the call forward to be prevented from being enabled.

Thanks in advance.

dkgroot, 2017-03-05:
Normally these softkeys are removed from the softkeyset when cfwdall=off for example.
When this setting is changed and reload is used, the softkeyset is not reloaded (todo, reload softkeyset, when one of these changes has happened).

Committed a simple fix, to recheck the setting, and generate and error message. If you reload the module, the softkeyset will be removed, and the user will not even have the option to press the button.

jamesmz, 2017-03-06:
Hi, thanks for the response. I've updated and reloaded but it doesn't seem to be working I'm afraid, as it's still allowing me to set up call forwarding when cfwdall=off. I'm testing on a 7940. James.

dkgroot, 2017-03-06:
f8a00d

dkgroot, 2017-03-06:
Hi Jamesmz,

What branch/revision are you on at this moment (should be develop/f8a00d).
```
git log |head -n1
sccp show version
```

After reloading the module, normally the softkeys should not even visible on the phone (behaviour from before the patch i committed). 

This flag is only available in the [general] and [device] sections. When set correctly, it should show
```
Can CFWDALL                                    : on
Can CFWBUSY                                    : on
Can CFWNOANSWER                                : on
```
when you do 'sccp show device SEPMACMACMAC'

jamesmz, 2017-03-06:
Hi. Looks OK now. Turns out my sccp.conf had cfwdall twice for some reason, so I ended up with both =off and =on. I've removed the duplicate entry just leaving the =off, and it is working as it should. Thanks for your help :)

closed 2017-03-06 by jamesmz

dkgroot, 2017-03-06:
:-)
Thanks for updating the issue report !

commit 33caeff0 (2017-03-05): Fix: generate tone and display message when cfwdall/cfwdbusy is not active

Note: normally these softkeys are removed from the softkeyset when cfwdall=off for example
Note: when this setting is changed and reload is used, the softkeyset is not reloaded (todo).

Fixes #386