chan-sccp/chan-sccp #389: Oneway audio during call transfer on Cisco 7912

jozzus, 2017-03-07:
When trying to make call transfer on Cisco 7912 user doesn't hear remote party.

**Reproduction Steps:**

1. Setup call from any phone(sip or sccp) to Cisco 7912
2. On Cisco 7912 press transfer soft key
3. Enter extension 
4. Wait while remote party will answer

**Expected behavior:**

We should hear user voice

**Observed behavior:**

We doesn't hear user voice or only "electronic" fragments of voices.

**Screenshots**

I set up basic test environment with Sip SoftPhone(ext.70000), Cisco 7912(ext. 70100) and Cisco 7942(ext.70200).
I made some network dumps and noticed that after pressing Transfer asterisk sends(StopMediaTransmission & CloseReceiveChannel) but it keep sending RTP media to the phone(asterisk:15636->ext70100:16384) After transfer destination(ext. 70200) answered, Asterisk opens secondary media channel (asterisk:16538->ext70100:16384). From this moment we have two media streams with came destination port 16384. I think it happened because Cisco 7912 phone uses only one udp port for media. When i did transfer on Cisco 7942 it every time uses different udp ports for media a this issue doesn't appear.

**chan-sccp version: Release: 4.3.0 develop - 7e76034M

**asterisk version: Asterisk 13.14.0 

**Addition / Optional Information:**
[-] Problem started happening recently, didn't happen in an older version of Chan-SCCP: Yes
- Problem can be reliably reproduced, doesn't happen randomly: Yes
- Platform:
  - VMWare
 - Centos Linux 3.10.0-514.10.2.el7.x86_64

jozzus, 2017-03-07:
I can't attach network dumps directly, so i uploaded it to gdrive:
https://drive.google.com/drive/folders/0B9OHN79aPdI3TlN1Rjd3TmF0bGc

dkgroot, 2017-03-07:
Hi jozzus,

would you mind running a little test with branch 'issues/389' to see if that remedies the situation:
```
git fetch --all
git pull
git checkout issues/389
make && make reload
```
This branch, changes the behaviour when putting the call on hold.

dkgroot, 2017-03-07:
Once you have tested that, would you mind helping out with another branch as well ? 
```
git clone https://github.com/chan-sccp/chan-sccp.git uniform_rtp_ack
cd uniform_rtp_ack
git checkout feature/uniform_rtp_ack
git pull origin issues/389 --squash
./configure [...your regular configure flags...]
make && make reload
```
This will combine a new branch (which tries to be more accurate about openReceive/startMedia RTP flow. It will pull in the changes from issues/389 (see previous comment) as well. Please do this in a seperate directory (as described above), which you can throw away after testing.

jozzus, 2017-03-09:
Hi, dkgroot
I've tested these two branches and seems it helped. But i found another issue which appeared in all branches with or without current fix, I opened another ticket with it #391

dkgroot, 2017-03-09:
Hi Jozzus, Thanks for reporting back. Will merge the issue/389 back into develop, to solve this issue. Will then look into the (unrelated) #391 issue.

closed 2017-03-09 by dkgroot