chan-sccp/chan-sccp #390: asterisk 10 is crashing during load of chan_sccp

hhornung, 2017-03-07:
Loaded chan_sccp.so => (Skinny Client Control Protocol (SCCP). Release: 4.3.0 TRUNK - unknown (built by 'root' on '2017-03-07 00:00:00 UTC')
)
[2017-03-08 00:30:25] ERROR[20224]: sccp_hint.c:1335 sccp_hint_notifySubscribersViaPbx: SCCP: Could not queue AST_EVENT_DEVICE_STATE_CHANGE event.

Latest commit of chan_sccp was b045f1f (developer tree)

dkgroot, 2017-03-09:
- Asterisk-10 is end of life, and has been for over 4 years (See: https://wiki.asterisk.org/wiki/display/AST/Asterisk+Versions). There has not been any development/fixes etc since it was declared end of life.
    - Maybe we should also remove compile support for these version : 1.6 / 1.8 / 10 / 12
- We are handling the error (about queueing problem) returned from asterisk, correctly
- When reporting a segfault / crash, we require a backtrace, to see what when on. See: [Reporting-Bugs](https://github.com/chan-sccp/chan-sccp/wiki/Reporting-Bugs), without a backtrace there is not a lot we can do.

hhornung, 2017-03-09:
backtrace not available. I will use an older version of chan_sccp Tx!

closed 2017-03-09 by hhornung

dkgroot, 2017-03-09:
I don't think an older version of chan-sccp is going to help (with the information we have in the report above).
I tried to recommend upgrading your asterisk to either 11 or 13. 

In any case, Please do enable 'dumpcore' (ie: 'asterisk -g' so that you have a coredump when something does happen, that way you can generate a backtrace (using gdb), and we can pick up from there. Without a backtrace, it's impossible to say way actually happened.