chan-sccp/chan-sccp #391: Voice loopback when directrtp=on

jozzus, 2017-03-09:
When option directrtp is ON called user hears itself instead of remote party.

**Reproduction Steps:**

1. Ensure directrtp  option is ON for both ends.
2. From one phone call to enother
3. Pick up phone on remote end

**Expected behavior:**

Caller must hear called party and vice versa.

**Observed behavior:**

Caller doesn't hear anything, called party hears itself.

**Screenshots**
tcpdum shows that Asterisk sends to called party  StartMediaTransmission message with called party IP and port.

[url]

**chan-sccp version: 4.3.0 develop - 01e4148M,  4.3.0 develop - b045f1fM

**asterisk version: 13.14.0

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: yes
- Problem can be reliably reproduced, doesn't happen randomly: yes

dkgroot, 2017-03-09:
Hi Jozzus,
Can you give me an indication of the previous revision you where running, when the issue was not present. We can then use 'git bisect' to figure out when the regression was introduced.

jozzus, 2017-03-09:
Sorry, I don't know from which point it started... I noticed it only when tested transfer, before it this option always was off.

dkgroot, 2017-03-09:
You can use git reflog, to see where you came from (what you did to your local repository).

jozzus, 2017-03-09:
Every time I removed source folder and start from the beginning so it hasn't any history.

dkgroot, 2017-03-09:
Ooh. This isn't really necessary (but ok). :-)
In that case, i would advice moving the old/previous folder into a backup folder, so that you have a little history.
Would you be able to guess the date of you last checkout (for your previous version) ?

jozzus, 2017-03-09:
I can only assume it's somewhere after 0e64a1e (issue #340 ), but I'm not sure..

dkgroot, 2017-03-09:
Ok thanks. Will switch my machine to directRTP tomorrow and do some git bisect starting at 0e64a1e, to figure out what changed and when.
If you like, you can git bisect a try to see if you find the 'bad' revision before i do :-)

jozzus, 2017-03-10:
I tried some oldest versions from develop and master branches but all of them works the same way..

dkgroot, 2017-03-10:
Not 100% sure what you mean by 'all of them work the same way'. Does this mean it does work, or it doesn't ?
I am going to run a couple of tests with asterisk-11 and an older version of asterisk-13 to see if direcrtp does set up the call correctly with them.

dkgroot, 2017-03-10:
Directrtp does seem to work correctly with asterisk-11. Switching to asterisk-13.12 to see if that gives the same result. From there i can work forward and see what changed.

jozzus, 2017-03-10:
> Not 100% sure what you mean by 'all of them work the same way'. Does this mean it does work, or it doesn't ?

I mean i can't find combination where it worked.
Do you reproduced this issue? Or it's something wrong with my environment?

dkgroot, 2017-03-10:
I was able to reproduce your issue with asterisk-13.14 + chan-sccp lastest 'develop' branch. 
I also tested with asterisk-11 and was not able to reproduce the issue. I am now going back to asterisk-13.12 to see if the issue occurs there as well. If not we know something in asterisk-13 changed and we need to figure out how to fix it.
So you can relax (switch back to asterisk-11 or in-directrtp for the moment), while i try to figure out what needs to be done.

dkgroot, 2017-03-14:
@jozzus 
Jitterbuffer (jbenable=yes in sccp.conf), is preventing the directrtp bridge being formed when using asterisk-13.
If you set jbenable=no, then directrtp should start working.

A patch is required, which will only attempt to enable the jitterbuffer, when in-direcrtp is used. It does not really make sense to have jitterbuffer enabled, when using direcrtp. 

It took me quite a bit of debugging to figure this one out :-) Will push a fix later this week.
Thanks for bringing the issue to my attention.

jozzus, 2017-03-15:
Oh yes i just recently turned it off, after that this issue appeared. 
I didn't think it could be connected somehow. :)

dkgroot, 2017-03-15:
I didn't realize it either, but it does make sense. For a jitterbuffer to be able to do it's work, it would have to intervene in the RTP stream. In asterisk-13 this is interpreted literally, meaning that having a jitterbuffer active, will prevent a directrtp connection.

dkgroot, 2017-03-15:
@jozzus:
Committed a fix for the issue. Can you run a couple of test with the latest develop revision.

New results should be:

jbenable | jbforce | direcrtp | result
-------- | ------- | -------- | -----------------------------
false    | false   | false    | indirect media (no jb)
true     | false   | false    | indirect media (+jb)
true     | true    | false    | indirect media (+jb)
false    | false   | true     | direct media (no jb)
true     | false   | true     | direct media (no jb)
true     | true    | true     | indirect media (+jb)

jozzus, 2017-03-22:
Hi @dkgroot!
I just yesterday had a chance to test this fix and faced with the same issue. It seems we are missing something..
I dumped current chan_sccp configuration of my environment and made test call with sccp debug rtp enabled.

dkgroot, 2017-03-22:
@jozzus 

Warning: Are you sure you want to be using 'earlyrtp = immediate'..  This requires a very specific type of dialplan ! the immediate setting is and expert setting !

Looked at your debug log, and i don't understand:
```
  == SEP001794C26D3A: (asterisk113_update_rtp_peer) stage: RTP, remote codecs capabilty: (alaw|none) (139744612827656), nat_active: 0
 SEP001794C26D3A: (asterisk113_update_rtp_peer) new remote rtp ip = '10.192.60.34:26510'
 (d->directrtp: yes && !d->nat: (Auto)Off && !remote->nat_active: yes && d->acl_allow: yes) => directmedia=yes
    -- SEP001794C26D3A: (asterisk113_update_rtp_peer) Result: 0
  == SEP44E4D944DD8A: (asterisk113_update_rtp_peer) stage: RTP, remote codecs capabilty: (alaw|none) (139744612827512), nat_active: 0
 SEP44E4D944DD8A: (asterisk113_update_rtp_peer) new remote rtp ip = '10.192.60.34:26510'
 (d->directrtp: yes && !d->nat: (Auto)Off && !remote->nat_active: yes && d->acl_allow: yes) => directmedia=yes
  == SEP44E4D944DD8A: (sccp_rtp_set_peer) Restart media transmission on channel 19
  == SEP44E4D944DD8A: (updateMediaTransmission) Stop media transmission on channel 19
    -- SEP44E4D944DD8A: Stop mediatransmission on channel 19 (KeepPortOpen: YES)
  == SEP44E4D944DD8A: (updateMediaTransmission) Start/Update media transmission on channel 19
    -- SEP44E4D944DD8A: Tell Phone to send RTP/UDP media from 10.192.60.34:26510 to 10.192.60.34:26510 (NAT: (Auto)Off)
```
Both devices seem to be pointing at the same destination address. In the end this causes the last line, where it's telling the phone to send RTP from / to one and the same address. That does not make any sense.

dkgroot, 2017-03-22:
Can you please remove the earlyrtp line in your general section (going back to progress).

jozzus, 2017-03-22:
Changed on both ends to:
Early RTP: Progress 
issue still in place...

dkgroot, 2017-03-22:
In the header you mentioned, that the issue only started happening recently. Do you know which revision you used before, that did not have the issue. (`git reflog` can help to find out which revision you used). If we know which revision did not have the problem, we can use git bisect to find the revision that caused this change in behaviour on your system.

dkgroot, 2017-03-22:
Currently directrtp seems to be broken... I am trying to figure out what happened / changed. Will take me a little while (directrtp can be a little complex to wrap your had around)

jozzus, 2017-03-22:
First time i started to test chan_sccp was:
Asterisk 13.10 + schan_sccp from MASTER branch
Then I moved to develop branch..
But all of that configurations was with:
jbenable = on
jbforce = on
I think because of it I didn't noticed this issue. 
If I'm setting these options in current version(Release: 4.3.0 develop - c49f1a2M) the issue disappearing.

dkgroot, 2017-03-22:
@jozzus
When you set jbenable + jbforce, it will prevent directrtp from being allowed and you will get indirectrtp. 
I have re-opened the issue. And will go through this step by step. I can reproduce the issues, so should be able to figure out how to fix it.

dkgroot, 2017-03-23:
@jozzus
With @voipAsteriskGuy help we got direcrtp working again in asterisk-11 (#394). Asterisk-13 is still failing but now that we have a working solution for 11, it will be a lot simpler to get asterisk-13 working again as well. So making progress.

dkgroot, 2017-03-23:
@jozzus 
Can you please checkout issues/394 branch:
```
git fetch -a
git pull
git checkout issues/394
git pull
make && make reload
```
This revision should fix the sequencial ordering required to setup rtp flow when dealing with directrtp. On asterisk-13 the flow updated the peer before updating the local side, causing trouble.

dkgroot, 2017-03-23:
Limitted Testing Done:
- asterisk-11 + directrtp
- asterisk-11 + indirectrtp
- asterisk-11 + directrtp + conference
- asterisk-13 + directrtp
- asterisk-13 + indirectrtp
- asterisk-13 + directrtp + conference

Still requiring testing:
- Other conference participant channel variants (SCCP/SIP etc).
- directrtp + NAT
- indirectrtp + NAT
- Other tests (direct/indirect/transfer/hold/resume/sharedline/pickup/nat/other channel drivers).

jozzus, 2017-03-24:
@dkgroot 
I tested this fix, it started to work different way but still not yet fixed.
I tried to present it in table form:

[image]

Debug log for each test:
[issues-394.txt](https://github.com/chan-sccp/chan-sccp/files/866948/issues-394.txt)

Also I noticed that I can't change earlyrtp  in general section. No matter what i set in conf file it shows me that "Early RTP : Immediate" when i'm checking  "sccp show globals". Even after reloading asterisk. But at the same time if I'm trying to change it in the device configuration it's working correctly. 
Note: I'm using realtime configuration for deevices and lines.

jozzus, 2017-03-24:
@dkgroot 
I tried to test this fix (Skinny Client Control Protocol (SCCP). Release: 4.3.0 issues/394 - b36c4efM (built by 'root' on '2017-03-24 06:10:57 UTC')) it seemed RTP comletely broken.

Results for the same test can be completely different and appears randomly. For example it can be same result just after full reboot whole asterisk(core restart now), but the same call but second time after reload can gives us different result. Also it depends from which phone model i tried first time. It happens either for direct and indirrect media. The results was like one way audio, correct audio in both ways and silence on both ends.

dkgroot, 2017-03-24:
@jozzus 
I spent quite a bit of work trying to fix the issue, and on my system the new version seems to work in a stable fashion. Can you set 'debug = core, channel, pbx, indicate, rtp' and show me a couple of these calls (you can send the mail directly to my mailbox, or attach them here). Please make a note in the logs about the audio state (no-way / one-way / both-ways).
Thanks for your help on this.

jozzus, 2017-03-24:
@dkgroot 
just did  fore calls without any changes and had three different results.
Dump and configs:

[log-394.txt](https://github.com/chan-sccp/chan-sccp/files/868036/log-394.txt)

[asterisk.sql.txt](https://github.com/chan-sccp/chan-sccp/files/867995/asterisk.sql.txt)

[extconfig.conf.txt](https://github.com/chan-sccp/chan-sccp/files/867997/extconfig.conf.txt)

[sccp.conf.txt](https://github.com/chan-sccp/chan-sccp/files/868014/sccp.conf.txt)

jozzus, 2017-03-24:
@dkgroot 
I'm sorry, seems it's my fall.
It was an issue with iptables. I'll re-test all of this.

dkgroot, 2017-03-24:
Are you routing these calls through NAT ?
From the logging it looks like '[general]: direcrtrtp=on' but never attempted. it's always doing AST_RTP_TRY_PARTIAL, which will never result in a fully directrtp call. If this is still the case when you have solved your ip-tables issues, we need to figure out why.

BTW: might be a good idea to power cycle all the phones you used for testing, just to make sure they are not stuff with an open rtp port.

dkgroot, 2017-03-24:
Working DirectRTP call looks something like this
```
    -- SEP381C1AB6B976: Indicate SCCP new state:CONNECTED, current channel state:OFFHOOK on call:SCCP/98031-00000002, lineInstance:1 (previous channelstate:RINGING)
    -- SEP381C1AB6B976: channel SCCP/98031-00000002 payloadType 2
    -- SEP381C1AB6B976: Ask the device to open a RTP port on channel SCCP/98031-00000002. Codec: G.711 A-law 64k, echocancel: ON
    -- SCCP/98031-00000002: rtp server instance created at [::]:12654
  == Using SCCP RTP TOS bits 184
  == Using SCCP RTP TOS bits 184 in TCLASS field.
  == Using SCCP RTP CoS mark 6
    -- SCCP/98031-00000002: (createAudioServer) updated phone Audio RTP destination to : [::ffff:10.15.15.195]:12654, family:IPv4, mapped: True
    -- SEP381C1AB6B976: Started RTP on channel SCCP/98031-00000002
    -- SEP381C1AB6B976: Open receive channel with format G.711 A-law 64k[2], payload 2, echocancel: YES, passthrupartyid: 4294967293, callid: 2
    -- SEP381C1AB6B976: video support 0 
    -- SEP381C1AB6B976: Remote Indicate state CONNECTED (30) with reason: NORMAL (0) on remote devices for channel SCCP/98031-00000002
    -- SEPE0D173E11D95: Getting the active channel on device.
    -- SEPE0D173E11D95: No active channel on device.
    -- SEP381C1AB6B976 -> SEPE0D173E11D95: indicate remote connected (lineInstance: 1, callid: 2 )
    -- SEP381C1AB6B976: Finish Indicating state CONNECTED (30) with reason: NORMAL (0) on remote device SEPE0D173E11D95 for channel SCCP/98031-00000002
    -- SEP381C1AB6B976: Finish to indicate channel state:CONNECTED on call:SCCP/98031-00000002, lineInstance:1. New channel state:CONNECTED
    -- SEP381C1AB6B976: Answered channel SCCP/98031-00000002 on line 98031
    -- SEP381C1AB6B976: Got OpenChannel ACK.  Status: 'Media Status: OK' (0), Remote RTP/UDP '10.15.15.226:16436', Type: DirectRTP, PassThruPartyId: 4294967293, CallID: 2
    -- SEP381C1AB6B976: Getting the active channel on device.
    -- SEP381C1AB6B976: Starting Phone RTP/UDP Receive (State: CONNECTED[30])
    -- SEP381C1AB6B976: Set the active channel 2 on device
    -- SCCP: (asterisk113_setPhoneRTPAddress) Update PBX to send RTP/UDP media to '10.15.15.226:16436' (new remote) (NAT: no)
    -- SEP381C1AB6B976: Tell Peer  to send RTP/UDP media from 10.15.15.195:12654 to 10.15.15.226:16436 (NoNat)
    -- SEP381C1AB6B976: Tell Phone to send RTP/UDP media from 10.15.15.226:16436 to 10.15.15.195:12654 (NAT: (Auto)Off)
    -- SEP381C1AB6B976: Using codec: G.711 A-law 64k(2), TOS 184, Silence Suppression: OFF for call with PassThruId: 4294967293 and CallID: 2
    -- SEP381C1AB6B976: send callInfo on SCCP/98031-00000002 with lineInstance: 1
    -- SCCP/98031-00000002 answered SCCP/98011-00000001
    -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel'
    -- SEP001B54CA499B: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel' (-1) condition on channel SCCP/98011-00000001 (readStat:0, writeState:0, rtp:no, default res:inband signaling (-1))
  == SEP001B54CA499B: (pbx_indicate) finish: send indication 'inband signaling' (-1)
    -- SCCP: sccp_pbx_answer
    -- SEP001B54CA499B: sccp_pbx_answer checking parent channel
    -- SEP001B54CA499B: (sccp_pbx_answer) Outgoing call SCCP/98011-00000001 has been answered by remote party
    -- SEP001B54CA499B: recalculateReadformat
    -- preference: 2(alaw/64k)
    -- found first firstJointCapability 2(alaw/64k)
    -- found bestCodec as joint capability with remote peer 2(alaw/64k)
    -- SCCP/98011-00000001: (recalculateReadformat) 
	capabilities: [slin16, ulaw/64k, alaw/64k, g729b, g729ab, g729, g729a]
	preferences: [alaw/64k, g729, g729a, g729b, g729ab]
	remote caps: [alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab]
	READ:[alaw/64k]
    -- SEP001B54CA499B: recalculateWriteformat
    -- preference: 2(alaw/64k)
    -- found first firstJointCapability 2(alaw/64k)
    -- found bestCodec as joint capability with remote peer 2(alaw/64k)
    -- SCCP/98011-00000001: (recalculateWriteformat) 
	capabilities: [slin16, ulaw/64k, alaw/64k, g729b, g729ab, g729, g729a]
	preferences: [alaw/64k, g729, g729a, g729b, g729ab]
	remote caps: [alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab]
	Write:[alaw/64k]
    -- SEP001B54CA499B: Indicate SCCP new state:PROCEED, current channel state:RINGOUT on call:SCCP/98011-00000001, lineInstance:1 (previous channelstate:RINGOUT_ALERTING)
    -- SEP001B54CA499B: Finish to indicate channel state:PROCEED on call:SCCP/98011-00000001, lineInstance:1. New channel state:PROCEED
    -- SEP001B54CA499B: Indicate SCCP new state:CONNECTED, current channel state:PROCEED on call:SCCP/98011-00000001, lineInstance:1 (previous channelstate:RINGOUT)
    -- SEP001B54CA499B: channel SCCP/98011-00000001 payloadType 2
    -- SEP001B54CA499B: Ask the device to open a RTP port on channel SCCP/98011-00000001. Codec: G.711 A-law 64k, echocancel: ON
    -- SCCP/98011-00000001: rtp server instance created at [::]:16788
  == Using SCCP RTP TOS bits 184
  == Using SCCP RTP TOS bits 184 in TCLASS field.
  == Using SCCP RTP CoS mark 6
    -- SCCP/98011-00000001: (createAudioServer) updated phone Audio RTP destination to : [::ffff:10.15.15.195]:16788, family:IPv4, mapped: True
    -- SEP001B54CA499B: Started RTP on channel SCCP/98011-00000001
    -- SEP001B54CA499B: Open receive channel with format G.711 A-law 64k[2], payload 2, echocancel: YES, passthrupartyid: 4294967294, callid: 1
    -- SEP001B54CA499B: video support 0 
[... cut ...]
 SEP381C1AB6B976: (asterisk113_get_rtp_info) Channel SCCP/98031-00000002 Returning res: direct-rtp
 SEP381C1AB6B976: (asterisk113_get_vrtp_info) Asterisk requested RTP peer for channel SCCP/98031-00000002
       > 0x7f6ff8016430 -- Probation passed - setting RTP source address to 10.15.15.226:16436

```

Also i don't understand why changing [general] earlyrtp would not work. I cannot reproduce this issue on my system. Maybe you should set 'debug = core, config' in your sccp.conf and do a reload, it will show you exactly how the file get's parsed.

dkgroot, 2017-03-24:
@jozzus 
Please do check that permit/deny settings between the two devices include one another. Using permit/deny you can specify in which segments of your network direcrtrtp is allowed. If the two mask each other out, then directrtp will not even be attempted.

dkgroot, 2017-03-24:
btw: you are correct about earlyrtp. When you remark it out in the general section, it does get the default 'progress' value. Something is wrong in the config parser, i am trying to figure out what.

jozzus, 2017-03-27:
Hi, @dkgroot 
I tested wersion with last commit(Skinny Client Control Protocol (SCCP). Release: 4.3.0 issues/394 - 2e31356M (built by 'root' on '2017-03-27 04:40:27 UTC')) and found couple issues:

1. Delayed media with indirect RTP setup. Media connection occurs only when voice start to go from caller side(7912). Inside the dump  I marked each steps during voice setup:
[indirect_delayed.txt](https://github.com/chan-sccp/chan-sccp/files/871716/indirect_delayed.txt)

2. Ona way media with direct RTP setup. Media goes only from caller to called party.
[direct_oneway.txt](https://github.com/chan-sccp/chan-sccp/files/871726/direct_oneway.txt)

Still testing, will update if i found anything.

dkgroot, 2017-03-27:
@jozzus 
Thanks for testing !

1. When both devices have sent 'Got OpenChannel ACK', they should be able to receive audio. We do have to wait until asterisk actually bridges the call. 
From the moment where both devices have sent 'Got StartMediaTranmission ACK' you should be able to speak. This should happen after the bridge has formed for the answering party.
Both of this seems to be the case.

You can force the opening of the receiving rtp channel a little earlier, by setting earlyrtp=dial or earlyrtp=ringout.

I don't understand why your system is handling the flow so differently though.

2. 
Before Direct Bridging
    -- SEP44E4D944DD8A: Tell Peer  to send RTP/UDP media from 10.194.3.111:11908 to 10.192.60.34:24492 (NoNat)
    -- SEP44E4D944DD8A: Tell Phone to send RTP/UDP media from 10.192.60.34:24492 to 10.194.3.111:11908 (NAT: Auto)

After started bridging (looks ok)
    -- SEP001794C26D3A: Tell Peer  to send RTP/UDP media from 10.192.60.34:24492 to 10.192.61.17:16384 (NoNat)
    -- SEP001794C26D3A: Tell Phone to send RTP/UDP media from 10.192.61.17:16384 to 10.192.60.34:24492 (NAT: Auto)

The remote end for D8A, does not get updated to 10.192.61.17, which should have happened: We need to find out why this is produced instead.
"SEP44E4D944DD8A: (asterisk113_update_rtp_peer) failed to get remote ip-address"

jozzus, 2017-03-27:
@dkgroot 
I found the cause of one way audio with direct media. I gathered network dump and found that directRTP only occurs from caller to called party at the same way reverse flow (from called to calling) still going through server via indirect way.
So, in my environment it didn't work because i have firewall between phone network and test server network, which allow only connections initiated from phone network. In our situation RTP from caller side never passed through firewall(it went dirrectly to called party). When I added the role which allow connections from server to phone network ona way issue also disappeared.

But nevertheless i think this behavior with assymetrical RTP should be fixed.

btw first issue was also caused by firewall.

jozzus, 2017-03-29:
@dkgroot 
I noticed that this strange behavior(in-direct received RTP) only happens with Cisco 7912 model. I even switched cisco 7942 to sip firmware, test results was the same. Can it happens due to differences in handling for this model and others models?

jozzus, 2017-03-29:
Asymmetrical RTP setup happens only when call initiated from Cisco 7912.

Also I found that  issue #389 still happening with direct RTP when incoming peer is SIP device.

dkgroot, 2017-03-29:
@jozzus 
The 7912 is limitted to one RTP stream at a time, which might be the cause for these issue. We do try to make sure we close the RTP stream before opening a new one. I don't own a 7912 which makes it a little harder for me to debug this.

RTP and Codecs:
Getting directrtp and codec negotiation right is harder than it might look at first. With sccp phones we can not change codecs on the fly, so codec negotiation has to happen as late as possible. And the sccp phones can be a little slow when opening a new RTP stream, this has to happen as early as possible. This puts us in a bit of a tough spot. Especially when performing a transfer where we are handling three call legs (bridge channels).

Can you send me (direct mail please) the tcpdump you captured, so i can have a look.

jozzus, 2017-03-29:
@dkgroot sent via email.

jozzus, 2017-03-30:
@dkgroot I fount this in my logs:

sccp_actions.c:3549 handle_startmediatransmission_ack: SEP44E4D944DD8A: Channel with passthrupartyid 4294967258 / callid 37 / callid1 37 not found, please report this to developer

Can it be interesting?

dkgroot, 2017-03-30:
@jozzus 
Yes that's very interesting. Can you reproduce that message ? If so, can you please set 'sccp debug core, channel, pbx, action, message, codec, rtp'. That should send the 'message' sent by the phone, for which we don't seem to have a call.
It might just be that the call was already hungup, and the startmediatransmissionack came back a little late. but it might also show us the cause for the one way audio issue you are having.

BTW: i did not get a mail from you (and i did check my spam folder)

jozzus, 2017-03-31:
@dkgroot 
I can consistently reproduce this warning.
The exact conditions is:
1. Calling party - sip trunk with direct-RTP
2. Called party - sccp Cisco 7942 with direct-RTP(i can't reproduce it with Cisco 7912)
3. Calling party(sip trunk) should end the call first.
I collected debug log with this warning:
[warning.txt](https://github.com/chan-sccp/chan-sccp/files/884548/warning.txt)

I uploaded dumps to gdrive: https://drive.google.com/drive/folders/0B_TOmjnGTOoCSEk4b3dzenpER2M?usp=sharing

I also added dump of call with warning message(sip-to-7942)

jozzus, 2017-05-04:
Hi, @dkgroot 

Have you a chance to investigate these bugs?

dkgroot, 2017-09-01:
Hi Jozzus,

Thanks for the report. I only just started to get back to working on the chan-sccp project. It will take me a while to order the issue reports by priority and work on them one by one.

Sorry for the inconvenience.

Regards,

Diederik

dkgroot, 2017-09-01:
References: #405

dkgroot, 2017-09-24:
@jozzus 
Would you mind checking the 'issues/394' branch

Related Issues:
#394
#391
#405

Once this branch (and Pull Request #416) has been check it can be merged back into the develop tree.

jozzus, 2017-09-25:
Hi @dkgroot,
Grate news, i'll test it soon.

jozzus, 2017-09-25:
[Sccp.zip](https://github.com/chan-sccp/chan-sccp/files/1328607/Sccp.zip)
@dkgroot,

I found an issue with audio when calling from sip device.
My current setup:

1. Asterisk 13.17.1
2. Skinny Client Control Protocol (SCCP). Release: 4.3.0 issues/394_1 - 6269025M
3. SIP phone ip:10.192.60.20 ext.70200
4. SCCP phone Cisco 7942 ip:10.192.60.34 ext.70790

Call from sip(70200) -> sccp(70790) = no audio in both way
Call from sccp(70790) -> sip(70200) = Ok

debug log and tcp dump attached.

dkgroot, 2017-09-25:
@jozzus 
Thanks for testing. Will have a look at your resuls and figure out what went wrong.
Btw: next time set 'sccp debug core, channel, rtp, codec, pbx' instead. That way the debug output is a little easier to interpret.

jozzus, 2017-09-25:
@dkgroot 
Debug log with your debug options.
[debug.txt](https://github.com/chan-sccp/chan-sccp/files/1329877/debug.txt)

dkgroot, 2017-09-25:
Committed a patch to prevent codec issues. To prevent
```
Got  RTP packet from    10.192.60.20:22138 (type 08, seq 001307, ts 2049463489, len 000160)
[2017-09-25 17:45:44] DEBUG[30010][C-00000010]: res_rtp_asterisk.c:4025 ast_rtp_write: Ooh, format changed from none to alaw
Sent RTP packet to      10.192.60.34:22154 (type 08, seq 028495, ts 2049463488, len 000160)
```
Can you recheck with the latest version, to see if this helped resolve your issue ?

Have tested several calls with SIP->SCCP and SCCP->SIP (asterisk-13).

If video is enabled on both directrtp bridging seems to succeed. If video is disabled or not possible on the SCCP phone, direcrtp is not performed but there is two way indirect audio. (This needs a little more research), you did not seem to run into this issue.

To be honest I am already happy if SCCP->SCCP directrtp works correctly (including hold/resume/transfer). And if SIP->SCCP at least results in two-way audio, even if we are not achieving remote briding (directrtp) in all circumstances (agreed ?).

jozzus, 2017-10-11:
Hi @dkgroot!

I've tested this new version:
Skinny Client Control Protocol (SCCP). Release: 4.3.0 issues/394_1 - c438b3eM (built by 'root' on '2017-09-25 17:13:46 UTC')

And found several issues:
1. One way audio between 7845 and other SCCP devices(7912, 7942) with direct media.
 2. No audio, when call setup from SIP(direct media) to SCCP(direct media).
3. One way audio when call setup from SCCP(direct media) to SIP(direct media), BUT when i enabled debug options the issue disappeared :)

[7845-7912.txt](https://github.com/chan-sccp/chan-sccp/files/1374521/7845-7912.txt)
[7845-sip(debug on-off).txt](https://github.com/chan-sccp/chan-sccp/files/1374522/7845-sip.debug.on-off.txt)
[SIP-7845.txt](https://github.com/chan-sccp/chan-sccp/files/1374523/SIP-7845.txt)

dkgroot, 2017-10-12:
@jozzus 
Thanks for reporting back. Will look at your debug files tomorrow.
Still a little strange why you are getting different results to me. Must be that we have different network-seyups.

jozzus, 2017-10-13:
@dkgroot 
I'm using test setup without NAT, internet, etc.
Now it's looks like this:
All my phones locates in same network(10.192.60.0/23)
SIP phone(Cisco 7942) - 10.192.60.34
Cisco 7912G (SEP001794C26D3A) 

model | protocol | s/w ver | mac | IP | exten
-- | -- | -- | -- | -- | --
Cisco7942 | SIP | CP7942G/8.5.3 | - | 10.192.60.34 | 70200
Cisco 7912G-A | SCCP | CP7912080004SCCP080108A | SEP001794C26D3A |10.192.60.25 | 70000
Cisco 7942G | SCCP | SCCP42.9-4-2SR3-1S | SEPF07F06B8DBE1 | 10.192.60.22 | 70400
Cisco 7845 | SCCP | SCCP894x.9-4-2SR3-1 | SEP580A20FBA215 | 10.192.60.23 | 70450

Working configuration of all the phones is in attached file phone_conf.txt

Asterisk server located in seporate network and has IP 10.194.3.111. 
Asterisk version: 13.17.2 
chan_sccp version: 4.3.0 issues/394_1 - c438b3eM

I have Cisco ASA firewall between Phone network and server but i opened everything between them(disabled any filtering)

After reloading server an all the phones i remake tests, to be more clear I'll post my results one by one.

Call from  Cisco 7845(70450) to Cisco 7912G(70000) - 7845to7912.txt
Results: I can hear voice coming to 7912 but no voice coming to 7845

From the log we can see that RTP from 7912 to 7845 is still going through the server but reverse stream directly. So, in my opinion cisco 7845 expects o receive RTP from 7912 directly and discarding RTP received from the server.

The same result with any combination of phone models where Cisco 7845 involved.
[7845to7912.txt](https://github.com/chan-sccp/chan-sccp/files/1381658/7845to7912.txt)
[phone_conf.txt](https://github.com/chan-sccp/chan-sccp/files/1381659/phone_conf.txt)

jozzus, 2017-10-13:
Tests between SIP and SCCP:
For sip phone changed option DirectMedia  : Yes
Everything reloaded, config for phones - conf(sip-direct).txt

Call from SIP to SCCP Cisco7912  - no issues
Call from SCCP Cisco 7912 to SIP - no issues but RTP is going through the server.
debug for both calls - sip-7912.txt
[conf(sip-direct).txt](https://github.com/chan-sccp/chan-sccp/files/1381880/conf.sip-direct.txt)
[sip-7912.txt](https://github.com/chan-sccp/chan-sccp/files/1381881/sip-7912.txt)

jozzus, 2017-10-13:
Call from sip to 7942 - no voice in both ways
call from 7942 to sip - no issues vith voice but RTP is going through server
[sip-7942.txt](https://github.com/chan-sccp/chan-sccp/files/1381919/sip-7942.txt)

[19 more replies]

commit c49f1a2f (2017-03-15): Fix: jitterbuffer_enable preventing direcrtp with > asterisk-12

Cause: jitterbuffer creates a channel-framehook, and all audiohooks / framehooks prevend directrtp to be established.

New results:

jbenable | jbforce | direcrtp | result
-------- | ------- | -------- | -----------------------------
false    | false   | false    | indirect media (no jb)
true     | false   | false    | indirect media (+jb)
true     | true    | false    | indirect media (+jb)
false    | false   | true     | direct media (no jb)
true     | false   | true     | direct media (no jb)
true     | true    | true     | in [... cut]