chan-sccp/chan-sccp #392: Commit 5e5f65a caused 7921 to lose G.722 support

zipiju, 2017-03-11:
I just pulled latest develop branch and it looks like commit 5e5f65a caused C7921 to lose G.722 capability (or any other codecs except G.711a/μ). Other phone types are behaving as before.

**Reproduction Steps:**

1. Compile rev. 5e5f65a or any beyond
2. Install and restart Asterisk/reload chan_sccp.so

**Expected behavior:**

C7921 has G.722 capability. With any rev. before 5e5f65a calls are working normally.

**Observed behavior:**

Phone lose G.722 codec support. With current config (and with disallow=all; allow=g722) calls won't go through and Asterisk is complaining about codec capabilities:

> Cannot determine best translation path since one capability supports no formats.
> No path to translate from SCCP/207-00000004 to SCCP/203-00000003.

Output of show sccp device SEP... with rev. 5e5f65a and beyond:
```
Protocol Version   : Supported '16', In Use '16'
Protocol In Use    : SCCP Version 16
Device Features    : 0x85600010,10000101011000000000000000010000
...
Config Phone Type  : 7921
Skinny Phone Type  : Cisco 7921(365)
Capabilities       : [ulaw/64k, alaw/64k]
Codecs preference  : [g722/64k, g722/56k, g722/48k]
```

Output of show sccp device SEP... before rev. 5e5f65a:
```
Protocol Version   : Supported '16', In Use '16'
Protocol In Use    : SCCP Version 16
Device Features    : 0x85600010,10000101011000000000000000010000
...
Config Phone Type  : 7921
Skinny Phone Type  : Cisco 7921(365)
Capabilities       : [ulaw/64k, alaw/64k, rfc2833, g729, g729a, g729b, g729ab, slin16, g722/64k, g722/56k, g722/48k, ilbc]
Codecs preference  : [g722/64k, g722/56k, g722/48k]
```

**Addition / Optional Information:**
- chan-sccp version: 4.3.0 develop - 5e5f65a
- Asterisk version: 13.13.1
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: Yes
- Problem can be reliably reproduced, doesn't happen randomly: Yes
- Log Files with rev. 5e5f65a: https://gist.github.com/zipiju/162aaf12c46c74bbf0c21ee17fccee48

dkgroot, 2017-03-13:
@zipiju thank you for reporting this issue and being so helpfull, to pinpoint the actual commit that caused the regression. The commit was meant to correct potential issues, and from the diff it's not completely clear how this could cause the side effect you reported.

Can you set 'sccp debug core, device, action, message' and restart one of these devices so that we can collect some of the debug. Normally device capabilities will be sent shortly after the device has registered.

dkgroot, 2017-03-13:
@zipiju I just pushed a branch to help fix/debug this issue called 'issues/392'. Can you check it out and see if it helps correct the issue, or at least show where the error is caused.
```
git fetch --all
git pull
git checkout issues/392
make && make reload
sccp debug core, device, codec
```

Thanks !

zipiju, 2017-03-13:
Hello Diederik, thanks, this fix works.
Calls are working okay, other phone types are also okay with this patch.
Didn't do debugs as it is working. If you need them please let me know and I'll dump them.
Thank you!
```
serenity*CLI> sccp show version
Skinny Client Control Protocol (SCCP). Release: 4.3.0 issues/392 - 17c9013M (built by 'root' on '2017-03-13 12:30:58 UTC')

serenity*CLI> sccp show device SEP002155xxxxxx

--- SCCP channel driver device settings ----------------------------------------------------------------------------------
MAC-Address                                    : SEP002155xxxxxx
Protocol Version                               : Supported '16', In Use '16'
Protocol In Use                                : SCCP Version 16
Device Features                                : 0x85600010,10000101011000000000000000010000
...
Config Phone Type                              : 7921
Skinny Phone Type                              : Cisco 7921(365)
Capabilities                                   : [ulaw/64k, alaw/64k, g729, g729a, g729b, g729ab, slin16, g722/64k, g722/56k, g722/48k, ilbc]
Codecs preference                              : [g722/64k, g722/56k, g722/48k]
```

dkgroot, 2017-03-13:
Merged the issues/392 branch into develop. Issues closed.
Thanks for reporting the issues and checking the patch !

closed 2017-03-13 by dkgroot