chan-sccp/chan-sccp #393: How to display dial out person name in Cisco 7945g with Chan-sccp-b?

shoaibraza, 2017-03-21:
I have successfully implemented chan-sccp-b and everything works perfectly. Corporate directory is added to Cisco 7945g using directoryURL. When I select a person to dial out from Corporate directory, it dials out by number and doesn't display name. Phone directory shows name and number but doesn't display name after pressing dial softkey.

<directoryURL>http://192.168.1.10/cisco_menu/directory/menu.php</directoryURL>

menu.php generates below XML

```
<CiscoIPPhoneDirectory>
<Title>Asterisk CID Directory</Title>
<Prompt>Please Select</Prompt>
<DirectoryEntry>
<Telephone>1234567</Telephone>
<Name>Test Caller</Name>
<URL>Dial:1234567</URL>
</DirectoryEntry>
</CiscoIPPhoneDirectory>
```

What's worng done by me? help required.

dkgroot, 2017-03-21:
The only entry send from the phone to chan-sccp is the "Dial:1234567" string. It does not send the Name to chan-sccp/asterisk. They only way for you to show the name would be to make sure asterisk dialplan can resolve the destination and set's to correct name and number for the called party. For example:
```
exten => _1234567,1,Noop(Calling:Test Caller at 1234567)
 same => n,Set(CHANNEL(calledparty)="Test Caller <1234567>")
 same => n,Dial(SIP/1234567@provider, 20)
 same => n,Hangup()
```
If 1234567 was an internal destination (like SIP/SCCP etc), you would have only have to set the correct cid_num and cid_name in sip.conf / sccp.conf. When these extensions are dialed, the name and number will be display automatically. So the method above would only be necessary when calling number which are not local to your PBX.

You could write a litte AGI script to lookup the name / number from the same database behind your php script, to make this more flexible. There are also a number of scripts that will check incoming/outgoing number against internet sources (like phonebook / white pages / yellow pages etc).

Please close this 'issue' report if you have no further questions on this topic. BTW: these sort of questions are better asked on the mailinglist :-)

shoaibraza, 2017-03-21:
Thanks dkgroot for your prompt response.

No issue while dialing to internal extensions.

The issue is when to dial external number i.e. 91234567 where 9 is prefix for outgoing dial plan and 1234567 is a number. I have stored 91234567 with Test User in phonebook and can view phonebook in 7945g. As I press dial softkey, it display number only and doesn't show name i.e. Test User in this case.

Appreciate if you can share some information to solve the issue as I'm stuck with this problem for more than two months.

dkgroot, 2017-03-21:
The phones do not check the phonebook to resolve names. The names are only provided by the PBX (if known). You will have to write an agi script to make this happen. The agi script would be able to lookup the number in the same database (that was used to fill the phonebook). The ability script can then set the channel calledparty and it will all work. You can use the php language to write your agi script. You will have to do a little bit of (research and) work to make this work. You can start with the example I gave above, and replace the set(channel()) line with the agi script.

As this is a PBX issue and not directly related to chan-sccp I am closing this issue for now.

closed 2017-03-21 by dkgroot

dkgroot, 2017-03-21:
Examples:
- [PHP lookup mysql callerid](https://www.voip-info.org/wiki/view/Asterisk+simple+php+lookup+mysql+database+to+set+callerid+name)
- [php lookup pgsql callerid](https://www.voip-info.org/wiki/view/Asterisk+simple+php+lookup+up+callerid+name+from+Horde+Turba)
- [very complete lookup script](https://stuff.purdon.ca/?page_id=267)
- [how to manipulate callerid / called party info](https://wiki.asterisk.org/wiki/display/AST/Manipulating+Party+ID+Information)
- [agi explanation](https://www.voip-info.org/wiki-Asterisk+AGI)

dkgroot, 2017-03-22:
@shoaibraza 
Examples:
- [PHP lookup mysql callerid](https://www.voip-info.org/wiki/view/Asterisk+simple+php+lookup+mysql+database+to+set+callerid+name)
- [php lookup pgsql callerid](https://www.voip-info.org/wiki/view/Asterisk+simple+php+lookup+up+callerid+name+from+Horde+Turba)
- [very complete lookup script](https://stuff.purdon.ca/?page_id=267)
- [how to manipulate callerid / called party info](https://wiki.asterisk.org/wiki/display/AST/Manipulating+Party+ID+Information)
- [agi explanation](https://www.voip-info.org/wiki-Asterisk+AGI)