chan-sccp/chan-sccp #394: hold/resume call using directrtp

dkgroot, 2017-03-22:
Issue reported by Salvador
Call is establised correctly and direcrtp is working between both endpoints
Call can be put on hold and Music on hold is heard by remote party
After resume:
```
  == SEP001E7A262593: (asterisk111_update_rtp_peer) stage: RTP, remote codecs capabilty: (ulaw|alaw) (295082160), nat_active: 0
 SEP001E7A262593: (asterisk111_update_rtp_peer) new remote rtp ip = ''
 (d->directrtp: yes && !d->nat: (Auto)Off && !remote->nat_active: yes && d->acl_allow: yes) => directmedia=yes
  == SEP001E7A262593: ( sccp_rtp_set_peer ) remote information are invalid, don't change anything
    -- SEP001E7A262593: (asterisk111_update_rtp_peer) Result: 0
    -- SEP001E7A262593: (pbx_indicate) start indicate 'AST_CONTROL_UNHOLD: Indicate call left hold' (17) condition on channel SCCP/53558-00000004 (readStat:2, writeState:2, rtp:yes, default res:inband signaling (-1))
    -- Stopped music on hold on SCCP/53558-00000004
  == SEP001E7A262593: (pbx_indicate) finish: send indication 'outofband signaling' (0)
  == SEP58971ECC3279: (asterisk111_update_rtp_peer) stage: RTP, remote codecs capabilty: (ulaw|alaw) (295041312), nat_active: 0
 SEP58971ECC3279: (asterisk111_update_rtp_peer) new remote rtp ip = '172.22.200.56:32494'
 (d->directrtp: yes && !d->nat: (Auto)Off && !remote->nat_active: yes && d->acl_allow: yes) => directmedia=yes
  == SEP58971ECC3279: (sccp_rtp_set_peer) remote information is equal to the current info, ignore change
    -- SEP58971ECC3279: (asterisk111_update_rtp_peer) Result: 0
  == SEP001E7A262593: (asterisk111_update_rtp_peer) stage: RTP, remote codecs capabilty: (ulaw|alaw) (295082160), nat_active: 0
 SEP001E7A262593: (asterisk111_update_rtp_peer) RTP not ready
    -- SEP001E7A262593: (asterisk111_update_rtp_peer) Result: 0
```
The audio RTP channel is not re-established.

dkgroot, 2017-03-23:
AST_CONTROL_UNHOLD from sccp_channel_resume to openReceiveChannelAck to fix the serial ordering

(required for asterisk-13 which runs 'update_rtp_peer' in parallel / reverse order (compared to asterisk-11), causing a race)

Changes required for >asterisk-12:

Local
1. OpenReceive
2. OpenReceiveAck
3. setRtpPhone to newly acquired ip-address
4. startMediaTransmission
5. Indicate AST_CONTROL_UNHOLD -> kicks peer into action

Peer
6. Peer will re-bridge
7. Which will then use the newly acquired ip-address of it's remote side (our local side)

This used to work correction, when we only stopped media transmission, but did not tear down the openreceive channel. This did however
cause issues on older phone, which only support 1 active RTP channel.

dkgroot, 2017-03-23:
Limitted Testing Done:
- asterisk-11 + directrtp
- asterisk-11 + indirectrtp
- asterisk-11 + directrtp + conference
- asterisk-13 + directrtp
- asterisk-13 + indirectrtp
- asterisk-13 + directrtp + conference

Still requiring testing:
- Other conference participant channel variants (SCCP/SIP etc).
- directrtp + NAT
- indirectrtp + NAT
- Other tests (direct/indirect/transfer/hold/resume/sharedline/pickup/nat/other channel drivers).

dkgroot, 2017-09-25:
Please set
```
sccp debug core, channel, rtp, codec, pbx
core set debug 3
core set verbose 5
rtp set debug on
```
during testing.

dkgroot, 2017-10-15:
I Think i fixed directrtp (again) :-)

Pushed another set of commites (cd1eac7)
Can you give it another try
(with "sccp debug core, channel, pbx, indicate, rtp")

I made one major change in Signalling: Which does cause somewhat altered behaviour.

When the Phone Receiving the incoming call starts ringing, it will send PROGRESS() to asterisk. So that the other side (the Dialing Party), can start setting up RTP. That way we have the ip-address:port combination a little earlier on, before bridging starts.

We do need to test this rigorously, to make sure it does not have negative effects (like with NAT / in-directrtp / HOLD / RESUME / Transfer / etc).

dkgroot, 2017-10-19:
I finaly figure out the cause for the directrtp issues. 

Create a new branch called 'issues/394_2'
Please check if this fixes the issue you reported.
Previous 'issues/394_1' will be dropped/removed shortly.

Ran quite a number of tests and it seems to work. But would love to hear back from you, to make sure.

nickvsh, 2019-11-06:
Hi.
Asterisk **certified/16.3** 
Skinny Client Control Protocol (SCCP). Release: 4.3.2 (HEAD - **558d2b6**M)

Hold/Resume functions are OK, ONLY IF use it on ONE device during the call. 
If press Hold on both phones simultaneously during the call/one after another (hold on 7912<-> then hold on 7960)  - Resume "resumes" RTP from the last phone only after second attempt.

dkgroot, 2019-11-06:
@nickvsh
I have not been spending a lot of time lately making sure directrtp is still working correctly (sorry). It looked like there was little to no interest, and there were other more pressing matters.

And this branch talked about above, was never tested/checked by anyone, so did not get committed/merged. I would have to spend some time to revive this branch. Once it is actual again then i would have to ask you to help test it and we can find out why hold/resume is failing in this situation.

nickvsh, 2019-11-11:
Recently I make a test accoring to [this](https://github.com/chan-sccp/chan-sccp/issues/405#issuecomment-551203324) 
```
Connected to Asterisk certified/16.3 currently running on hq (pid = 11068)
CLI> sccp show version 
Skinny Client Control Protocol (SCCP). Release: 4.3.2 issues/394_2 - 558d2b6M (built by 'voip' on '2019-11-08 15:06:18 UTC')

```
**Call between Cisco 7912 devices - is almost OK. Phone's firmware - CP7912080004SCCP080108A**

Hold on 1-st device, then hold on 2-nd -> Then Resume on each = the voice is OK in both directions.
But Moh while 1 device is on Hold is only in one direction. I assume it depends on who first put call on Hold.

**Call between Cisco 7960 devices - have some problems here. Phone's firmware - P0030801SR02**

In general, if normally using Hold/Resume - only one device putting another on Hold/Resumes - Voice is OK.
But if put one device on hold, then on the second device press Hold -> **there is a problem in restoring 2-directional RTP between devices.**
It **restores only after 2-nd attempt Holding/Resuming** on one or each phones.

Here is detailed test call with comments. I hope it helps.

```
Device with ext. 7317 calls 5720 extension.

[Nov 10 14:51:18]   == Using SCCP RTP TOS bits 184
[Nov 10 14:51:18]   == Using SCCP RTP CoS mark 6
[Nov 10 14:51:18]     -- Executing [5720@pzu_international:1] Gosub("SCCP/7317-00000143", "sub-call-dn-SCCP,start,1(5720)") in new stack
[Nov 10 14:51:18]     -- Executing [start@sub-call-dn-SCCP:1] NoOp("SCCP/7317-00000143", "") in new stack
[Nov 10 14:51:18]     -- Executing [start@sub-call-dn-SCCP:2] Verbose("SCCP/7317-00000143", "Call to cisco7912/DN:start FROM: 7317/NAME-1") in new stack
[Nov 10 14:51:18] Call to cisco7912/DN:start FROM: 7317/NAME-1
[Nov 10 14:51:18]     -- Executing [start@sub-call-dn-SCCP:3] Set("SCCP/7317-00000143", "CHANNEL(language)=ru") in new stack
[Nov 10 14:51:18]     -- Executing [start@sub-call-dn-SCCP:4] ChanIsAvail("SCCP/7317-00000143", "SCCP/5720") in new stack
[Nov 10 14:51:18]     -- Executing [start@sub-call-dn-SCCP:5] NoOp("SCCP/7317-00000143", "flag is ") in new stack
[Nov 10 14:51:18]     -- Executing [start@sub-call-dn-SCCP:6] GotoIf("SCCP/7317-00000143", " 0?sticky_call:") in new stack
[Nov 10 14:51:18]     -- Executing [start@sub-call-dn-SCCP:7] GotoIf("SCCP/7317-00000143", " 0?not_available:") in new stack
[Nov 10 14:51:18]     -- Executing [start@sub-call-dn-SCCP:8] Dial("SCCP/7317-00000143", "SCCP/5720,30,r(ring)") in new stack
[Nov 10 14:51:18]     -- Called SCCP/5720
[Nov 10 14:51:18]     -- SCCP/5720-00000145 is ringing
[Nov 10 14:51:18]        > 0x7fe59c02fc70 -- Strict RTP learning after remote address set to: 172.16.13.11:23806
[Nov 10 14:51:18]        > 0x7fe59c02fc70 -- Strict RTP switching to RTP target address 172.16.13.11:23806 as source
[Nov 10 14:51:20]   == Using SCCP RTP TOS bits 184
[Nov 10 14:51:20]   == Using SCCP RTP CoS mark 6
[Nov 10 14:51:20]     -- SCCP/5720-00000145 is making progress passing it to SCCP/7317-00000143
[Nov 10 14:51:20]        > 0x7fe5a4035500 -- Strict RTP learning after remote address set to: 172.16.13.14:17142
[Nov 10 14:51:20]     -- SCCP/5720-00000145 answered SCCP/7317-00000143
[Nov 10 14:51:20]     -- Channel SCCP/5720-00000145 joined 'simple_bridge' basic-bridge <34687c15-eaeb-4877-a71b-57156b2445ed>
[Nov 10 14:51:20]     -- Channel SCCP/7317-00000143 joined 'simple_bridge' basic-bridge <34687c15-eaeb-4877-a71b-57156b2445ed>
[Nov 10 14:51:20]        > Bridge 34687c15-eaeb-4877-a71b-57156b2445ed: switching from simple_bridge technology to native_rtp
[Nov 10 14:51:20] NOTICE[10452][C-000000a1]: sccp_rtp.c:242 sccp_rtp_set_peer: SEP003094C39A23: ( sccp_rtp_set_peer ) Set new remote address to 172.16.13.14:17142
[Nov 10 14:51:20] NOTICE[10452][C-000000a1]: sccp_rtp.c:242 sccp_rtp_set_peer: SEP003094C3CF05: ( sccp_rtp_set_peer ) Set new remote address to 172.16.13.11:23806
[Nov 10 14:51:20]        > Remotely bridged 'SCCP/7317-00000143' and 'SCCP/5720-00000145' - media will flow directly between them

Press HOLD on "7317" device
[Nov 10 14:51:45]     -- Started music on hold, class 'default', on channel 'SCCP/5720-00000145'

Press RESUME on "7317" device
 -> Voice is OK

[Nov 10 14:52:02]        > 0x7fe59c02fc70 -- Strict RTP learning after remote address set to: 172.16.13.11:28218
[Nov 10 14:52:02] NOTICE[10454][C-000000a1]: sccp_rtp.c:242 sccp_rtp_set_peer: SEP003094C3CF05: ( sccp_rtp_set_peer ) Set new remote address to 172.16.13.11:28218
[Nov 10 14:52:02]     -- Stopped music on hold on SCCP/5720-00000145

Press HOLD on "5720" device

[Nov 10 14:52:53]     -- Started music on hold, class 'default', on channel 'SCCP/7317-00000143'

Press RESUME on "5720" device

[Nov 10 14:53:04]        > 0x7fe5a4035500 -- Strict RTP learning after remote address set to: 172.16.13.14:27564
[Nov 10 14:53:04] NOTICE[10452][C-000000a1]: sccp_rtp.c:242 sccp_rtp_set_peer: SEP003094C39A23: ( sccp_rtp_set_peer ) Set new remote address to 172.16.13.14:27564
[Nov 10 14:53:04]     -- Stopped music on hold on SCCP/7317-00000143
[... 18 lines cut ...]
I hear MoH on "7317" device... wtf??? It seems like channel is not Resumed...

Putting device "5720" on Hold

[Nov 10 15:01:47]     -- Started music on hold, class 'default', on channel 'SCCP/5720-00000145'

Device "5720" - hear MoH.

[Nov 10 15:01:50]        > 0x7fe59c02fc70 -- Strict RTP learning after remote address set to: 172.16.13.11:21450
[... cut ...]
Press HOLD on "5720" device. But no logs like "Started music on hold, class 'default', on channel 'SCCP/7317-00000143'"
Press RESUME on "5720" device

[Nov 10 15:03:18]        > 0x7fe5a4035500 -- Strict RTP learning after remote address set to: 172.16.13.14:23430

At last... Now we see
[Nov 10 15:03:18] NOTICE[10452][C-000000a1]: sccp_rtp.c:242 sccp_rtp_set_peer: SEP003094C39A23: ( sccp_rtp_set_peer ) Set new remote address to 172.16.13.14:23430
[Nov 10 15:03:18]     -- Stopped music on hold on SCCP/7317-00000143

And NOW Voice is OK in both directions!
```

dkgroot, 2019-11-11:
@nickvsh Ok pretty good news. I normally don't test putting both sides on hold, because that is a bit of an edge case (TBH). Thank you very much for testing this and reporting your findings in this detail, that is very helpfull going forward.

The next step will have to be to rebase the changes made in 394_2 on top of the current develop tree to see if the solution is still upheld. This will be a little involved though, because the last commit to this branch is quite some time ago. I will see what i can do later this week. In the meantime i guess you have to resort to using indirectrtp or using 394_2 + single hold/resume.

When that is done, we can check the reason for the edgecase presented above.

nickvsh, 2019-11-12:
Hi, @dkgroot 

> that is a bit of an edge case (TBH)
You are 100% right! Just saying about the issue :) 

> Thank you very much for testing this and reporting your findings in this detail, that is very helpful going forward.
You are always welcome! I am trying to help as possible as I can.

I am using Release: 4.3.2 - 558d2b6M at the moment.

RTP in calls between the SCCP devices flows directly!
The "indirectrtp" is only the case when calling pjsip-endpoints on asterisk (like trunks to ITSP, Cisco CUCM) or sip-voice gateways (Cisco CUBE/OpenVox).
As I don't record phone calls, my interest was to setup DirectRTP between as much devices as possible to reduce traffic and load on asterisk server and also exclude any RTP/voice quality related issues in the future (caused by asterisk/server itself).

The only issues with stability/inconvenience that I sometimes have with chan_sccp/asterisk:

- when I applying new config after adding new lines/new devices with "sccp reload"  - if there is no calls with SCCP staff - all is OK. But if  there are active channels with SCCP - I have high load from asterisk process during long period of time until OK again, or high load persists which leads to RTP/voice quality issues and, maybe, crash and asterisk reboot in several days. But again, if no SCCP active channels + sccp reload - all is OK.

- some SCCP channels don't hangup after call ending and are still there, like this one. I suppose it can be connected with "sccp reload". And I simple can't do "channel request hangup CHANNEL_NAME", because  channel don't exist anymore.

```
> SCCP/5611-000000AF   ***_national         056***42*2          1 Ringing AppDial      (Outgoing Line)           5611            07:50:20                                             
> SCCP/7736-00000ACF   ***_national         7235                2 Up      Dial         PJSIP/7235@cucm,30,r(ring 7736            00:01:18                         b39599a0-94b3-479c-9
> SCCP/5608-000000AD   pzu_national         056***42*2          1 Ringing AppDial      (Outgoing Line)           5608            07:50:20                                               
```
Maybe, you can give me some advice in this situation.

In general, chan_sccp is pretty stable, as I can say, if you setup it and don't change anything))
Cool project - thank you very much!

dkgroot, 2019-11-12:
@nickvsh 
DirectRTP makes the most sense if you have multiple devices on remote sites. For local phones to a local asterisk server, the minimal amount added network bandwidth and processing should be negligent.

__sccp reload__:
You can get rid of having to perform `sccp reload` for the most part if you switch to using a realtime database for the devices and lines. Only changes to the [generic] section in sccp.conf would require a reload, and that doesn't happen that often. If you add/remove/change any device or line, just `sccp restart XXX` those devices that are involved. The database will be checked and reloaded for the devices in question, during register.

__left_over_channels__:
Channels should be hungup and removed during normal operation and `sccp reload` should not have anything to do with. We would need to track down these left of channels in debug to find out what caused them (in a separate issue report). The issue may have already been fixed since 558d2b6, i don't know.

dkgroot, 2019-11-12:
BTW: Working on the rebase now, should be done shortly.

nickvsh, 2019-11-12:
> if you have multiple devices on remote sites.

Exactly! Multiple phones in multiple branch VPN-networks, that connects to the headquarters, where asterisk is located. That is why DirectRTP was 'a must have' feature for my setup. Now all internal SCCP-calls have direct media flow, except sccp <-> pjsip calls.

> if you switch to using a realtime database for the devices and lines.

Thank you for hint! In fact, planning to migrate to FreePBX 15 + [sccp_manager](https://github.com/PhantomVl/sccp_manager/wiki) + yours phone [provisioner ](https://github.com/dkgroot/provision_sccp) (tested it yesterday - cool thing!)

After migration I will test the latest asterisk available in FreePBX + chan_sccp develop branch and will give you feedback about Hold/Resume/DirectRTP staff.

Thank you and have a nice day!

dkgroot, 2019-11-12:
Rebase has been done. Please keep a copy of your current 558d2b6 based directory (because it has been overwritten and removed as a side effect of the rebase). The clone a new copy of chan-sccp using branch `issues/394_2` to test the current behaviour. I have not made any changed except for rolling the 558d2b6 changes on top of the latest develop branch. We should progress from that stance to see what further work needs to be done. Please let me know your findings.

Please run this latest version in a test situation/environment first to make sure there are no major regressions.

In the meantime i will have to do a couple of tests using directrtp and pjsip to see where the direcrtp flow is being blocked.

If you are running in a VM you might want to snapshot before migrating, to make sure you have a smooth rollback.

nickvsh, 2019-11-12:
@dkgroot thank you for quick response!

I have tested the mentioned branch
```
hq*CLI> sccp show version 
Skinny Client Control Protocol (SCCP). Release: 4.3.2 issues/394_2 - 345dc6b (built by 'voip' on '2019-11-12 17:14:35 UTC')
```
DirectRTP between SCCP endpoints works.

But today I found out, that even on old **issues/394_2 558d2b6M** branch code with **directrtp=on** option , there is  a "single directional RTP" issue after attended transfer:

- NO RTP **from device**, that was transfer destination. 
- but RTP flow **to device** is OK.  

The same, of course, is on tested "issues/394_2 - 345dc6b".   

If set **directrtp=off** - RTP is bidirectional after attended transfer. All is OK.

Here is some call logs. Later if needed I can trace all asterisk protocol debug messages.

```
[Nov 12 19:24:10]   == Setting global variable 'SIPDOMAIN' to '192.168.250.2'
[Nov 12 19:24:10]     -- Executing [38057***10*1@from-cube:1] Verbose("PJSIP/cube-00000005", "log:inbound:38057***10*1 from 38067***4*6*") in new stack
[Nov 12 19:24:10] log:inbound:38057***10*1 from 38067***4*6*
[Nov 12 19:24:10]     -- Executing [38057***10*1@from-cube:2] Gosub("PJSIP/cube-00000005", "sub-sticky-in,start,1") in new stack
[Nov 12 19:24:10]     -- Executing [start@sub-sticky-in:1] NoOp("PJSIP/cube-00000005", "Inbound sticky logic") in new stack
[Nov 12 19:24:10]     -- Executing [start@sub-sticky-in:2] Set("PJSIP/cube-00000005", "__sticky_flag=1") in new stack
[Nov 12 19:24:10]     -- Executing [start@sub-sticky-in:3] Set("PJSIP/cube-00000005", "client=38067***4*6*") in new stack
[Nov 12 19:24:10]     -- Executing [start@sub-sticky-in:4] Set("PJSIP/cube-00000005", "DID=38057***10*1") in new stack
[Nov 12 19:24:10]     -- Executing [start@sub-sticky-in:5] NoOp("PJSIP/cube-00000005", "Call from '38067***4*6*' to Line '38057***10*1'") in new stack
[Nov 12 19:24:10]     -- Executing [start@sub-sticky-in:6] Set("PJSIP/cube-00000005", "sticky_number=") in new stack
[Nov 12 19:24:10]     -- Executing [start@sub-sticky-in:7] NoOp("PJSIP/cube-00000005", "38067***4*6*") in new stack
[Nov 12 19:24:10]     -- Executing [start@sub-sticky-in:8] Set("PJSIP/cube-00000005", "CALLERID(num)=9067***4*6*") in new stack
[Nov 12 19:24:10]     -- Executing [start@sub-sticky-in:9] NoOp("PJSIP/cube-00000005", "9067***4*6*") in new stack
[Nov 12 19:24:10]     -- Executing [start@sub-sticky-in:10] Set("PJSIP/cube-00000005", "CALLERID(name)=+38067***4*6*") in new stack
[Nov 12 19:24:10]     -- Executing [start@sub-sticky-in:11] GotoIf("PJSIP/cube-00000005", "1?call_DID") in new stack
[Nov 12 19:24:10]     -- Goto (sub-sticky-in,start,14)
[Nov 12 19:24:10]     -- Executing [start@sub-sticky-in:14] Verbose("PJSIP/cube-00000005", "log:Sticky = Failed -> call DID. inbound_NOT_sticky_call from '38067***4*6*' to cucm/057***10*1") in new stack
[Nov 12 19:24:10] log:Sticky = Failed -> call DID. inbound_NOT_sticky_call from '38067***4*6*' to cucm/057***10*1
[Nov 12 19:24:10]     -- Executing [start@sub-sticky-in:15] Dial("PJSIP/cube-00000005", "Local/057***10*1@did-all,60") in new stack
[Nov 12 19:24:10]     -- Called Local/057***10*1@did-all
[Nov 12 19:24:10]     -- Executing [057***10*1@did-all:1] Verbose("Local/057***10*1@did-all-00000003;2", "log:inbound:057***10*1 from 9067***4*6*") in new stack
[Nov 12 19:24:10] log:inbound:057***10*1 from 9067***4*6*
[Nov 12 19:24:10]     -- Executing [057***10*1@did-all:2] Dial("Local/057***10*1@did-all-00000003;2", "SCCP/8100&SCCP/8101,30") in new stack
[Nov 12 19:24:10]     -- Called SCCP/8100
[Nov 12 19:24:10]     -- Called SCCP/8101
[Nov 12 19:24:10]     -- SCCP/8101-00000006 is ringing
[Nov 12 19:24:10]     -- SCCP/8100-00000005 is ringing
[Nov 12 19:24:10]     -- Local/057***10*1@did-all-00000003;1 is ringing
[Nov 12 19:24:17]     -- SCCP/8101-00000006 is making progress passing it to Local/057***10*1@did-all-00000003;2
[Nov 12 19:24:17]   == Using SCCP RTP TOS bits 184
[Nov 12 19:24:17]   == Using SCCP RTP CoS mark 6
[Nov 12 19:24:17]     -- SCCP/8101-00000006 connected line has changed. Saving it until answer for Local/057***10*1@did-all-00000003;2
[Nov 12 19:24:17]        > 0x7f2020096e00 -- Strict RTP learning after remote address set to: 172.16.13.17:16384
[Nov 12 19:24:17]     -- SCCP/8101-00000006 answered Local/057***10*1@did-all-00000003;2
[Nov 12 19:24:17]     -- Local/057***10*1@did-all-00000003;1 answered PJSIP/cube-00000005
[Nov 12 19:24:17]     -- Channel SCCP/8101-00000006 joined 'simple_bridge' basic-bridge <d186d62c-d764-4a31-a535-4f016c1d1323>
[Nov 12 19:24:17]     -- Channel Local/057***10*1@did-all-00000003;2 joined 'simple_bridge' basic-bridge <d186d62c-d764-4a31-a535-4f016c1d1323>
[Nov 12 19:24:17]        > 0x7f2014051a30 -- Strict RTP learning after remote address set to: 192.168.250.1:25132
[Nov 12 19:24:17]     -- Channel Local/057***10*1@did-all-00000003;1 joined 'simple_bridge' basic-bridge <e8ad2779-c11d-491b-9fa5-66d57d9380c0>
[Nov 12 19:24:17]     -- Channel PJSIP/cube-00000005 joined 'simple_bridge' basic-bridge <e8ad2779-c11d-491b-9fa5-66d57d9380c0>
[Nov 12 19:24:17]        > 0x7f2014051a30 -- Strict RTP switching to RTP target address 192.168.250.1:25132 as source
[Nov 12 19:24:17]        > Move-swap optimizing Local/057***10*1@did-all-00000003;2 <-- PJSIP/cube-00000005.
[Nov 12 19:24:17]     -- Channel PJSIP/cube-00000005 left 'simple_bridge' basic-bridge <e8ad2779-c11d-491b-9fa5-66d57d9380c0>
[... cut ...]
[Nov 12 19:24:58] NOTICE[16469][C-00000006]: sccp_channel.c:1268 sccp_channel_closeAllMediaTransmitAndReceive: SCCP/8101-00000006: (closeAllMediaTransmitAndReceive) called without a valid device
[Nov 12 19:25:28]     -- Channel SCCP/8100-00000009 left 'simple_bridge' basic-bridge <d186d62c-d764-4a31-a535-4f016c1d1323>
[Nov 12 19:25:28]     -- Channel PJSIP/cube-00000005 left 'simple_bridge' basic-bridge <d186d62c-d764-4a31-a535-4f016c1d1323>
[Nov 12 19:25:28]   == Spawn extension (sub-sticky-in, start, 15) exited non-zero on 'PJSIP/cube-00000005'

```

dkgroot, 2019-11-12:
Thank you for checking the new version of the branch (happy to see there are no further regressions). Also transfer and hold/resume are sort of related, because during a transfer a hold is being issued so we can call the new destination).

Looking at your debug logging, i notice that you are using a Local Channel and having multiple destinations in the Dial() command. This is making analysis unnecessarily difficult especially when it comes to directrtp. Analysing directrtp calls especially with remote networks becomes harder because of a couple of things. Having to deal with multiple ip-ranges and not being able to see the rtp packets (which either flow directly or get lost somewhere). So going forward we should agree to start with simple SCCP->SCCP calls, and then upgrade to SCCP->PJSIP etc. Second when debugging rtp, we should set `sccp debug core, channel, pbx, indicate, rtp`.

dkgroot, 2019-11-12:
BTW
```
[Nov 12 19:24:47] NOTICE[16471][C-00000005]: sccp_rtp.c:254 sccp_rtp_set_peer: SEP000DBC979ABC: ( sccp_rtp_set_peer ) Set new remote address to 172.16.13.37:16384
[Nov 12 19:24:47] NOTICE[16471][C-00000005]: sccp_rtp.c:254 sccp_rtp_set_peer: SEP000F348584A2: ( sccp_rtp_set_peer ) Set new remote address to 172.16.13.17:16384
```
This at least is propably wrong, normally two different phones should be using two different ports, we do normally try to use the same port on the phone and the server side, but not reuse the same port for both devices. In this case that would lead to problems when the phones are remote, because the ports would map to the same address on the firewall.

nickvsh, 2019-11-13:
> This at least is probably wrong, normally two different phones should be using two different ports
Maybe, but that is what I see in logs.
[image]

> So going forward we should agree to start with simple SCCP->SCCP calls, and then upgrade to SCCP->PJSIP etc. Second when debugging rtp, we should set sccp debug core, channel, pbx, indicate, rtp. And your inclusion of core set verbose 5 and core set debug 1 is already perfect.

OK 

>Also little advise, please include a little diagram of the phones/firewal/server ip-addresses when returnning any debug. It's hard enough to keep it all apart when reading log info

Got it!