chan-sccp/chan-sccp #396: BLF / Privacy Light Issue

KnightRider89, 2017-03-28:
I'm in the process of setting up 4 Cisco 7975G Phones with FreePBX 13 and the most recent revision of CHAN-SCCP.  The phones are working fine, however I'm having difficulty with the BLF lights. I've attempted to use the following code for the BLF speeddial to other extensions:

`button = speeddial,John Doe, 201, 201`

This code allows the BLF to work, but has two issues.  The main "issue" is that this staff member can see who John Doe is on the phone with.  I would like for him to see that John Doe is on the phone, but not the phone number of the person he is talking to.  The second very minor issue is that the phone logo sits on top of the speed dial logo, making the system look a little bit cluttered.  Is there a way to fix these issues?

I attempted to use the following code instead.  It fixed the image overlay issue, however it disabled all BLF functionality.

`button = speeddial,John Doe, 201`

Sorry if I posted this in the wrong location.

dkgroot, 2017-03-28:
@KnightRider89 
You can disable the callerid information on remote hinted speeddials using
```
./configure --disable-dynamic-speeddial-cid
```
This will only disable the display of remote call information, but not the monitoring feature of hinted a hinted speeddial, which is what you asked for.

We can not change the icon used to display a hinted speeddial, this is functionality built-in to the phone, so fixed.

Next time it might be better to ask questions like this via our mailing list. The github issues are mostly used for problems you are having with the code. On the mailinglist other users will be able to help out answering your question , speeding up the process.

dkgroot, 2017-03-28:
You can close the issue when you have no further questions, leaving it open for 1-2 more days

closed 2017-04-03 by dkgroot