chan-sccp/chan-sccp #397: hung channels?

jrodder, 2017-03-29:
It's not really so much of a problem with this current catch, but I had an issue previously with the same symptoms where a user could not receive calls from the queue, and I had to restart asterisk to clear the "ghost" channels. When this happened before it was a different description, but there were 2 "ghost" channels associated with the extension. If there is a way to manually clear those out without restarting asterisk, I couldn't find it. If the func-apply-siphead gives a clue as to the trigger and how I can run a stacktrace, I would gladly do it.

**Reproduction Steps:**

1. No currently found way to create trigger condition other than system operating for a few days

**Expected behavior:**

lines show up associated with extensions that are in the channels dialog
SCCP/00205-000002AA  s@func-apply-siphead Down    Return()

**Observed behavior:**

Same as observed?

**Screenshots**

[http://imgur.com/a/kLQTI](url)

**chan-sccp version:** 
Skinny Client Control Protocol (SCCP). Release: 4.3.0 master - 337b104M (built by 'root' on '2017-02-08 06:39:21 UTC')

**asterisk version:**
Asterisk 13.12.1 built by mockbuild @ jenkins2.schmoozecom.net on a x86_64 running Linux on 2016-10-28 03:40:04 UTC

dkgroot, 2017-03-29:
@jrodder 
There are some issues with the current master branch. We are trying to get the develop branch ready for release, but is going slower that we had hoped. The current develop branch is more stable than the master branch (under normal usage), so i would recommend switch to the develop branch. The ghost channel issue you are seeing in master should not happen there.
Please use git to clone our repository and give it a go.
Note: i have not been able to find a way to terminate these ghost channel. These remnants cannot be hangup or ended.

jrodder, 2017-03-29:
Thanks for the quick response. I will look at how to do a git clone properly sometime in the near future. It's our production system, and it's a busy day so I'm not trying to go breaking things. :)

dkgroot, 2017-03-29:
@jrodder

Initial checkout:
``` 
git clone https://github.com/chan-sccp/chan-sccp.git chan-sccp_dev
cd chan-sccp_dev
./configure [...the configure flags you want to use...]
make -j2 && make install
```

after that:
```
cd chan-sccp_dev
git pull
make -j2 && make reload
```

So that's pretty easy and simple (i think). Please copy/clone your production machine so that you can practice a little, and make sure the 'develop' branch is working correctly with your system.

Closing this issue report.

closed 2017-03-29 by dkgroot