chan-sccp/chan-sccp #398: Asterisk service is disconected when try to send a call to a sccp phone using a .call file

miguelsanzpardo, 2017-04-03:
When you try to send a call to a sccp phone using a .call file the asterisk service is disconected suddenly.

**Reproduction Steps:**
1. You call to a virtual extension which execute a .sh
2. The .sh mv a .call file from /var/spool/asterisk/outgoing/temp to /var/spool/asterisk/outgoing/temp
3. The system try to generate a call to an sccp phone

**Expected behavior:**
The sccp phone is offhooked and then can listen an audio.

**Observed behavior:**
The system asterisk go down.

**Dialplan, .sh and .call used

extensions.conf
[from-internal-hospital]
...
exten => 980,1,NoOp()
same => n,Playback(/var/lib/asterisk/sounds/es/barcelona)
same => n,System(sh /var/spool/asterisk/outgoing/temp/coffee_hour_id.sh)
same => n,Hangup()

exten => _98X,1,NoOp()
same => n,Wait(1)
same => n,Playback(/var/lib/asterisk/sounds/es/valencia)
same => n,Wait(1)
same => n,Hangup()

coffee_hour_id.sh
sleep 1
cp /var/spool/asterisk/outgoing/temp/emilio_coffee.call /var/spool/asterisk/outgoing/emilio_coffee.call

emilio_coffee.call
Channel: SCCP/1104
Context: from-internal-hospital
Extension: 982
Priority: 1

NOTE:
If the device is using SIP the problem doesnt appear and the call is generated
When the problem appears it is generated a core file.

**chan-sccp version:**
4.3.0- Release: 4.3.0 develop - 128c709M

**asterisk version:** (include revision and/or branch if relevant)
Asterisk 11.21.2

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: I dont know
- Problem can be reliably reproduced, doesn't happen randomly: Yes
- Platform:
  - [CPU]
  - [64 Bit]
  - [Operating System] --> Debian 8.4

- Log Files (attach .txt / [pastbin](http://pastebin.com/) / [gist](http://gist.github.com/)): [url]
[full-03-04-2017.txt](https://github.com/chan-sccp/chan-sccp/files/889810/full-03-04-2017.txt)

dkgroot, 2017-04-03:
Hi Miguel,
What are you trying to achieve ? 
Using System and .call originate scripts from the dialplan is not threadsafe (i think). You should try to avoid SYSTEM() and SHELL() if at all possible. If you need to DIAL a specific extension, your script should have returned the extension to call and passed it back to the dialplan.
You should try to keep the call flow within a single thread of execution. You can use dialplan macro's and/or AGI scripts to extend the dialplan logic.

dkgroot, 2017-04-03:
@miguelsanzpardo What did the coredump/segfault look like (backtrace?)

dkgroot, 2017-04-03:
/var/lib/asterisk/agi-bin/test.sh
```
#!/bin/bash
echo 'VERBOSE "we are running the bash agi script" 1'
echo 'SET VARIABLE dial_endpoint "SCCP/1104"'
exit 0
```

dialplan
```
exten => 980,1,NoOp()
same => n, Answer()
same => n, Playback(/var/lib/asterisk/sounds/es/barcelona)
same => n, AGI(test.sh)
same => n, DIAL(${dial_endpoint}, 20, tr)
same => n, Hangup()
```
Now AGI calls and processes the script. The test.sh script sets an asterisk variable in this case. The DIAL() command uses this asterisk variable.

**Note: using bash as an AGI script is not the best choice. You should use php/python/etc.**

Because you haven't shown what you coffee script does, i don't know what the intent of this whole procedure is. Also without a backtrace there is no way, to know what happened.

miguelsanzpardo, 2017-04-03:
[coredump_info.txt](https://github.com/chan-sccp/chan-sccp/files/890220/coredump_info.txt)

miguelsanzpardo, 2017-04-03:
I have test with something like that:

exten => 990,1,NoOp()
same  => n,Answer()
same  => n,Playback(/var/lib/asterisk/sounds/es/barcelona)
; same  => n,Originate(SIP/1004,app,Playback,/var/lib/asterisk/sounds/es/valencia,,20)
same  => n,Originate(SCCP/1104,app,Playback,/var/lib/asterisk/sounds/es/valencia,,20)
same  => n,Hangup() 

and the problems appears again with SCCP but dont appear with SIP

miguelsanzpardo, 2017-04-03:
I think that the fast/best solution will be something like that:

exten => 990,1,NoOp()
same => n,Playback(/var/lib/asterisk/sounds/es/barcelona)
same => n,Page(SIP/1004&SCCP/1104,dnA(/var/lib/asterisk/sounds/es/valencia),30)
same => n,Hangup()

The unique problem will be that the calling party wont can hangup meanwhile the page is done.

miguelsanzpardo, 2017-04-04:
I share with you the core file which has been generated when using this dialplan:

exten => 990,1,NoOp()
same => n,Answer()
same => n,Playback(/var/lib/asterisk/sounds/es/barcelona)
same => n,Originate(SCCP/1104,app,Playback,/var/lib/asterisk/sounds/es/valencia,,20)
same => n,Hangup()

[core.txt](https://github.com/chan-sccp/chan-sccp/files/892694/core.txt)

dkgroot, 2017-09-13:
Issue resolved in 4b7ec6f / references #411

closed 2017-09-13 by dkgroot