chan-sccp/chan-sccp #399: chan-sccp-b causes system to lock-up entirely

NF3Q, 2017-04-06:
I currently have a virtual machine running Asterisk 13.3.2, with CentOS 6.9, kernel 2.6.32-696 (64 bit) on an ESXi hypervisor, 6.5, build 5146846.  Upon updating the kernel from 2.6.2-642.15.1, the system crashes.  Recompiling chan-sccp-b from source, be it 4.2.3 or 4.3.0 (trunk) seems to not alleviate any issues under the new kernel.  Unloading (or not loading altogether) the chan_sccp.so allows the system to behave in a normal manner.  Given that this happens shortly after the module is loaded, gleaning diagnostic logs is seemingly impossible.

Reproducing:
* install Asterisk 13.3.2
* update CentOS host to version 6.9 (released April 5, 2017)
* ensure all kernel headers and debug packages are installed, as needed
* configure and compile chan_sccp.so module (./configure, make clean, make, make install)
* ensure module loads in /etc/asterisk/modules.conf (load=>chan_sccp.so)

drtmk, 2017-04-07:
Maybe try updating Asterisk first. 13.3.2 is 2 years old. Lots of bugs have been fixed since that time.

NF3Q, 2017-04-07:
Done, problem persists, but this time I caught in action with 4.3.0/trunk ...

[Apr  7 17:06:22] NOTICE[28155]: chan_sccp.c:154 sccp_prePBXLoad: preloading pbx module
  == Starting Threadpool
    -- Created thread 0(0x7fab23cba700) in pool
    -- Created thread 1(0x7fab23c3e700) in pool
    -- Threadpool Started
  == Starting event system
  == SCCP: Starting devstate system
  == SCCP: Starting MWI system
  == SCCP: Starting hint system
    -- Added line 'Hotline' to Glob(lines)
  == Platform byte order   : LITTLE ENDIAN
[Apr  7 17:06:22] NOTICE[28155]: sccp_config.c:2744 sccp_config_getConfig: Config file 'sccp.conf' loaded.
    -- Added line '2001' to Glob(lines)
    -- Added line '2002' to Glob(lines)
    -- Added line '2003' to Glob(lines)
    -- Added line '2004' to Glob(lines)
    -- Added line '2005' to Glob(lines)
    -- Added line '2006' to Glob(lines)
    -- Added line '2007' to Glob(lines)
    -- Added line '2008' to Glob(lines)
    -- Added line '2009' to Glob(lines)
    -- Added line '2010' to Glob(lines)
    -- Added line '2011' to Glob(lines)
    -- Added line '2012' to Glob(lines)
    -- Added line '2013' to Glob(lines)
    -- Added line '2014' to Glob(lines)
    -- Added line '2015' to Glob(lines)
    -- Added line '2016' to Glob(lines)
    -- Added line '611' to Glob(lines)
    -- Added line '14124894111' to Glob(lines)
    -- Added line '17242341333' to Glob(lines)
    -- Added line '17245262011' to Glob(lines)
    -- Added line '17245262022' to Glob(lines)
    -- Added line '17245262093' to Glob(lines)
    -- Added line '17245262094' to Glob(lines)
    -- Added line '17245263383' to Glob(lines)
    -- Added line '17246022810' to Glob(lines)
    -- Added line '17249543800' to Glob(lines)
    -- Added line '17249543815' to Glob(lines)
    -- Added line '17249543816' to Glob(lines)
    -- Added line '18444539823' to Glob(lines)
    -- Added line '18445353947' to Glob(lines)
    -- Added line '18555353947' to Glob(lines)
    -- Starting Threadpool JobQueue:0x7fab23c3e700
    -- SCCP listening on 0.0.0.0:2000
    -- Starting Threadpool JobQueue:0x7fab23cba700
Asterisk Ready.
*CLI>
*CLI>
*CLI>
*CLI>
*CLI>
*CLI>     -- SCCP: Accepted Client Connection from 172.30.101.205:51941
    -- SCCP: Connected on server via 172.16.255.159:39812
  == Alarm Type: LastOutOfServiceInformation
  == Reason Enum: 12
  == Last Event Sent: 0:KeepAlive
  == Last Event Received: 256:KeepAliveAck
    -- SEP00152B7B208C: Device Not NATTED. Device IP '172.30.101.205' falls in localnet scope
       > SEP00152B7B208C: Asked for our protocol capability (22).
       > SEP00152B7B208C: Phone protocol capability : 22
       > SEP00152B7B208C: Our protocol capability   : 22
    -- SEP00152B7B208C: Joint protocol capability : 22
    -- SEP00152B7B208C: Ask the phone to send keepalive message every 46 seconds
    -- SEP00152B7B208C: Got rtpPort:3500 which the device wants to use for media
    -- SEP00152B7B208C: using softkeyset: default (0x16d73b0)!
    -- SEP00152B7B208C: Sending ConfigStatMessage, lines 8, speeddials 0
    -- SEP00152B7B208C: Device has 8 Capabilities
    -- SEP00152B7B208C: Got Forward Status Request.  Line: 1

NF3Q, 2017-04-07:
I might add that the particular handset (SEP00152B7B208C) is a Cisco 7971G/GE, running SCCP firmware SCCP70.9-4-2SR2-2S

NF3Q, 2017-04-26:
Any response here? I suspect this will be a wide, routine problem.

dkgroot, 2017-08-31:
Hi NF3Q,

- Asterisk-13 or Asterisk-14 should both be fine.
- When you do experience a crash or segfault, we require a so called backtrace. See: https://github.com/chan-sccp/chan-sccp/wiki/Reporting-Bugs
A backtrace is the only thing that would shed some light on the issue.

BTW: Linux kernel 2.6.xx is pretty ancient, would it not be time to move forward in time a little ?

Sorry for my long absence. Trying to pickup the slack.

closed 2017-09-28 by NF3Q

dkgroot, 2017-09-28:
@NF3Q 
Hi John, Was the issue resolved in the meantime / Did you find out the cause / Did it not re-appear any more ?

NF3Q, 2017-09-28:
I didn't find the cause, despite trying to grab a backtrace.  Since the issue appeared to be something that wouldn't be looked at, I replaced the PBX, changed all of the phones to SIP firmware, and yanked SCCP altogether.

dkgroot, 2017-09-29:
Hi John,
I'm sorry to hear that. We never received the backtrace (as far as i know). You might be interested in the http://usecallmanager.nz project running SIP on cisco phones.
Regards,
Diederik