chan-sccp/chan-sccp #400: Conferences (ConfBridge) crash / unstable?

kb4fxc, 2017-04-07:
I've been testing chan-sccp-b over the last couple weeks a using a custom build from source of Asterisk  13.14.0.  I've tried  the "stable" release driver (Chan_SCCP-4.2.3_STABLE_r6728.tar.gz) and the most recent development code, downloaded from github (as of 2017-04-05), both compiled with the '--enable-conference' option. ConfBridge seems properly operational from Asterisk, I setup a stand-alone conference accessible via the dial plan and it works fine.  I have the 2.05 dahdi drivers loaded.

The Linux platform is Ubuntu 14.04.5 LTS, x86_64, currently running on a VM for testing.

I'm testing with a mixture of 7961G, 7965G and 7976G phones. I've tried uLAW and G729 CODECS.

I am able to establish a properly functional conference between 3 phones initially. Once I end the conference, if I try the exact same steps a second time, the channel driver crashes. Sometimes this ends with a segfault, at other times the driver is left broken. Once this happens, I'm enable to shutdown Asterisk (core stop now) and must kill it manually.  This problem seem reproducible 100% of the time.

I do have music on hold enabled.

Note that this is the only problem I've found, so far. Everything else seems to work properly (transfers, parking calls, forwarding, pick-up groups, etc.). I've not tried the MeetMe functionality.

Any ideas??

Thanks,

David McGough

kb4fxc, 2017-04-07:
BTW, for firmware versions, I running:

SCCP41.9-4-2SR3-1S     (7961G phones)
SCCP45.9-4-2SR3-1S     (7965G phones)
SCCP75.9-4-2SR3-1S     (7975G phones)

Thanks,

David McGough

kb4fxc, 2017-04-10:
Well, I'm mixing apples and oranges here, but, I moved backwards to Asterisk version 11.25.1 (built from source) leaving as many of the other variables as possible alone, and the conferences and everything else seems to be working properly, no problems so far.  I'll test carefully this week.

I'm running the chan-sccp-b version from git, downloaded on April 05, 2017.

Thanks,

David McGough

kb4fxc, 2017-04-14:
Anyone else notice sccp conference problems using Asterisk 13.14.x ????

kb4fxc, 2017-05-01:
Asterisk version 11.25.1 is still working great. Anyone else noticing conference problems using Asterisk 13.14.x ????

jaganthoutam, 2017-05-18:
Have problem with Asterisk Asterisk 13.12.2

#0  0x00007f0d37dee61f in ast_rtp_codecs_payload_code (codecs=0x7f0cb88c61a0, asterisk_format=asterisk_format@entry=1, format=format@entry=0x7f0d39557160, code=code@entry=0) at rtp_engine.c:926
#1  0x00007f0c9927224f in bridge_p2p_rtp_write (hdrlen=12, len=172, rtpheader=0x7f0cb8876120, instance=0x7f0cb875b9a8) at res_rtp_asterisk.c:4302
#2  ast_rtp_read (instance=0x7f0cb875b9a8, rtcp=<optimised out>) at res_rtp_asterisk.c:4519
#3  0x00007f0c8cb3e406 in sip_rtp_read (faxdetect=<synthetic pointer>, p=0x7f0cb8490a98, ast=0x7f0cb8921af8) at chan_sip.c:8524
#4  sip_read (ast=0x7f0cb8921af8) at chan_sip.c:8630
#5  0x00007f0d37d41962 in __ast_read (chan=0x7f0cb8921af8, dropaudio=dropaudio@entry=0) at channel.c:3898
#6  0x00007f0d37d42727 in ast_read (chan=<optimised out>) at channel.c:4323
#7  0x00007f0d37d0f991 in bridge_handle_trip (bridge_channel=0x7f0cb8912c28) at bridge_channel.c:2416
#8  bridge_channel_wait (bridge_channel=0x7f0cb8912c28) at bridge_channel.c:2586
#9  bridge_channel_internal_join (bridge_channel=bridge_channel@entry=0x7f0cb8912c28) at bridge_channel.c:2732
#10 0x00007f0d37cfe353 in ast_bridge_join (bridge=0x7f0cb889ef48, chan=0x7f0cb8921af8, swap=swap@entry=0x0, features=0x7f0cb8911aa0, tech_args=tech_args@entry=0x0, flags=flags@entry=(unknown: 0))
    at bridge.c:1712
#11 0x00007f0c98dc5ceb in sccp_conference_thread (data=data@entry=0x7f0cb8911a00) at sccp_conference.c:649
#12 0x00007f0d37e3d37e in dummy_start (data=<optimised out>) at utils.c:1235
#13 0x00007f0d35b06184 in start_thread (arg=0x7f0c6e1dc700) at pthread_create.c:312
#14 0x00007f0d350e237d in __ecvt_r (value=1.9950297492085153e+161, ndigit=0, decpt=0x0, sign=0x0, buf=0x7f0c6e1dc9c0 "\300\t$t\f\177", len=139691363780352) at efgcvt_r.c:218
#15 0x0000000000000000 in ?? ()

dkgroot, 2017-09-14:
@kb4fxc / @jaganthoutam 
Sorry i took so long to get around to this.
Looking into the issue today. When setting up the conference, are you including other channel drivers (like sip etc), into the conference or only sccp ?

@jaganthoutam : Your backtrace obviously shows that in your case chan_sip is also involved.

kb4fxc, 2017-09-14:
Hi,
In my situation, all conference members were running 7965G or 7961G phones, with chan_sccp.  I didn't even have any SIP phones, etc., setup (at that point). I have not tested with Asterisk 13.x.y in months, only Asterisk 11.25.1....And, 11.25.1 continues to run perfectly.

I'll try again with Asterisk 13.x.y in a few days, using the latest chan_sccp from github.

Thanks,

David McGough

dkgroot, 2017-09-14:
@kb4fxc 
Please give me a little time to figure out what is going on, and how to possibly fix it, before retesting. Good to hear that ast11 is still stable.

ipnvvoip, 2017-10-27:
We too have a customer experiencing this issue, when a conf is tried it crashes the whole system.

Aaron

dkgroot, 2017-10-29:
I have not been able to reproduce the issue. Can you send me another backtrace (created from the coredump) to help me out, find the cause ?

kb4fxc, 2017-10-29:
Hi Diederik,

I was using Asterisk 13.14.0, from February, 2017. My testing was only with SCCP phones (see my original posts, above). The instability I saw happened reliably, every time.  I then moved back to Asterisk version 11.25.1 and the crashing vanished. I changed no other variables--same phones, firmware, Linux release and PC hardware....What Asterisk release are you using, where you don't see this problem---I'll try it, too! 

Thanks,

David McGough, KB4FXC

dkgroot, 2017-11-02:
@kb4fxc 
Asterisk GIT
Version : 13-13.15.0
revision : g844f70f301

Can you tell me your specific version / revision. Maybe i can go back to that one, to see what is causing the segfault on yours.
If you do get a segfault, please produce a gdb backtrace from the core file and post the gdb output here, that would be very helpful.

kb4fxc, 2017-11-02:
Please see all the details as to versions at the top of this web page:

https://github.com/chan-sccp/chan-sccp/issues/400

In the meantime, I'll test against Asterisk from GIT.

Thanks,

David McGough, KB4FXC

On Thu, 2 Nov 2017, Diederik de Groot wrote:

> @kb4fxc
> Asterisk GIT
> Version : 13-13.15.0
> revision : g844f70f301
>
> Can you tell me your specific version / revision. Maybe i can go back to that one, to see what is causing the segfault on yours.
> If [...]

dkgroot, 2017-11-06:
@kb4fxc 
I managed to reproduce the issue (using asterisk-15). I think i have managed to fix the issue. I created a new branch for you to test/check.

Please do the folowing:
```
cd chan-sccp                 # change path to your chan-sccp source folder
git pull
git fetch -a
git checkout issues/400  $ Moves you to the new branch related to issue/400
make -j2 && make reload
```
When testing, please set "`sccp debug core, channel, pbx, conferences, rtp`".

kb4fxc, 2017-11-06:
Hi, I will test ASAP!

On Mon, 6 Nov 2017, Diederik de Groot wrote:

> @kb4fxc
> I managed to reproduce the issue (using asterisk-15). I think i have managed to fix the issue. I created a new branch for you to test/check.
>
> Please do the folowing:
> ```
> cd [...]

dkgroot, 2017-11-06:
Thanks. Note: changes were implemented for asterisk-15 and (only) backport to asterisk-14. Have not had time to backport them to 13 (yet).

dkgroot, 2017-11-08:
@kb4fxc 
Have run a couple more tests (Incoming & Outgoing calls / SIP Participants etc). They all seem to work correctly now (after the latest patches).
Currently backporting the changes made for asterisk-14 and asterisk-15 to make conference work again, to asterisk-13 (pbx_impl).

kb4fxc, 2017-11-08:
Hi Diederik, Thanks. I downloaded the released Asterisk 13.18.1, but 
haven't tested yet.

On Wed, 8 Nov 2017, Diederik de Groot wrote:

> @kb4fxc 
> Have run a couple more tests (Incoming & Outgoing calls / SIP Participants etc). They all seem to work correctly now (after the latest patches).
> Currently backporting the changes made for asterisk-14 and asterisk-15 to make conference work again, to asterisk-13 (pbx_impl).
> 
>

jcroy, 2017-11-21:
Any progress on whether this worked or not? I cant test it on our test environment properly.

dkgroot, 2017-11-21:
Hi julien,

Have pushed a fix for the issue to a separate branch so it can be tested. Have not heard back yet (from @kb4fxc ). As far as I can see the fix should work. 

@jcroy what is the reason you cannot test this on your environment ? Or did you mean you cannot replicate the issue when using the issues/400 branch?

I normally wait until I get a positive reply before merging a branch back into develop branch

jcroy, 2017-11-21:
@dkgroot The issue only occurs on cisco 79xx and we only have access to one in the office as our other working spare decided to do its own thing.  I tried to test it out using a cisco 79xx and a softphone but softphones and cisco phones have separate issues (no audio ...codec issue possibly) but we did notice that hangup signals were being sent and it seemed to be working better.  So obviously its hard to tell if its fixed or if progress was made at all.

dkgroot, 2017-11-21:
@jcroy ok I understand. Not sure what would be causing the audio issues (is that a separate issue or unrelated ?).

Talking about softphones, you can always use a couple of Cisco IP communicator instances to simulate skinny phones when needed 😉

I have done quite a number of tests in the meantime that I feel quite sure. So I think I might go ahead and merge the fix.

dkgroot, 2017-11-21:
Have merged the issues/400 branch into develop (automatically closed this issue report as well). Please continue checking/testing using the develop branch. If there still is an issue the either reopen this issue or create a new report.

Thank you for reporting the issue !

kb4fxc, 2017-11-21:
Hi,

Yes, this patch seems to work in my test environment. I haven't tested 
with a production-load on it, yet.

Thanks,

David, KB4FXC

On Tue, 21 Nov 2017, Diederik de Groot wrote:

> Hi julien,
>
> Have pushed a fix for the issue to a separate branch so it can be tested. Have not heard back yet (from @kb4fxc ). As far as I can see the fix should work.
>
> @jcroy what is the reason [...]

dkgroot, 2017-11-21:
As the issue was not really load related, and the actual cause was found, we should be ok 😉
The issue is now considered closed. Thanks for reporting back!

jcroy, 2017-11-22:
@dkgroot just used this today. It works, thank you so much.

dkgroot, 2017-11-23:
Thanks for checking and reporting back. Much appreciated
One more issue solved 😃

jcroy, 2017-12-19:
@dkgroot this issue still sorta is happening.  So doing the above did fix the problem but if you hit the 3 caller limit that the phones have it crashes the system in the same way that it did before.  Is there anyway that we can prevent it from crashing at the 3 person limit?  One of our clients just crashed the server because they added a 4th person.  Thanks!

dkgroot, 2017-12-22:
@jcroy 
Will have another look after Christmas, to see what might be going on here. I thought i had it solved :-(

jcroy, 2017-12-29:
@dkgroot I wonder if a hardcoded limit for number of users can be set so that it will not allow any more users to join would that fix the isse?

dkgroot, 2018-01-04:
Hi jcroy, i do not really like adding a hardlimit of 3 for no good reason. The conference should be able to do more than just mix three users. I just need to find some time to dive in and figure out what might be causing the issue.
The problem is I am quite busy with other stuff during this time of year :-( Sorry

dkgroot, 2019-02-22:
No reply since opening this bug report.
Not sure if the issue is still there, if it is please reopen.
Closing the issue.

PR #419 (2017-11-21): Conference causing segfault when using asterisk-13 or higher
Fix: Segfault cause by running conference on >asterisk-14 (NativeFormat was not set correctly when entering the brige)
Fix: Codec selection during channel_request
Enh: Update config allow/disallow handling for video codec preferences
Enh: Display video capabilies and preferences in CLI/AMI output
Fix: Erroneous use of rtp.audio where rtp.video was meant
Backport: Conference related fixed made for ast115 and ast114, backported to ast113

Fixes #400
closed 2019-02-22 by dkgroot