chan-sccp/chan-sccp #402: 8945 - DND Label not correctly showing

bcdaus, 2017-04-29:
When activating DND on a 8945 DND status on-screen does not change.
Other phones show this extension is in DND mode as expected.

**Reproduction Steps:**
Press DND Button
Check CLI to see if Phone in DND Mode
Call phone to see if it is in DND Mode

**Expected behavior:**
Display shows DND Mode

**Observed behavior:**
Display does not show DND mode.

**Screenshots**
When selecting DND on phone.
[image]
Command line shows
-- SEP6CFA897380F4: change dnd to reject
on press of Phone DND button

sccp show device SEP6cfa897380f4
<SNIP>
--- SCCP channel driver device settings ------------------------------------------------------------------
MAC-Address                                    : SEP6CFA897380F4
Protocol Version                               : Supported '22', In Use '22'
Protocol In Use                                : SCCP Version 22
Device Features                                : 0x85702016,10000101011100000010000000010110
<...>
Config Phone Type                              : 8945
Skinny Phone Type                              : Cisco 8945(585)
Softkey support                                : yes
Softkeyset                                     : default => default (0x87e2620)
BTemplate support                              : yes
linesRegistered                                : yes
Image Version                                  : SCCP894x.9-4-2SR3-1
<..>
DND Feature enabled                            : yes
DND Status                                     : Reject
DND Action                                     : Reject
<...>

**chan-sccp version:** (include revision and/or branch if relevant)
Skinny Client Control Protocol (SCCP). Release: 4.3.0 develop - 46f48e9M (built by 'root' on '2016-08-28 09:20:03 UTC')

**asterisk version:** (include revision and/or branch if relevant)
Asterisk 13.7.2

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: Not Sure
- Problem can be reliably reproduced, doesn't happen randomly: Yes
- Platform:
Incredible PBX 13-12.2
Kernel Version - 2.6.32-573.12.1.el6.i686 (SMP) i686
Distro Name - Icon Scientific Linux release 6.7 (Carbon)
CPU - Dual Core Intel(R) Celeron(R) 2955U @ 1.40GHz (ChromeBox)

dkgroot, 2017-09-14:
@bcdaus 
I have no idea how to fix this issue on 8945 phones. They do not seem to follow the same protocol rules as 79xx phones (We have not figure out these protocol differences completely). I would require a cucm tcpdump to see how they get the phone to display this state. 
You can download a trial version of cucm and install it in a VM, if you would like to gather this information and help us out improving 8945 support.

gplindstrom, 2017-09-21:
Diederik, not sure on this, but it may not display a long indication.  On my 9971 SIP phones, I only see a brief message indicating DND is on/off (well, mostly, I will explain later).  Docs at:

https://www.cisco.com/c/en/us/td/docs/voice_ip_comm/cucme/ip_phones/8941_45/english/user/guide/b_8941_45_ug_release8_8/b_8941_45_ug_release8_8_chapter_01.html#concept_97A6A0AB85B74597AE5F307757F327CB 

indicate:

Step 1 | Press the DND button to turn on DND.  **Visual confirmation displays briefly**.
Step 2 | Press the button again to turn off DND.  **Visual confirmation displays briefly**.

I don't know if this is happening for @bcdaus.  

Now, that said, on my 9971 phones, when DND is active, the speeddial buttons change to the "Speed-dial, call log, or directory listing (line status unknown)" icon, I assume because the phone rejects the notify updates for BLF status (it is after all in DND mode).  

Interestingly, if I put a chan_sccp phone in DND (Busy) mode (but not Silent mode) the usecallmanager phones that have that extension with a BLF/Speeddial button will change the icon to the "BLF-monitored line is in-use" icon and the led turns red.  Chan_sccp phones with speeddials indicate DND next to the description and change the icon/led.  But putting a usecallmanager phone in DND mode does not change the speeddials on either usecallmanager or chan_sccp phones, so chan_sccp and usecallmanager operate a bit differently.

One other possible option is to set the <dndCallAlert> and <dndReminderTimer> xml tags in the SEPxxxxxxxxxxxx.cnf.xml file.   This works with my SIP firmware, but does not work with the SCCP firmware.  I don't know if the firmware needs something from chan_sccp or not to make it work. or if it is a bug in the SCCP firmware.  Maybe I don't have something configured right.

Maybe I should get myself a 894x phone so I could look at things like this, but also to see how it works with the transfer stuff that you are working on and if I can figure it out with my transfer patch....

Just my input, FWIW....

dkgroot, 2017-09-21:
@gplindstrom 
Using the line icon to show DND status is a possible solution for the 8945 i had not thought of. I will see if this can be incorporated easily.
Remote indication of DND is not a standard asterisk feature (as far as i know). We might be using a different hint setting then 'usecallmanager' causing different remote indications. Our shared line DND setting might also be different (We only switch on DND for the complete shared line, if all shared-line participants switch on DND on their device, if only one has DND enable, we just exclude ringing that device).
As mentioned before, we are not trying to replicate callmanager or usecallmanager behaviour. But if you can provide the usecallmanager hint value when DND is enabled, we might be able to use the same hint value.

When DND:Busy is enabled on a single line in chan_sccp, the hint setting is 'busy'
When DND:Silent is enabled on a single line in chan_sccp, the hint setting is 'idle' (so that we can show the incoming call, just ringing is suppressed).

dkgroot, 2017-09-21:
@gplindstrom 
Could not find an easy solution, to change the icon on the Line, without causing other strange behaviour. We really need to CUCM tcpdump + 8945 + DND to see what they do.

brummski, 2018-01-10:
Got some CP-6921, Same Problem. DND Button is working, but no indication on the phone.
[image]

FYI: "Ruhe" is DND in german

dkgroot, 2018-01-10:
@brummski The 69XX phones are quite different from any/all of the other SCCP/Skinny phones (even worse than the 89XX phones). We have had a lot of trouble with them, especially when it comes to displaying Status(bar) information. It would help to have some CUCM tcpdump + 69xx + DND. Without that information there is no easy way to figure out how to make this work.

brummski, 2018-01-10:
Hi! I ofcourse understand - i am sorry - i have no CUCM around... but if anybody needs a 6921 for testing - i can provide one.

gthompsongcsc, 2019-03-01:
I have 6921 phones and Cisco Call Manager 7.2 here. I can provide the tcpdump if someone can provide directions on how to get that dump... Is it as easy as using Wireshark and capturing packets between CUCM and the Phone while pressing the DND button?

dkgroot, 2019-03-01:
@gthompsongcsc Yes that would do the trick nicely.

Could you maybe start the wireshark capture, and then restart the phone, so that we also get the registration messages as well.

There might be some other issues with 69xx, so it would be really nice to have someone with a cucm on hand (even if it is a little older). For example not all status messages are displayed on the 69xx phones (when used with chan-sccp). Would love to get a full set of 69xx messages and know how they displayed these (description/video/picture). For example during transfer, call forward, gpickup, pickup etc etc.

gthompsongcsc, 2019-03-11:
@dkgroot Sorry this took so long.. But here is a capture of the 6921 Registering to CUCM. I then Toggle DND on and off 5-6 times, and then turn on Call Forwarding a couple of times... Let me know if you need anything else.

[Cisco6921Cap.pcapng.zip](https://github.com/chan-sccp/chan-sccp/files/2952415/Cisco6921Cap.pcapng.zip)

dkgroot, 2019-03-11:
Would also love to see a "call transfer" and "missed call", ". It's mostly the stuff that's not working very well with chan-sccp and 69xx phones. The biggest thing seems to be the missing control over the status line.

gthompsongcsc, 2019-03-11:
@dkgroot Here ya go.. Missed call and Transfer. Same IP info as above.

[6921MissedCallandTransfer.pcapng.zip](https://github.com/chan-sccp/chan-sccp/files/2952504/6921MissedCallandTransfer.pcapng.zip)

dkgroot, 2019-03-11:
Wow, that was quick... I am not even half way through the first one :-) Much appreciated. It will take a while to dig through it all :-). And it will hopefully improve our 6921 support.

gthompsongcsc, 2019-03-11:
Great, I am looking forward to it, I have about 150 6921's in use and would love to switch from SIP to chan_sccp for them.

dkgroot, 2019-03-11:
If you have some time, it might be easier to use [gitter chat](https://gitter.im/chan-sccp/chan-sccp)..if you like of course

dkgroot, 2019-03-11:
Already found two interesting things... that i need to test.
The ForwardStatRes message uses different values to activate the callforward (2 and 8), where we use 4 for all of them. I need to run some tests to see if that makes a difference. Have not got a clue about that these number do beyond switch the forward on (ie: they are higher that 0).
Question: When callforward is active on the phone does it display the status message "Forwarded to 40" ?

dkgroot, 2019-03-11:
@gthompsongcsc Thanks to your tcpdump, i made the first small changes in a new git branch called "isssues/402"
You can give it a try using:
```
git fetch -a
git pull
git branch
git checkout issues/402
./configure [....]
make && make install && make reload
```

(please do this on you test system and not production (as always).

dkgroot, 2019-03-12:
Status Update:
- 69xx dnd status fixed... using a popup and a building dnd message only (no more fancy "<<<")
- 89xx does now show the dnd state, but only briefly :-( so this still needs work. Why might be able to show the permanent state on the line button instead (red cross or something)
- 79xx lots it's fancy "<<<" and uses the more relaxed/demure "DND (busy)" / "DND (silent)" in tristate mode. In toggle mode it only says "Do not disturb is active" and notifies when it turns it off.

dkgroot, 2019-03-14:
Add Relation to #404

dkgroot, 2019-03-15:
Merged the issues/402 branch into develop to get the current 69xx/89xx related fixed into main stream. Will continue exploring the issue when greg get's back

dkgroot, 2019-03-17:
Note: Setting DND on the 8945 will show a popup for about 5 seconds and is then dropped again... on every user action the popup re-appears. it's not perfect but at least the user is informed. Switching DND off also generates a similar popup. However i noticed only yesterday, that the "Do not disturb is not active" message keeps re-apparearing after each user interaction, which is wrong.
I have not figured out how to get a "stable" DND message on the phone yet, that remains. I will look into the "not active message" which is annoying.

dkgroot, 2019-08-29:
@gthompsongcsc @gplindstrom 
According to the 8945 manual, the DND popup is how this phone is supposed to work. With one difference that when DND is active the change the line light to AMBER. I have not figured out how to make that happen through. For now we just popup the DND warning when you go OFFHOOK/ONHOOK, so the user is aware.
We do have someone with a CUCM now and he is checking out if he can make the amber light appear (no luck yet), once we have that, we should be able to replicate the effect.

dkgroot, 2019-09-15:
After some heavy experimenting using the 8945 phones (on CUCM). We were not able to change the light status of the line button (as the phones manual portrait), so we are assuming this was never actually implemented.

We have decided, that using the "Popup Method" (described above / added some months ago) is as much as we can achieve on the 8945 (for now).

Closing this issue. Thank you for reporting the issue. And sorry that DND isn't that clear on these devices. Your report was responsible for producing the 'Popup Method' though, so there is that.

closed 2019-09-15 by dkgroot

dkgroot, 2020-03-11:
Thanks to @tensi0n519 we figured out how CME manages to have DND on the displaylabel continuesly. Instead of actually fixing the problem with the phones, they just send a displaylabel with 35 second timeout. And then clear the display and send the same message again 34 seconds later. It's an awefull kludge, but i guess it does work. I am considering this solution, but i am not happy about having to do something like this.