chan-sccp/chan-sccp #403: 8945 - Call forwarding disable does not correctly update GUI

bcdaus, 2017-04-29:
When disabling call forwarding on 8945, the Phone GUI does not correctly reflect the call forwarding status until a menu pop up is opened on the phone.

**Reproduction Steps:**
Enable call forwarding to know extension - gui will show forwarding icon and text
[image]
Press Forward Off

**Expected behavior:**
Phone line icon will remove forarding notification.  Forward OFF button text will return to Forward All.  Forwarding text in header will be cleared.

**Observed behavior:**
Phone Icon removes forwarding Icon.
Button remains as Forward Off text
Header retains forwarding info
[image]

Bringing up an on screen menu (say the Quick Dials option and then closing it clears the button name and header as expected.
[image]
[image]

CLI shows correct clearing of forwarding status
```
-     -- SEP6CFA897380F4: Call Forward disabled on line 98030
    --     -- SEP6CFA897380F4: db clear SCCP/SEP6CFA897380F4/98030
```

**chan-sccp version:**
Skinny Client Control Protocol (SCCP). Release: 4.3.0 develop - 46f48e9M (built by 'root' on '2016-08-28 09:20:03 UTC')

**asterisk version**: (include revision and/or branch if relevant)
Asterisk 13.7.2

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: Not Sure
- Problem can be reliably reproduced, doesn't happen randomly: Yes
- Platform:
Incredible PBX 13-12.2
Kernel Version - 2.6.32-573.12.1.el6.i686 (SMP) i686
Distro Name - Icon Scientific Linux release 6.7 (Carbon)
CPU - Dual Core Intel(R) Celeron(R) 2955U @ 1.40GHz (ChromeBox)

**Phone Firmware**
Image Version : SCCP894x.9-4-2SR3-1
Also happened on Image 9-4-2-SR2-2

dkgroot, 2018-03-01:
@bcdaus 
Sorry that it took some time to get around to looking at this issue. The handling of the skinny protocol on 69xx and 89xx phones differes quite a bit from what we were used to when developing for the 79xx series. The 89xx phones do seem to expect some events to happen in a pre-specified order, which we have not been able to find out (yet). If you do have access to a cucm machine, it would be very helpfull if you could capture some of this information (via tcpdump/wireshark).

dkgroot, 2019-01-09:
I can reproduce the issue now. Have not had time to look into a possible solution just yet

dkgroot, 2019-08-22:
***Git***
_Checkout this temporary branch_
```
git pull
git fetch -a
git branch
git checkout issues/403
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
git checkout develop
```

commit b480f5fa (2019-08-22): Fix callforward:Off display update on 89xx Series phones

---
894x phones have a bit of a quirky interface and we are having some trouble adapting
to them. For example when cancelling a callforward the cfwd status is not removed from
the display.

Daljit Parmar provided a pcap which showed a solution to updating the display after
cancelling call forward on the 8945 phones. In retrospect the soluton is pretty simple
and obvious (but contra to how skinny protocol works on other phones). Just go offhook,
cancel callforward and go back onhook.
---

References #465
Fixes #403
commit bc5fc3b6 (2019-08-23): Fix callforward:Off display update on 89xx Series phones

---
894x phones have a bit of a quirky interface and we are having some trouble adapting
to them. For example when cancelling a callforward the cfwd status is not removed from
the display.

Daljit Parmar provided a pcap which showed a solution to updating the display after
cancelling call forward on the 8945 phones. In retrospect the soluton is pretty simple
and obvious (but contra to how skinny protocol works on other phones). Just go offhook,
cancel callforward and go back onhook.
---

References #465
Fixes #403
closed 2019-08-22 by dkgroot