chan-sccp/chan-sccp #404: Transfer Operation not consistent between technologies

gplindstrom, 2017-05-12:
Initial problem:

When "transfer_on_hangup = yes" is set and placing call to phone, the CONNTRAN softkey set is selected instead of CONNECTED.

Longer Description:

I am attempting to make a mixture of SIP and SCCP phones operate as closely to the same as possible (less training and end-user confusion).  On a commercial service we use (not sure of software... not call manager though, and definitely not Asterisk) that supports Cisco SCCP phones, and the USECALLMANAGER patch for Cisco SIP firmware for Asterisk, the operation seems to proceed (more or less) as:

- CONNECTED softkeys when phone call is received and taken off hook.  Transfer can then be initiated by pressing Transfer softkey.

- While I don't have 69xx phone (I have 79xx and 9971), looking at the code, for the 69xx phones, on the first stage, the KEYMODE_OFFHOOKFEAT is used rather that KEYMODE_DIGITSFOLL, which is what both the commercial service and the USECALLMANAGER patch seem to use.  This may be a bug in chan_sccp I spotted???

-  At this point, the CONNTRAN softkey set is displayed (which may consist of as little as endcall, transfer).  A transfer can be completed by either hanging up the handset/speaker or pressing the transfer softkey.  If the remote party rejects call and hangs up, or we select the remote party and press endcall, the ONHOLD softkey set is presented where we can then resume the call (probably back to CONNECTED) and explain that the person we tried transferring to is unavailable.

The advantage to this is that we can do both, complete a transfer by simply hanging up the handset/speaker, or pressing the transfer softkey.  Additionally, the conntran softkey set is kept simple, with most functions in the connected key set, and operation would match across technologies).  While I have not used callmanager, from reading the docs, I think it probably works this way too, but not sure.

It works mostly as described when transfer on hangup is "no", but of course you lose the ability to simply hangup.  I have given this a shot at changing the code (mostly in sccp_device.c and the sccp_dev_set_keyset function), but only with partial success.  I can't seem to keep the CONNTRAN softkey set during the transfer process.  It keeps changing to ONHOLD or ONHOOK.   Someone who knows the code better than I probably can do this, assuming this would be accepted as a patch/feature upgrade/enhancement :)

Using latest devel release of chan_sccp and Asterisk 13.15.0.

Thanks!

Gary

gplindstrom, 2017-05-16:
OK, think I figured this out.  Still needs some testing, but I think this works.  See patch.  Feel free to modify to match style of developers if desired.  But I would like to see this integrated into devel if possible.  For what seems simple, this took a while as I had to figure out existing code, phone API, etc.  Set "transfer_on_hangup = yes", set softkeys to something like (modify as desired):

onhook=redial,newcall,cfwdall,dnd,pickup,gpickup,private
offhook=redial,endcall,cfwdall,dnd,pickup,gpickup,private
offhookfeat=redial,endcall
digitsfoll=back,endcall,dial
ringout=empty,endcall,transfer
connected=hold,endcall,transfer,idivert,conf,conflist,park,monitor,vidmode
conntrans=empty,endcall,transfer
connconf=hold,endcall,conf,conflist,join,vidmode
onhold=resume,newcall,endcall,idivert
ringin=answer,idivert
onhint=barge,newcall,redial,pickup,gpickup
onstealable=redial,newcall,cfwdall,pickup,gpickup,dnd,intrcpt
holdconf=resume,newcall,endcall,join

Also, this patch includes a modification to sccp_indicate.c to expand the data displayed in the prompt line.  It does have one problem which I have not yet figured out...  if the phone has a message about missed calls, the incoming information is not displayed.  Have not yet figured out how to override that message.  If anyone knows and cares to share, that would be great!  I use the expanded info to show the DID dialed that reached me, the extension that call forwarded to me, etc by setting the REDIRECTING info in the dialplan, and think this additional info is useful to phone users.

Got a couple more things I will work on in the future, but I will open separate issues for them.

[sccp-patch-gpl20170515.patch.txt](https://github.com/chan-sccp/chan-sccp/files/1003428/sccp-patch-gpl20170515.patch.txt)

dkgroot, 2017-08-31:
Hi Gary,

Thank you for the patch, and sorry for my really late reply (I have been out of the loop for a bit, because of some health issues). 

I take it that you have had this patch in production for a while now, without ill effects ? 

I will have to review your patch to check that it does not have any ill effects, if not it will be accepted and merged into the current develop branch.

dkgroot, 2017-08-31:
If you require any information about the sources or require any help with one of your other patches/project please feel free to ask.

dkgroot, 2017-08-31:
In principle (for 79xx devices) it does not really matter what the difference keymodes are called, as long as they make sence to the person setting up the system. The devices don't use these names in any way. 

I am absolutely fine with any of the transfer code and indications sent to the phones. Changing the current keymodes and button assignments is however difficult. Mostly because we allow poeple ot setup their own softkeys sets (which some people have done/used). Using your patch as is, without modifying the the softkeyset on their systems, means that transfer does not work anymore for them. Especially phones with 'hard' transfer keys will have a problem (ie: 8945 / 69xx).

Maybe we can work on your patch together, to see if we can find a middle ground.

Note: our current transfer code is a bit of a mess (really big functions), which i already started to try and cleanup back in april/may of this year, but never got finished. Maybe we can combine these two projects. If we fix transfer in the process, it might be more acceptable to require a changing in softkeysets.

gplindstrom, 2017-08-31:
Yes, the code has been working good for me.  I have been slammed at work, so have not been able to look at stuff much recently.  I have several other things I want to change too.

I will have to go and modify my test environment to see if I can understand the problem you are describing.  I don't have an 8945/68xx to try this with.  I do have 9971 phones but they are SIP only, and 79xx phones (SIP and SCCP).  I thought what I did was basically to fix the transfer process :)  It works much more smoothly for me anyway.

The other goal was/is to make the SIP-only phones and the SCCP phones work as close to the same as possible so that users are not confused when moving from phone to phone.  For me, this was also accomplished.

I will get back when I get a chance to look at this and see what you are referring to...
Gary

dkgroot, 2017-08-31:
Hi Gary,

You might want to have a look at the 'feature/transfer' branch (just updated it). It tries to clarify the way transfers are handled into seperate/smaller functions. It's already working for asterisk-<13, but still needs a little work for the newer asterisk versions. Note this is a work in progress. Would love to work on this together.

Regarding the changes you made. Your version works perfectly fine, with the accompanied softkeyset on 79xx phones. There are some problems with 69xx/89xx phones though, these phones expect certain keymodes only in particular circumstances. They cannot be ordered around like the 79xx's can. This makes it a little bit more tricky to change things. They seem to refuse changing to a particular mode without following a certain order of events (had to look at a number of cucm dumps to see what order they had to be in).

Also we need to take into account that some of our users already have softkeysets defined in their sccp.conf, which would cause problems when we change the definition of the keymodes. If necessary we can change them, but would require an announcement on the mailinglist and potentially helping out users using old definitions. If possible i would like to prevent a change like this.

gplindstrom, 2017-09-04:
Hi Diederik,

Ok, I tried the feature/transfer branch.  For me at least, it has a few problems.  I only have asterisk 13 (13.17.0 with usecallmanager patch, and a patch for notifycid that makes it work with sip phones) so that may be part of the problem (not asterisk-<13).  I also use FreePBX 14 (part of the Sangoma 7.3.1701 distro) but that should not be a factor here.  I also have a mixture of sccp and sip phones that I want to work as closely the same as possible to minimize confusion with my users when they move from phone to phone.  I would prefer to use chan_sccp simply due to the fact that I have a enhanced/fixed version of sccp manager (yes, it works with the newer versions of php) that I and/or my staff can use to do all the configuration of the phones without using the bash shell.  That is not the case with the usecallmanager patch, but I still need/use it for those phones that are sip or sip-only as the case may be now.

The most glaring issue with this release is the transfer on hangup seems completely broken.  I haven't tracked it down, yet, but I think the transfer is just being completely cancelled when the phone is hungup.  Does seem like there is some callerid issues, but have not paid attention to them yet.  Additionally, same problem with focus on the transfer line is lost after dialing the extension to transfer to, requiring the arrow keys to move back to the transfer.  I have included a patch again for that part.

I would like to work with you to get this addressed.  I don't want to isolate myself on an old version of chan_sccp unless I re-patch newer versions.  I may or may-not be a lot of help because my last serious programming was in the 1990's (using Pascal, Cobol, and a few other old languages), but I can make due when necessary with C and PHP.  I also don't have any 69xx/89xx phones at this point to test with...  All our phones are Cisco 79XX, some Cisco 99xx, a bunch of Aastra 67xx phones, and the occasional odd phone like a SNOM, Grandstream, and some softphones.

So, I guess the first thing we would need to decide if my goal to make the sccp and sip phones operate similarly is reasonable even if there is some incompatibility in configuration and past user operation.   If so, this may require modifying the default keyset internally to chan_sccp to make all functions work as desired.  Don't know what percentage of people defined custom keysets, but it might require some modifications on their part.  Perhaps a big warning notice in a readme file :)  I have done upgrades in the past that required modifications to configurations, so it would not be unheard of.   Maybe there is another solution?

One of the other glaring inconsistencies is the pickup/gpickup feature.  According to https://www.cisco.com/en/US/docs/voice_ip_comm/cucm/admin/3_1_2/ccmsys/a07cpick.html pickup should automatically dial the group and get the call, and gpickup prompts for number to pickup.  However, chan_sccp seems to reverse this function and instead gpickup automatically picks up the call, and pickup asks for the number to be dialed.   For a simple (but not correct) fix, I swapped the functions in sccp_softkeys.c  This works, but the C code documents just the opposite.  If this is modified, it is a change incompatible with old versions (user experience) unless perhaps a config option is added (runtime or compile-time), but it matches how Cisco Call Manager and the asterisk usecallmanager patch function.
[gpl-transfer-patch.txt](https://github.com/chan-sccp/chan-sccp/files/1273358/gpl-transfer-patch.txt)

Anyway, thoughts, comments, ideas?  I will do what I can though, within the limitations of my current skills and work obligations.

Gary
[gpl-transfer-patch.txt](https://github.com/chan-sccp/chan-sccp/files/1273361/gpl-transfer-patch.txt)

dkgroot, 2017-09-04:
Hi Gary,

The feature/trasnfer branch is just in it's infancy, currently callinfo (display and phonebook) is a bit of a mess when using this version. Have not looked at the transfer on hangup for this either. I just started out rewriting the whole transfer procedure, to get cleaner code, it is not anywhere near finished though.

Would you mind sharing your version of SCCP Manager with the rest of chan-sccp users ? If you made any changes, maybe you could send a Pull Request to the original developer.

In regard to softkey keymodes, I think it should be possible to come up with something compatible, without having to chnage the softkeysets in any great way. The keymodes might not end up being named the same, but just giving the same user experience. Would you be ok with that. If so, maybe we should start by making a little state diagram (mapping keymodes used by chan_sccp to the ones used by usecallmanager etc). So instead of renaming the keymodes we just need to figure out, under which circumstance which one is used. That way we (can hopefull) preserve historic compatibility. The phones don't care what we call a keymode, they are sent as a number (index) to the phones. Maybe the usecallmanager names for the keymodes make more sense, but in principal the names don't really matter during programming or users using them.

I didn't know that pickup/gpickup were reversed when using callmanager. Name wise that does not make to much sense to me :-) BTW: We never attempted emulating callmanager in any way (We did not even look at callmanager functionality when implementing chan_sccp). If i am not mistaken chan_skinny uses the same functionality as we do (pickup is not implemented, and gpickup just uses the first callgroup that matches the pickupgroup). I am a little apprehensive switching the behaviour mid version. Maybe we should add a sccp.conf flag called 'emulate_callmanager=true/false'. That way we could implement your requested functionality, without changing behaviour for other users. This could also be used for the keymodes if necessary.

I don't care to much about your lesser c experience, it doesn't really matter in chan_sccp in regards to changing user-behaviour functionality like this :-)

dkgroot, 2017-09-04:
Hi Gary,
Added the 'emulate_callmanager' boolean flag to the global section. Default: off
Incorporated most of your patch into the feature/transfer branch.

Would you be able to provide a little schema showing the different keymodes used by 'usecallmanager' during which events (not sure if the list below is complete)

Even                       |  usecallmanager | callmanager | chan_sccp
------------------------ | --------------------- | ---------------- | ---------------
Dialing                    | | |
Ringing                   | | |
Connected              | | |
Hold | | |
Resume | | |
Started Transfer (Transfering) | | |
Completing Transfer | | |
After Transfer | | |
Hangup | | |

Thanks !

gplindstrom, 2017-09-17:
[chan_sccp_transfer_table.xlsx](https://github.com/chan-sccp/chan-sccp/files/1309429/chan_sccp_transfer_table.xlsx)

Hi Diederik,

Sorry this took so long to get done.  Like I said, been slammed at work but grabbed some time this weekend between things.  I changed your table a bit to make it more like the call flow.  Actually, a good old fashion flow chart showing events/states/flow/etc would be good but I don't have anything to draw that with other than pencil and paper.  Anyway, I think my table more-or less does that.  Hopefully this helps some.  Note I don't actually have Cisco call manager to test with, but if I had to guess from what I have read, it would operate like usecallmanager does.  If you need more info, or you need me to test a specific scenario I did not test, or it does not make sense and you need clarification, or just questions in general, let me know.

Gary

gplindstrom, 2017-09-17:
Hmm...  just looked at table that was uploaded.  Export from LibreOffice seems to have messed up row height...  but I am sure you can adjust it to make it look better :)
Gary

gplindstrom, 2017-09-17:
Correction:  On the first two rows (Incoming-Call or Call-in-progress and Answer if incoming-call), the chan_sccp column is NOT the same as usecallmanager.  'Connected Transfer' keyset is displayed instead of 'Connected'.  I discovered this because I tried the latest transfer version, and it did not have the "Transfer" key which is the default for "Connected", making it hard to do a transfer :)  So the state is correct, but the keyset will need to be adjusted to reflect the new correct state.

[chan_sccp_transfer_table.xlsx](https://github.com/chan-sccp/chan-sccp/files/1309462/chan_sccp_transfer_table.xlsx)