chan-sccp/chan-sccp #405: directrtp=on - no audio

dandrzejewski, 2017-05-13:
This may be a configuration issue, but I can't get audio between two 7971s with directrtp=on.

**Reproduction Steps:**

1. set directrtp=on
2. make a call between two phones

**Expected behavior:**
Both callers should be able to hear each other.

**Observed behavior:**
Neither caller can hear each other.

**Screenshots**

[2017-05-13 19:33:22] DEBUG[3224]: chan_iax2.c:2715 peercnt_remove: ip callno count decremented to 4 for 159.203.160.218
pbx*CLI>
[2017-05-13 19:33:22] DEBUG[3224]: chan_iax2.c:2715 peercnt_remove: ip callno count decremented to 3 for 159.203.160.218
[2017-05-13 19:33:22] DEBUG[3126]: manager.c:6327 process_message: Running action 'Ping'
    -- SEP0015626974AF: Got stimulus=BusyLampField Speeddial (21) for instance=5, callreference=0, status=1
    -- SEP0015626974AF: Speeddial Button (5) pressed, configured number is (50200)
    -- SEP0015626974AF: Getting the active channel on device.
    -- SEP0015626974AF: No active channel on device.
SEP0015626974AF: (getEmptyChannel) on line:5000, maybe_c:
SEP0015626974AF: (getEmptyChannel) getActiveChannel
    -- SEP0015626974AF: Getting the active channel on device.
    -- SEP0015626974AF: No active channel on device.
    -- SEP0015626974AF: Set the active channel 10 on device
    -- 0001: New channel number: 10 on line 5000
    -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 5000,
    -- SCCP:              cid_num: "5000"
    -- SCCP:             cid_name: "David Andrzejewski"
    -- SCCP:          accountcode: "79011"
    -- SCCP:                exten: ""
    -- SCCP:              context: "from-internal-xfer"
    -- SCCP:             amaflags: "0"
    -- SCCP:            chan/call: "SCCP/5000-0000000A"
    -- SCCP:combined capabilities: "[slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, g729b, g729ab, g729, g729a]"
    -- SCCP:  reduced preferences: "[ulaw/64k]"
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/5000-0000000A
[2017-05-13 19:33:25] DEBUG[3833][C-00000004]: channel.c:5689 set_format: Channel SCCP/5000-0000000A setting read format path: slin16 -> slin16
[2017-05-13 19:33:25] DEBUG[3833][C-00000004]: channel.c:5689 set_format: Channel SCCP/5000-0000000A setting write format path: slin16 -> slin16
[2017-05-13 19:33:25] DEBUG[3833][C-00000004]: channel.c:5689 set_format: Channel SCCP/5000-0000000A setting read format path: ulaw -> ulaw
    -- SCCP/5000-0000000A: (recalculateReadformat)
        capabilities: [slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, g729b, g729ab, g729, g729a]
        preferences: [ulaw/64k]
        remote caps: []
        READ:[ulaw/64k]
[2017-05-13 19:33:25] DEBUG[3833][C-00000004]: channel.c:5689 set_format: Channel SCCP/5000-0000000A setting write format path: ulaw -> ulaw
    -- SCCP/5000-0000000A: (recalculateWriteformat)
        capabilities: [slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, g729b, g729ab, g729, g729a]
        preferences: [ulaw/64k]
        remote caps: []
        Write:[ulaw/64k]
    -- 0001: Allocated asterisk channel SCCP/5000-0000000A
    -- SEP0015626974AF: Set asterisk state OffHook (2) for call 10
    -- SEP0015626974AF: Finish to indicate channel state:SPEEDDIAL on call:SCCP/5000-0000000A, lineInstance:1. New channel state:OFFHOOK
    -- SEP0015626974AF: (sccp_pbx_softswitch) New call on line 5000
    -- SEP0015626974AF: (sccp_pbx_softswitch) Dial Extension 50200
    -- SEP0015626974AF: Send the dialed number:50200, callid:10, lineInstance:1
    -- SEP0015626974AF: Finish to indicate channel state:DIALING on call:SCCP/5000-0000000A, lineInstance:1. New channel state:DIALING
    -- SCCP: (sccp_pbx_softswitch) set variable SKINNY_PRIVATE to: 0
    -- SCCP: (sccp_pbx_softswitch) set variable SKINNY_PRIVATE to: 0
    -- +- pbx extension matcher (50200          ): ---+
    -- |ignore     |exists     |can match  |match more|
    -- | no        |yes        |yes        |yes       |
    -- +----------------------------------------------+
 SEP0015626974AF: (sccp_pbx_softswitch) channel SCCP/5000-0000000A is dialing number 50200
    -- SEP0015626974AF: Set asterisk state Ring (4) for call 10
[2017-05-13 19:33:25] DEBUG[3833][C-00000004]: channel.c:5689 set_format: Channel SCCP/5000-0000000A setting write format path: slin -> ulaw
[... 215 lines cut ...]
[2017-05-13 19:33:50] DEBUG[3274]: chan_sip.c:3397 initialize_initreq: Initializing initreq for method OPTIONS - callid 086d327b34964c266b1e0ccc664947ce@10.1.50.3:5060
[2017-05-13 19:33:50] DEBUG[3274]: chan_sip.c:4538 __sip_ack: Stopping retransmission on '086d327b34964c266b1e0ccc664947ce@10.1.50.3:5060' of Request 102: Match Found
[2017-05-13 19:33:50] DEBUG[3261]: chan_iax2.c:3083 sched_delay_remove: schedule decrement of callno used for 104.131.98.27 in 60 seconds
[2017-05-13 19:33:52] DEBUG[3274]: chan_sip.c:9016 __sip_alloc: Allocating new SIP dialog for 27d9c9b500a3d9c77b0be70868138980@10.1.50.3:5060 - OPTIONS (No RTP)
[2017-05-13 19:33:52] DEBUG[3274]: chan_sip.c:8801 change_callid_pvt: SIP call-id changed from '27d9c9b500a3d9c77b0be70868138980@10.1.50.3:5060' to '1d32e21849211b2c612efe7d21a7537f@10.1.50.3:5060'
[2017-05-13 19:33:52] DEBUG[3274]: chan_sip.c:3397 initialize_initreq: Initializing initreq for method OPTIONS - callid 1d32e21849211b2c612efe7d21a7537f@10.1.50.3:5060
[2017-05-13 19:33:52] DEBUG[3274]: chan_sip.c:4538 __sip_ack: Stopping retransmission on '1d32e21849211b2c612efe7d21a7537f@10.1.50.3:5060' of Request 102: Match Found
    -- SCCP: Accepted Client Connection from 10.1.50.249:52446
    -- SCCP: Connected on server via 10.1.50.3:60386
[2017-05-13 19:33:52] DEBUG[3265]: chan_iax2.c:2683 peercnt_add: ip callno count incremented to 3 for 208.43.234.226
[2017-05-13 19:33:52] DEBUG[3266]: chan_iax2.c:3083 sched_delay_remove: schedule decrement of callno used for 208.43.234.226 in 60 seconds
[2017-05-13 19:33:52] DEBUG[3126]: manager.c:6327 process_message: Running action 'Ping'
pbx*CLI>

**chan-sccp version:** (include revision and/or branch if relevant)

pbx*CLI> sccp show version
Skinny Client Control Protocol (SCCP). Release: 4.3.0 develop - a2c0640M (built by 'root' on '2017-05-03 00:28:36 UTC')

**asterisk version:** (include revision and/or branch if relevant)

pbx*CLI> core show version
Asterisk 13.15.0 built by root @ raspberrypi on a armv7l running Linux on 2017-05-02 21:35:04 UTC

**Addition / Optional Information:**

dkgroot, 2017-09-24:
@dandrzejewski 
Would you mind checking the 'issues/394' branch

Related Issues:
#394 
#391 
#405

Once this branch (and Pull Request #416) has been check it can be merged back into the develop tree.

dkgroot, 2017-10-15:
@dandrzejewski
I Think i fixed directrtp (again) :-)

Pushed another set of commites (cd1eac7)
Can you give it another try
(with "sccp debug core, channel, pbx, indicate, rtp")

I made one major change in Signalling: Which does cause somewhat altered behaviour.

When the Phone Receiving the incoming call starts ringing, it will send PROGRESS() to asterisk. So that the other side (the Dialing Party), can start setting up RTP. That way we have the ip-address:port combination a little earlier on, before bridging starts.

We do need to test this rigorously, to make sure it does not have negative effects (like with NAT / in-directrtp / HOLD / RESUME / Transfer / etc).

dkgroot, 2017-10-19:
Create a new branch called 'issues/394_2'

Please check if this fixes the issue you reported.

nickvsh, 2019-11-07:
Hi. 
I have:

- Asterisk certified/16.3 
- Skinny Client Control Protocol (SCCP). Release: 4.3.2 (HEAD - 558d2b6M)

As far as I can tell DirectRTP between SCCP endpoints on asterisk (7912/7940/7960) working correctly.

But there is No DirectRTP in call between SCCP and PJSIP endpoint. 
RTP flow goes through the PBX for each call leg.

Is it possible at all, in case of different signaling protocols?
I suppose, "yes", theoretically... cause media uses RTP protocol which is independent from signaling. 

 According to the man-page https://github.com/chan-sccp/chan-sccp/wiki/DirectRTP
> When calling a sip device the same rules apply and the sccp:permit will be checked against the sip:directmediapermit to see if directrtp (directmedia) will be allowed.

But there is no such option **directmediapermit** in **chan_pjsip**. 
Maybe you can advise something?

Thanks!

**My settings:**

**Global pjsip settings:**

```
hq*CLI> pjsip show settings 

Global Settings:

 ParameterName                              : ParameterValue
 
 contact_expiration_check_interval          : 30
 debug                                      : no
 default_from_user                          : asterisk
 default_outbound_endpoint                  : default_outbound_endpoint
 default_realm                              : asterisk
 default_voicemail_extension                : 
 disable_multi_domain                       : false
 endpoint_identifier_order                  : ip,username,anonymous
 ignore_uri_user_options                    : false
 keep_alive_interval                        : 90
 max_forwards                               : 70
 max_initial_qualify_time                   : 0
 mwi_disable_initial_unsolicited            : false
 mwi_tps_queue_high                         : 500
 mwi_tps_queue_low                          : -1
 regcontext                                 : 
 send_contact_status_on_update_registration : no
 taskprocessor_overload_trigger             : global
 unidentified_request_count                 : 5
 unidentified_request_period                : 5
 unidentified_request_prune_interval        : 30
 use_callerid_contact                       : no
 user_agent                                 : Asterisk PBX certified/16.3

System Settings:

 ParameterName               : ParameterValue

 accept_multiple_sdp_answers : false
 compact_headers             : false
 disable_tcp_switch          : true
 follow_early_media_fork     : true
 threadpool_auto_increment   : 5
 threadpool_idle_timeout     : 60
 threadpool_initial_size     : 0
 threadpool_max_size         : 50
 timer_b                     : 32000
[... 254 lines cut ...]
Videosupport?                                  : no
Direct RTP                                     : on
Trust phone ip (deprecated)                    : off
Bind Address                                   : 172.16.*.11:51223
Server Address                                 : 192.168.*.241:2000
Deny/Permit                                    : deny:0.0.0.0/0.0.0.0,permit:127.0.0.0/255.0.0.0,permit:10.0.0.0/255.0.0.0,permit:172.0.0.0/255.224.0.0,permit:192.168.0.0/255.255.0.0,
PermitHosts                                    :  
Early RTP                                      : Progress
Accessory State                                : Headset:Off Hook
Last dialed number                             : 
Default line instance                          : 1
Custom Background Image                        : ---
Custom Ring Tone                               : ---
Use Placed Calls                               : off
PendingUpdate                                  : off
PendingDelete                                  : off
allow_conference                               : on
conf_play_general_announce                     : on
conf_play_part_announce                        : on
conf_mute_on_entry                             : off
conf_music_on_hold_class                       : default
conf_show_conflist                             : on
conflist_active                                : off
CallHistory Answered Elsewhere                 : Ignore
```

dkgroot, 2019-11-07:
> As far as I can tell DirectRTP between SCCP endpoints on asterisk (7912/7940/7960) working correctly.

Ooh nice... then fixing hold/resume should not be a whole lot of work !

> Is it possible at all, in case of different signaling protocols?

Yes it is possible. Both sides of the channel need to line up, regaring selected codec (DTMF etc).
And:
- sccp.conf device's:deny/permit must line/match up with
- sip.conf directmediadeny/directmediapermit  
When they line up, we allow direcrtp between both endpoints (this has been tested with chan_sip and chan_sccp).

 To be honest i am not sure what these are called in pjsip.conf. It might be that they do the same thing as chan_sccp and re-use the endpoints deny/permit.

dkgroot, 2019-11-07:
Can you do a small test of  'issues/394_2' branch to see if it is still works (and potentially fix the hold/resume issue already) ?

***Git***
_Checkout this temporary branch_
```
git pull
git fetch -a
git branch
git checkout issues/394_2
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
git checkout develop
```

nickvsh, 2019-11-11:
> Can you do a small test of 'issues/394_2' branch to see if it is still works (and potentially fix the hold/resume issue already) ?

Recently I tested "issues/394_2 - 558d2b6M " branch.

```
Connected to Asterisk certified/16.3 currently running on hq (pid = 11068)
hq*CLI> sccp show version 
Skinny Client Control Protocol (SCCP). Release: 4.3.2 issues/394_2 - 558d2b6M (built by 'voip' on '2019-11-08 15:06:18 UTC')
```
The results of this test are posted [here](https://github.com/chan-sccp/chan-sccp/issues/394#issuecomment-552459573), cause I think this is more appropriate place for this issue (Hold/Resume related troubles)