chan-sccp/chan-sccp #407: Ringtype parameter not working

NEAX2000, 2017-06-27:
The Parameter "ringtype" under the [general] section in sccp.conf is not working

When I do an SCCP reload I get:

debian1*CLI> sccp reload
SCCP reloading configuration. 0x7f0514000f40
  == Parsing '/etc/asterisk/sccp.conf': Found
[Jun 27 17:32:19] NOTICE[6426]: sccp_config.c:2664 sccp_config_getConfig: Config file 'sccp.conf' loaded.
[Jun 27 17:32:19] WARNING[6426]: sccp_config.c:384 sccp_config_object_setValue: SCCP: Unknown param at general:12:ringtype='inside'

Asterisk version 13.16
SCCP version 4.3

sccp.conf file:

; general section
;
[general]
servername = Asterisk
keepalive = 60
context = internal
dateformat = M.D.Y
bindaddr = my cm ip goes here
port = 2000
deny = 0.0.0.0/0.0.0.0
permit = internal
ringtype = inside
hotline_enabled = yes
hotline_extension = 111
hotline_label = hotline
;
;
;device section

The Parameter "ringtype" under the '[general]' section in sccp.conf is not working

I am attempting to follow the documentation here found here: https://github.com/chan-sccp/chan-sccp/wiki/Ringer-RingType-Alert_Info

gplindstrom, 2017-07-19:
I am the one that precipitated the addition of this parameter back in January 2017, and it works fine, but it won't be in the 4.3 branch.  Diederik was good enough to take what I submitted, make it better, and add it to the development branch.  You need to use the development branch.  In my opinion, this is the one to use anyway as it has A LOT of fixes!

marcelloceschia, 2017-07-20:
@NEAX2000 can you clone the master branch and us the master branch?

NEAX2000, 2017-08-21:
Hey everyone,

Got this working using the latest development branch.  You are correct, this release does appear to have a lot of fixes!

Thank you!

closed 2017-08-21 by NEAX2000