chan-sccp/chan-sccp #408: Can't register multiple remote extensions at same time

bryanhiller, 2017-06-28:
I have been able to successfully provision and register multiple remote phones with sccp, but i am unable to get more than one from the same remote site to register simultaneously.

**Reproduction Steps:**

1. conifgure and update firmware on 2 cisco 7961Gs at remote location behind nat.  
2. each phone registers when it is the only phone on the network
3. error on phone says Registration Rejected: Cross device ....

**Expected behavior:**

More than one phone should be able to register

**Observed behavior:**

only 1 phone at a time is able to register

**Screenshots**
###### My phone system sits on public WAN with whitelisted IP access only. (see attached conf files)

**chan-sccp version:** (include revision and/or branch if relevant)
chan-sccp-b-6836
**asterisk version:** (include revision and/or branch if relevant)
ateisk 11.19.0

[image]

[sccp.txt](https://github.com/chan-sccp/chan-sccp/files/1109153/sccp.txt)
[sccp_devices.txt](https://github.com/chan-sccp/chan-sccp/files/1109155/sccp_devices.txt)
[sccp_extensions.txt](https://github.com/chan-sccp/chan-sccp/files/1109154/sccp_extensions.txt)

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Yes/No]
- Problem can be reliably reproduced, doesn't happen randomly: [Yes/No]
- Platform:
  - [CPU]
  - [32/64 Bit]
  - [Big/Small Endian]
  - [Operating System]
  - [Operating System Version]
- Direct Contact information: [You can contact me here]
- Log Files (attach .txt / [pastbin](http://pastebin.com/) / [gist](http://gist.github.com/)): [url]
- Comments: [Other Information]

dkgroot, 2017-07-30:
please do update/upgrade you chan-sccp version (6836 is an svn revision number, we moved to git )
If you set 
```
Debug= core, device,action
Nat=auto
Externip=[external IP address of you may firewall]
```
 in sccp.conf. This will also increase the debug log level a little, so that you can see the registration information for the device. During registration it should show information about the automatic Nat detection  and state that it recognised that device is behind a Nat firewall.

dkgroot, 2019-02-22:
@bcharkansas 
Zero reply about this issue report since opening date. Closing this issue

closed 2019-02-22 by dkgroot