chan-sccp/chan-sccp #410: Bad Session?

jcroy, 2017-08-15:
**Problem**
Cisco 79XX phones stop working.  No dial tone and no inbound calls.  Looking at the asterisk side of things:

`[2017-08-15 09:39:06] WARNING[7714]: sccp_session.c:1064 sccp_session_check_crossdevice: Session(0x7f40f00280f0) and Device Session(0x7f40f001ece0) are of sync.
[2017-08-15 09:39:06] WARNING[7714]: sccp_actions.c:820 handle_register: SEP102BD53474BF: Cleaning previous session, come back later, state:OK
  == 10.x.x.x:59187: Previous session 0x7f40f001ef10 needs to be cleaned up and killed!`  

We "fixed" it by restarting the server but this is not optimal in the long run as we cant be restarting a production server whenever this problem occurs.    

**Reproduction Steps:**
No one was active or changing anything in the server, everything was running the way it normally should.  It was running fine without any issues and then out of no were this happened.  

**Observed behavior:**
Phones stopped working, as in no calling ability outbound or inbound.  

**chan-sccp version:**
Release: 4.3.0 develop 

**asterisk version:**
Asterisk 13.16.0

dkgroot, 2017-08-31:
Hi jcroy,

Can you please try the issues/410 branch:
```bash
git checkout issues/410
make
make install
```

To see if that fixes your issue. The biggest issue with testing this, is waiting for it to happen again :-(

AccessDG, 2018-06-25:
I'm having the same issues I think... I compiled and loaded the 410 branch but am still getting the same errors...

[2018-06-24 20:22:58] WARNING[5484]: sccp_session.c:1062 sccp_session_check_crossdevice: Session(0x7f00680240e0) and Device Session(0x7f00680235a0) are out of sync.
[2018-06-24 20:22:58] WARNING[5484]: sccp_actions.c:825 handle_register: SEP001B0CDB0D0F: Cleaning previous session, come back later, state:OK

I can restart asterisk and it stops for a short time and then starts again.. It affects 2 phones but the other 15 phones are not affected.

PhantomVl, 2018-06-25:
In the current version of sccp, after you reboot asterisk, you must reset the token on the wireless phones. 
>   sccp tokenack [SEPxxxxx]

AccessDG, 2018-06-25:
These phones are 7941's and 7961's...

AccessDG, 2018-06-25:
When I tried to use tokenack I got this error:

> sccp tokenack SEP001B0CDB0D0F
> SCCP CLI ERROR: SEP001B0CDB0D0F: We need to have received a token request before we can acknowledge it
> Command 'sccp tokenack SEP001B0CDB0D0F' failed.
> [2018-06-25 04:12:35] WARNING[21618]: sccp_cli.c:3377 sccp_tokenack: SEP001B0CDB0D0F: We need to have received a token request before we can acknowledge it

Maybe its because the phones arnt wireless.... egardless, it stil doesnt fix my issue

PhantomVl, 2018-06-25:
This solution is only for 792x phones....