chan-sccp/chan-sccp #411: No transfer when connection is established via TAPI

jonasius, 2017-09-08:
Hey there,

I use PhoneSuite TAPI for Asterisk to establish phone calls. The phone rings and after you pick up the phone asterisk dials. The problem is, that these calls can't be transferred. After pressing the transfer button the Cisco 7960 displays `Button not active` and the asterisk log shows `... Transfer request from line channel...`.

**chan-sccp version:** Release: 4.3.0 develop - a2c0640M
**asterisk version:** Asterisk 11.24.1

LogFile: [gist](https://gist.github.com/jonasius/fec26261ce4e75ee767ffca7daedcfbb)

dkgroot, 2017-09-08:
@jonasius 
I think we need a little more detailed logging, to see what is going on.
Can you set: ```sccp debug core, channel, line, device```
So we can see why the transfer is not being performed. 

BTW: I am working on a new transfer implementation (not finished yet), maybe you can give that one a try, just to see if it already fixes the issue above. Note: callinfo in the branch is not yet handled (well/at all). To give it a try: ```git checkout feature/branch```
To get back to the standard develop branch after testing, use: ```git checkout develop```

BTW: did you notice the externip warning. If you have a NAT firewall between your device and the pbx, you need to set sccp.conf externip  to the outside ip-address of the firewall. This is not related but might cause other preventable audio issues.

jonasius, 2017-09-09:
@dkgroot 
Thanks for your reply. Problem is, when i set `sccp debug core, channel, line, device` before establishing the call via TAPI, Asterisk is crashing.

At the moment I could only test with an Cisco 7975 but it shows the same behavior as the 7960.
In this log i activated debug mode after introduced to dial via TAPI. [Logfile](https://gist.github.com/jonasius/b73d8e6ff623b491997f8f62f7d5a4af)

I will try the new implementation and will give you feedback.

The problem with the extern ip is that the internet connection doesn't have an fixed outside ip-address. Is it possible to set an hostname with dynamic ip instead of ip-address?

dkgroot, 2017-09-09:
If you get a segfault/crash, we do need a gdb backtrace (see our wiki) so we can fix the cause of the segfault. If you are on an older version, then please upgrade to the latest develop branch and check again, before reporting the issue. Without the debugging or backtrace, it's hard for us to be able to help.

Regarding externip, you can old use externhost to set the name instead, it will be updated to the latest ip-address before the value is used. When you do see warnings about your config, you can check the sccp.conf.annotated to see all possible sccp.conf parameters. Most config parsing warnings will try to give you a clue how to fix the issue. If you do run into one, that is less helpfull, left us know.

miguelsanzpardo, 2017-09-10:
Hi Jonas,

When you use TAPI how is generated the call? Using a .call file? I remember
that in the past using call files with chan sccp produced asterisk crashing.

regards

jonasius, 2017-09-11:
Hi Miguel,

At this moment I can't tell you. What can I do to find out how if it uses .call files? The TAPI program connects to asterisk via the "Asterisk Call Management support" - configured with manager.conf.
To clarify, it only crashes when I enable debug-mode. Normally it works, but only without the possibility to transfer and put calls on hold.
Later I'll do further test.

regards

dkgroot, 2017-09-11:
Logically using talk should not make use of .call files (would be very impractical). More likely they hook up to the asterisk manager interface. Via the manager interface it can initiate an 'originate' just like a .call file would. So the end result should/would be the same, just no files involved. I am still not sure what cause issues when using originate, I do remember looking into the issue a while back when Miguel reported issues regarding .call files, not sure what the findings were to be honest.

Regards,

Diederik

jonasius, 2017-09-12:
@dkgroot 
With **chan-sccp version:** 4.3.0 origin/feature/transfer - 8087108M and **asterisk version:** 11.24.1
 keeps crashing if I set `sccp debug core, channel, line, device` and I initiate the call via TAPI.

I will try to do a gdb backtrace tomorrow. I know that I need to compile chan-sccp with --enable-debug and --disable-optimization. To compile asterisk I think I need to select something with debug in make menuselect?! 

Nevertheless I tried to put the TAPI initiated call on hold/transfer it in feature/transfer branch. Short Log of Errors below.

`[Sep 11 22:33:33] ERROR[15052]: sccp_channel.c:1231 sccp_channel_getEmptyChannel: SEP002699Exxxxx: Putting Active Channel SCCP/28-00000001 OnHold failed -> Cancelling new CaLL
[Sep 11 22:33:33] ERROR[15052]: sccp_channel.c:1276 sccp_channel_newcall: SEP002699Exxxxx: Can't allocate SCCP channel for line 28
[Sep 11 22:33:33] WARNING[15052]: sccp_channel.c:1917 _transfer_failed: SEP002699Exxxxx: Unable to create channel for transfer`

dkgroot, 2017-09-12:
@jonasius / @miguelsanzpardo 

**Ran a couple of command line tests**
asterisk-13 : originate (ie: channel originate sccp/xxxxx extension xxxx@default) 
asterisk-13 : originate + local channel (ie: channel originate local/xxxxx extension xxxx@default)
asterisk-11 : originate (ie: channel originate sccp/xxxxx extension xxxx@default) -> caused segfault during originate : Fix commited
asterisk-11 : originate + local channel (ie: channel originate local/xxxxx extension xxxx@default)

Once the call is established i transferred the call (both on the originator and called extension). All tests went ok (Can you run the same command on the commandline, to see if our test results are the same).

I did get a segfault when originating a new call using sccp/xxxx on asterisk-11, for which a fix has been committed (to both branches). Please pull the latest changes before testing again

I think i need to know who to replicate the behaviour that you are seeing. What does the originate command look like used by TAPI and/or .call file.

**BackTrace:**
You can try generating the backtrace without recompiling asterisk to see if it already generates worthwhile information (100% sccp bug in that case), If an asterisk recompile is required we can do that later. If you installed asterisk from rpm/deb you just need to install the asterisk-debuginfo package (no need to recompile).

jonasius, 2017-09-12:
@dkgroot 

**Command line test:**
With asterisk-11 and the fix asterisk doesn't crash.

With the fix you commited I was able to get a log with `sccp debug core, channel, line, device` enabled. See [here](https://gist.github.com/jonasius/8b16b922dad7affb767fbab9968d03d0).

It's nearly the same as I try to put a call initiated with `channel originate sccp/28 extension 01xxxxxxxxx@extern` on hold or transfer it. It's neither possible. 

If I initiate a (local) call with `channel originate local/900@voicemail extension 28@sccp` it's possible to put it on hold.

P.S.: I know that I need to fix the externip warning, but as I set externhost to my hostname callers wasn't able to hear me. As described in the wiki I need to define which devices are allowed talk to each other direct.

dkgroot, 2017-09-12:
@jonasius 
Good to here that the initial fix at least prevented the segfault.
Which side of the orignated call do you put on hold / transfer. The originating side or the destination ?
You can switch of directrtp when working on this issue. We have to review directrtp issues at a later stage.

Can you setup tapi to use a local channel just to see if that solves your problems for now. We still need to figure out why the sccp/xxxx causes trouble.

jonasius, 2017-09-12:
@dkgroot 
Before missunderstandings occur:

* Wrong direction: `channel originate local/01734349805@extern extension 28@sccp` -> hold/transfer on sccp phone works
* Right direction: `channel originate local/28@sccp extension 01734349805@extern` -> hold/transfer on sccp phone doesn't work

It seems that the local channel isn't the solution.

dkgroot, 2017-09-12:
@jonasius 
I reviewed your debug log a little more, and noticed that the originated channel ends up in state progress instead if connected. This is caused by the soemwhat unorthodox way an originate channel comes to live. I will have another look at the sources tomorrow, to see what needs to be changed to fix this issue (being carefull not to cause side effects for normal call setup). For testing I will have to create a separate branch based on this issue report. Will let you know when I have something for you to try.

miguelsanzpardo, 2017-09-13:
Nice to know that the segfault which appeared when yo use Originate(sccp/)
has been fixed :)

closed 2017-09-13 by dkgroot

dkgroot, 2017-09-13:
@jonasius / @miguelsanzpardo 
Apparently using Originate() with a SIP desination, can under circumstance (FreePBX) cuause it to send a PROGRESS indication after already having indicated CONNECTED (which is an ordering of events we did not expect). Made a change to the source to skip re-indication of Progress when we have already received a Connected indication). We do re-indicate Connected in this case, just to make sure the RTP stream is open.

Issues should be resolved in 'develop' and 'feature/transfer' branches. Please verify.

jonasius, 2017-09-13:
@dkgroot 
I've updated the system to the latest develop commit and it fixes the issue. Great work!

dkgroot, 2017-09-13:
@jonasius
Thanks for the feedback and reporting the issue !
One more problem resolved !
Viel Spass und Erfolg

commit 4b7ec6f5 (2017-09-13): Fix: ORIGINATE() to SIP does indicate PROGRESS after CONNECTED, causing trouble when the channel needs to be put on HOLD

Fixes #411