chan-sccp/chan-sccp #412: Cisco 7960 constantly boot-loops

philpem, 2017-09-13:
When connecting a 7960 to an Asterisk server which hasn't had the phone configured in sccp_hardware.conf, the phone continuously reboots.

Hot

There is a sequence of traffic between the phone and chan_sccp_b, ending with a SoftKeyTemplateRes packet.

I have a TCPdump log.

Confirmed with phone firmware P0030801SR02.

**Reproduction Steps:**

1. Connect a phone to the network, factory reset it. XML file should contain the callManager group and load information but nothing further. The phone's MAC address should not be 
2. Wait for the phone to boot -- it will reach the main app, spend a long time configuring the VLAN, then begin talking to the Asterisk server before suddenly rebooting after "Loading CM list".

The state of hotline_enabled seems to make little difference.

**Expected behavior:**

Phone to connect to chan_sccp_b, but show as Unprovisioned.

**Observed behavior:**

Phone continuously reboots.

Nothing relevant in the log, other than a note that the phone had been removed from the pool due to SCCP timing out.

**chan-sccp version:** Git 355def259314f1628431c2f2a87c4bb1120bd6f9

**asterisk version:** 13.16.0 

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: Have not tested any other version of Chan-SCCP
- Problem can be reliably reproduced, doesn't happen randomly: Yes
- Platform:
  - Intel Core Quad Q6600
  - 64 bit
  - Little endian
  - FreePBX 13.0.192.16
- Direct Contact information: philpem at gmail dot com

philpem, 2017-09-13:
[skinny_bootloop.pcap.gz](https://github.com/chan-sccp/chan-sccp/files/1300279/skinny_bootloop.pcap.gz)

7960 XML:
```
<device>
        <versionStamp>{7960 Sep 11 2017 19:20:00}</versionStamp>
        <devicePool>
                <callManagerGroup>
                        <members>
                                <member  priority="0">
                                        <callManager>
                                                <ports>
                                                        <ethernetPhonePort>2000</ethernetPhonePort>
                                                </ports>
                                                <processNodeName>10.0.0.26</processNodeName>
                                        </callManager>
                                </member>
                        </members>
                </callManagerGroup>
        </devicePool>
        <logServer>10.0.0.32</logServer>
        <loadInformation>P0030801SR02</loadInformation>
</device>
```

XMLDefault.xml only contains LoadInformation lines and blank auth, idle, information, messages, proxy server, services and directory URLs.

Configuration files are as per the FreePBX examples provided with chan-sccp-b.

dkgroot, 2017-09-13:
Hi Phil,
You can increase the sccp loglevel a little, to get more information, I suggest ```sccp debug core, socket, action, device```. Please do attach your sccp.conf (global section) and the gathered debug logging for this device. 
I think the device might be outside the global deny/permit range, which would exclude it from making a connection to the server. If it is allowed then a hotline e should be assigned. I did make a small change to the register code last week, so I will check your findings tomorrow.
If necessary the phone also has a debug log, which you can download via it's Webinterface, it might hold a clue.

philpem, 2017-09-13:
I haven't got the web interface enabled - and the phone reboots so fast, getting in is hard. It's rebooting to the Universal Loader, so whatever logs it has are gone.

I'll try to turn the debugging on and hook the phone back up.

Here's my sccp.conf:
```
;=========================================================================================
;
; general definitions
;
;=========================================================================================
[general]
servername = FreePbxSCCP
keepalive = 60
debug = 1
context = from-internal-xfer
dateformat = D.M.Y
bindaddr = 0.0.0.0
port = 2000
disallow=all
allow=alaw
allow=ulaw
firstdigittimeout = 16
digittimeout = 8
autoanswer_ring_time = 1
autoanswer_tone = 0x32
remotehangup_tone = 0x32
transfer_tone = 0
callwaiting_tone = 0x2d
musicclass=default
language=en
deny=0.0.0.0/0.0.0.0
permit=10.1.0.0/255.255.0.0
dndFeature = on
sccp_tos = 0x68
sccp_cos = 4
audio_tos = 0xB8
audio_cos = 6
video_tos = 0x88
video_cos = 5
echocancel = on
silencesuppression = off
private = on
callanswerorder=oldestfirst
hotline_enabled=yes                                     ; can devices without configuration register
hotline_context=default                                 ; context for hotline
hotline_extension=111                                   ; extension will be dialed on offHook

;=========================================================================================
;
; include files
;
;=========================================================================================
#include sccp_hardware.conf
#include sccp_extensions.conf

```

dkgroot, 2017-09-13:
You coukd add two more none existing member servers to the xmldefault to get a little more time if necessary 😀
I do know that hotline worked using the sources from 2-3 weeks ago, so you could role back a little to verify. Currently not near a PC so I can't check for you right now. Sccp debug logging should shed a bit of light on what is happening.

dkgroot, 2017-09-14:
sccp.conf
```
debug=core,device,action,socket
...
hotline_enabled=yes
hotline_context=default
hotline_extension=500
hotline_label=myhotline
```
ASTCLI:
```
*CLI> sccp restart SEPE0D173E11D9
    -- SCCP: Accept Connection
    -- SCCP: Accepted Client Connection from [::ffff:192.168.178.111]:35126
    -- SCCP: Connected on server via [::ffff:192.168.178.3]:49005
       > [::ffff:192.168.178.3]:49005: keepalive:62, keepaliveinterval:63
    -- <?xml version="1.0" encoding="UTF-8"?><x-cisco-alarm><Alarm Name="LastOutOfServiceInformation"><ParameterList><String name="DeviceName">SEPe0d173e11d95</String><String name="DeviceIPv4Address">192.168.178.111/24</String><String name="IPv4DefaultGateway">192.168.178.2</String><String name="DeviceIPv6Address"></String><String name="IPv6DefaultGateway"></String><String name="ModelNumber">CP-8945</String><String name="NeighborIPv6Address"></String><String name="NeighborIPv4Address">192.168.178.226</String><String name="NeighborDeviceID">SEP381C1AB6B976</String><String name="NeighborPortID">Port 1</String><Enum name="DHCPv4Status">1</Enum><Enum name="DHCPv6Status"></Enum><Enum name="TFTPCfgStatus">1</Enum><Enum name="DNSStatusUnifiedCM1">4</Enum><Enum name="DNSStatusUnifiedCM2">0</Enum><Enum name="DNSStatusUnifiedCM3">4</Enum><String name="VoiceVLAN">4095</String><String name="UnifiedCMIPAddress">192.168.178.3</String><String name="LocalPort">10377</String><String name="TimeStamp">1505374186</String><Enum name="ReasonForOutOfService">23</Enum><String name="LastProtocolEventSent">0x27</String><String name="LastProtocolEventReceived">0x0</String></ParameterList></Alarm></x-cisco-alarm>
    -- <?xml version="1.0" encoding="UTF-8"?>
    -- <x-cisco-alarm>
    -- <Alarm Name="DeviceTLInfo">
  == Alarm Type: DeviceTLInfo
    -- <ParameterList>
    -- <String name="DeviceName">SEPe0d173e11d95</String>
    -- <String name="IPv4Address">192.168.178.111</String>
    -- <String name="IPv6Address"></String>
    -- <String name="CTL_Signature">Not Installed</String>
    -- <String name="CTL_TFTP_Server">N/A</String>
    -- <String name="ITL_Signature">Not Installed</String>
    -- <String name="ITL_TFTP_Server">N/A</String>
    -- <Enum name="StatusCode">6</Enum>
    -- </ParameterList>
    -- </Alarm>
    -- </x-cisco-alarm>
 SEPE0D173E11D95: is registering, Instance: 1, UserId: 0, Type: Cisco 8945 (585), Version: 22 (loadinfo 'SCCP894x.9-4-2SR1-2')
    -- SCCP: Device 'SEPE0D173E11D95' not found in realtime table 'sccpdevice'
    -- SCCP: Create Device
    -- SEPE0D173E11D95: Device State is 'On Hook'
    -- Init MessageStack
  == buttonconfig status before check
  == buttonconfig status after check
  == buttonconfig status after adding new buttons
    -- Added device 'SEPE0D173E11D95' to Glob(devices)
    -- SEPE0D173E11D95: Allocating device to session (27) ::ffff:192.168.178.111
    -- SEPE0D173E11D95: Our Session IP4 Address 192.168.178.111:0
    -- SCCP: searching for our capability for device protocol version 22
    -- SCCP: searching for our capability for device protocol SCCP
    -- SCCP: found protocol version '22' at 22
    -- SEPE0D173E11D95: (sccp_dev_set_registered) Setting Registered Status for Device from None to Progress
    -- SEPE0D173E11D95: Registration State is 'Progress'
       > SEPE0D173E11D95: keepalive:62, keepaliveinterval:61
    -- SEPE0D173E11D95: Accessory 'Headset' is 'On Hook' (0)
    -- SEPE0D173E11D95: Clear the display notify message
    -- SEPE0D173E11D95: Clear the status prompt on line 0 and callid 0
    -- SEPE0D173E11D95: ( sccp_device.c:1994:void sccp_dev_check_displayprompt(const sccp_device_t *const) ) sccp_dev_displayprompt '�' for line 0 (0)
    -- SEPE0D173E11D95: Display prompt on line 0, callid 0, timeout 0
    -- SEPE0D173E11D95: Accessory 'Headset' is 'On Hook'
    -- SEPE0D173E11D95: Clear the display notify message
    -- SEPE0D173E11D95: Clear the status prompt on line 0 and callid 0
    -- SEPE0D173E11D95: ( sccp_device.c:1994:void sccp_dev_check_displayprompt(const sccp_device_t *const) ) sccp_dev_displayprompt '�' for line 0 (0)
    -- SEPE0D173E11D95: Display prompt on line 0, callid 0, timeout 0
    -- SEPE0D173E11D95: Accessory 'Handset' is 'On Hook'
    -- SEPE0D173E11D95: Sending servers message (SEPE0D173E11D95)
    -- SEPE0D173E11D95: Building button template Cisco 8945(585), user config 
    -- SEPE0D173E11D95: Allocated '14' buttons.
    -- Added extension 'Hotline' priority 1 to dundiextens
    -- SEPE0D173E11D95: Sending ConfigStatMessage, lines 1, speeddials 0
    -- SEPE0D173E11D95: Device has 12 Capabilities
    -- SEPE0D173E11D95: searching for softkeyset: default!
    -- SEPE0D173E11D95: ONHOOK          => |0 :Redial   |1 :NewCall  |2 :CFwdALL  |3 :DND      |4 :PickUp   |
    -- SEPE0D173E11D95: CONNECTED       => |0 :Hold     |1 :EndCall  |2 :Park     |3 :Video Mode|4 :Select   |5 :CFwdALL  |6 :CFwdBusy |8 :Record   |
[... 35 lines cut ...]
    -- SEPE0D173E11D95: enable video mode softkey
    -- SEPE0D173E11D95: Device has 1 Video Capabilities
    -- SEPE0D173E11D95:         level: 43
    -- SEPE0D173E11D95:         profile: 64
    -- SEPE0D173E11D95:         Codec has 3 levelPreferences:
    -- SEPE0D173E11D95:         0:     transmitPreference: 50
    -- SEPE0D173E11D95:                format: 2: qcif (176x144)
    -- SEPE0D173E11D95:                maxBitRate: 20000
    -- SEPE0D173E11D95:                minBitRate: 100
    -- SEPE0D173E11D95:                MPI: 1
[... cut ...]
    -- SEPE0D173E11D95:                MPI: 1
    -- SEPE0D173E11D95:                serviceNumber: 0
    -- SEPE0D173E11D95:         2:     transmitPreference: 52
    -- SEPE0D173E11D95:                format: 4: 4cif (704x576)
    -- SEPE0D173E11D95:                maxBitRate: 20000
    -- SEPE0D173E11D95:                minBitRate: 100
    -- SEPE0D173E11D95:                MPI: 1
    -- SEPE0D173E11D95:                serviceNumber: 0
    -- SEPE0D173E11D95: Accessory 'Headset' is 'Off Hook'
```

dkgroot, 2017-09-14:
I hooked up my 7960 andi can reproduce the issue. Will check what happened/changed and fix the issue.

dkgroot, 2017-09-15:
Fix committed in revision: 688bc3b4

closed 2017-09-15 by dkgroot

philpem, 2017-09-16:
Confirmed fixed! Thank you!

(Now I just need to figure out why my SIP phones can't dial the SCCP extensions... probably got something configured wrong in FreePBX...)

dkgroot, 2017-09-16:
Hi Phil,
Thanks for reporting back !

Can't really help you with setting up FreePBX (I don't use FreePBX myself). You can ask the mailinglist if necessary. I do know that you need to add a 'Custom Extension' for the sccpline that you added. When you add a custom extension it sometimes get's added to the astdb with a FreePBX 'DND setting' active (I have seen reports about this on our maillinglist in the past). This would cause it to not be reachable. You can have a look at the astdb using 'database show' in the Ast CLI. Also do have a look at the debug output in the CLI when dialing this extension, it should show you the reason.

This Issue report is now closed.

commit 688bc3b4 (2017-09-15): Fix: Anonymous/hotline causing device to reboot during registration

Fixes #412