chan-sccp/chan-sccp #414: Hard hold key on SPA30x and SPA50x

patrickkeys, 2017-09-22:
When pressing the hardware hold button on an SPA30x or SPA50x phone, it only ever works on line 1. If a hold softkey is defined, this works perfectly for all lines. When attempting to use the hardware hold button on other lines, it is effectively ignored, however the logs do indicate the cause of the problem:

```
[Sep 22 10:12:52] NOTICE[5285] sccp_session.c:     -- SEPC472956702FF: (session_dissect_msg) Incoming message is smaller(24) 
than known size(28).
[Sep 22 10:12:52] NOTICE[5285] sccp_actions.c:     -- SEPC472956702FF: >> Got message Stimulus Message (0x5)
[Sep 22 10:12:52] NOTICE[5285] sccp_actions.c: SEPC472956702FF: SCCP Handle Message: Stimulus Message(0x0005) 24 bytes length
[Sep 22 10:12:52] NOTICE[5285] sccp_line.c:     -- SEPC472956702FF: Looking for line with instance 1.
[Sep 22 10:12:52] NOTICE[5285] sccp_line.c:     -- SCCP: Found line 2005
[Sep 22 10:12:52] NOTICE[5285] sccp_channel.c:     -- SCCP: Looking for channel by state '30'
[Sep 22 10:12:52] NOTICE[5285] sccp_channel.c:     -- 0001: Could not find active channel with state CONNECTED(30) on line
[Sep 22 10:12:52] NOTICE[5285] sccp_channel.c:     -- SCCP: Looking for channel by state '32'
[Sep 22 10:12:52] NOTICE[5285] sccp_channel.c:     -- 0001: Could not find active channel with state HOLD(32) on line
```

In sccp_protocol.h, the comments for StimulusMessage.lel_stimulusInstance suggest that this field is normally set to '1' except for speed dial and line. However, in sccp_actions.c:handle_stimulus(), this instance value is used to determine the current line. Comments suggest that there is an outstanding 'todo' to use the CallReference field, but for now, the issue can be resolved by resetting the instance to 0 if the stimulus is a hold key. This in turn forces the function to determine the instance by active line instead. I know the following is a horrible bodge, but it works:

```
diff --git a/src/sccp_actions.c b/src/sccp_actions.c
index d003fd3..6899a0b 100644
--- a/src/sccp_actions.c
+++ b/src/sccp_actions.c
@@ -2360,6 +2360,9 @@ void handle_stimulus(constSessionPtr s, devicePtr d, constMessagePtr msg_in)
        if(!instance && stimulus == SKINNY_STIMULUS_LASTNUMBERREDIAL && d->redialInformation.lineInstance > 0) {
                instance = d->redialInformation.lineInstance;
        }
+       if(instance && stimulus == SKINNY_STIMULUS_HOLD) {
+               instance = 0;
+       }
        if (!instance) {                                                                                        /*! \todo also use the callReference if available */
                sccp_log((DEBUGCAT_CORE)) (VERBOSE_PREFIX_3 "%s: Instance 0 is not a valid instance. Trying the active line %d\n", d->id, instance);
                if ((l = sccp_dev_getActiveLine(d))) {
```

This has been tested on git snapshots a2c06409 (Apr 8 2017) and 688bc3b4 (Sep 15 2017) and works fine.

dkgroot, 2017-09-22:
Thanks for reporting this issue. I rarely use hardkeys, when the softkeys are available, so this one escaped me. Thanks a lot for doing the research on this (makes my life a lot easier this way).

I do see the issue, apparently the SPA phones send the wrong information via the instance variable. In comparison the 8945 also has a hard HOLD key, and it does send the correct line number via the instance variable.
Because some of the other phones with hard hold keys, do present the correct information, we need to check for the device type before making any decisions. When we cannot use the instance information, we should use the callreference information instead (which does seem to be correct in both cases).
Working on a solution. Will create a seperate branch 'issues/414' for testing.

dkgroot, 2017-09-22:
Hi Patrick,
Created a seperate branch 'issues/414' for you to test with. This should solve the issue nicely (or as nice as i can make it). I do hate the way different devices tend implement the skinny protocol differently, causing us to make exceptions like these :-(
When you have done your tests (maybe you have other devices with hard buttons for hold (or other stimuly), just give me a shout, so the branch can be merged back into develop.

dkgroot, 2017-09-22:
BTW you detailed analysis and preliminary patch really helped a lot. Making turn around on this issue really short ;-)

dkgroot, 2017-09-22:
Replaced your patch with:
```C
        /* SPA phones always send instance=1 instead then the hard hold button is pressed. using callid to find the correct line instead */
        if (stimulus == SKINNY_STIMULUS_HOLD && sccp_session_getProtocol(s) == SPCP_PROTOCOL) {
                AUTO_RELEASE(sccp_channel_t, c, sccp_channel_find_byid(callId));
                if (c) {
                        l = sccp_line_retain(c->line);
                        for (instance = SCCP_FIRST_LINEINSTANCE; instance < d->lineButtons.size; instance++) {
                                if (d->lineButtons.instance[instance] && d->lineButtons.instance[instance]->line == l) {
                                        break;
                                }
                        }
                }
        }
```
We check if we are dealing with a SPA device (which use the SPCP protocol)
It first looks for the channel by callid, when found, it set's the line for later use. 
It also looks up the instance of the line on the device, so that we can update the instance variable to what we had expected to receive. Just in case any later functions depend on the instance variable (not the case at this moment, but doesn't do any harm either)

closed 2017-09-22 by patrickkeys

dkgroot, 2017-09-22:
Branch has been merged back into develop. Thanks for checking the patch and testing it.

Looking forward to future issues with attached solutions 😃

PR #415 (2017-09-22): Fix: SPA phones send instance=1 when hard hold key is pressed
Regardless of the active line (Issue reported by patrickkeys)

Fixes #414
commit 7f1b848e (2017-09-22): Fix: SPA phones send instance=1 when hard hold key is pressed, regardless of the active line (Issue reported by patrickkeys) (#415)

Fixes #414