chan-sccp/chan-sccp #418: Call Park Dropping calls

AccessDG, 2017-10-24:
When someone tries to pickup a call that is parked instead of connecting the call is dropped.  The caller who was listening to on hold music just hears a click and then nothing.  The person that is trying to pick up the call just has dead air (silence).  The problem is sporadic and I havent been able to replicate it on demand yet.  If there's anything I can do to try to get this problem solved let me know.  We are using asterisk 13.17.2 and chan_sccp v4.3

Jonathan

dkgroot, 2017-10-24:
@AccessDG 
Sporadic issues are nasty and difficult to fix. Is there anythjng in the asterisk full log, that would indicate what is going on ? To perform a pickup from park, we actually do nothing special, you enter the parkinglot extension and we dial it. Most likely this might be an asterisk issue and not chan-sccp related. Have you checked the asterisk changelogs after 13.17.2 to see if this may have been fixed already upstream ?

Can you increase the sccp debug log to 'core, channel, pbx' so that we might pickup some information.

dkgroot, 2019-02-22:
@AccessDG 
No reply since opening this bug report. Closing the issue.
If the issue is still there please reopen

closed 2019-02-22 by dkgroot