chan-sccp/chan-sccp #420: SIP -> Local -> SCCP one way audio becouse of wrong codec after eb61206

pnlarsson, 2017-12-12:
Call SIP with G722 -> Local -> SCCP gives oneway audio -> asterisk is sending G722 to the phone and are getting G711a back (should be G711a in both directions)

**Reproduction Steps:**

Install revision eb61206
Make a call with local channel from a sip phone with (PJSIP) allow: (g722|alaw|ulaw|opus|gsm|g726|g729) and with g722 as first priority in the invite.
Asterisk is then sending G722 to the 7965 and no sound

**Expected behavior:**

Before eb61206 there was sound in both directions

**chan-sccp version:** develope eb61206 and later

**asterisk version:** 13.17

dkgroot, 2017-12-22:
Hi Niklas,
Have been a little busy on a different project the last couple of days (sorry). And now all this Christmas stuff is preventing me to make some progress on this. I will have to schedule a nice debug session next week, to figure out what might be going on.

Merry Christmas in advance to you and your Family !

milaq, 2018-02-24:
**Asterisk 13.14.1**

Can confirm transcoding doesn't get applied after c461215b30b8c02f9c96ab620b319d6e426e6250 when G722 is prioritized and calling on a alaw/ulaw line.

This _can_ be worked around by putting the call on hold and resuming but obviously this is not intended.
`ReadFormat` then falls back to `alaw`.

Before hold with c461215b30b8c02f9c96ab620b319d6e426e6250 (called party can hear me, I cannot be heard):
```
State: Up (6)
NativeFormats: (g722|alaw|ulaw|g729)
WriteFormat: alaw
ReadFormat: g722
WriteTranscode: No 
ReadTranscode: No 
```

After hold with c461215b30b8c02f9c96ab620b319d6e426e6250 (both audio channels fine):
```
State: Up (6)
NativeFormats: (g722|alaw|ulaw|g729)
WriteFormat: alaw
ReadFormat: alaw
WriteTranscode: No 
ReadTranscode: No 
```

Intended behaviour before c461215b30b8c02f9c96ab620b319d6e426e6250:
```
State: Up (6)
NativeFormats: (g722)
WriteFormat: slin16
ReadFormat: g722
WriteTranscode: Yes (slin@16000)->(g722@16000)
ReadTranscode: No
```

dkgroot, 2018-02-28:
@pnlarsson I finally got/made some time to look into this.

To me both before and after revision c461215 looks wrong. Before c461215 it was doing a partial (one leg) transcoding (juck but explainable). After c461215 it does not do transcoding but still has the one legged/partial codec selection.

I just tested this with chan_sip which does not seem to be showing this, and both legs seem to be kept in sync.

I need to switch to pjsip and see if i can reproduce the issue, with that.

dkgroot, 2018-03-08:
Hi Niklas,
Would you mind checking that branch 'issues/420' reverts the c461215 changes correctly, and brings the state back to the 'working' situation, for now.
```
git fetch -a
git checkout issues/420
make && make install && make reload
```
I tried fixing the issues reported but did not find an easy route forward from the current situation. Better to first revert this c461215 change and then move forward from there.
Regards,
Diederik

pnlarsson, 2018-03-08:
Hi,
Tested with issues/420 and it works as it should call between two phones one with alaw the other with ulaw:
Got  RTP packet from    83.218.94.20:31872 (type 08, seq 040418, ts 114073429, len 000160)
Sent RTP packet to      83.218.94.20:31800 (type 00, seq 001235, ts 114073424, len 000160)
Got  RTP packet from    83.218.94.20:31800 (type 00, seq 058584, ts 1131239189, len 000160)
Sent RTP packet to      83.218.94.20:31872 (type 08, seq 030890, ts 1131239184, len 000160)

Latest develop (sound was ok but wrong rtp, using other codecs makes one-way audio):
Got  RTP packet from    83.218.94.20:26458 (type 08, seq 040563, ts 2800199412, len 000160)
Sent RTP packet to      83.218.94.20:26384 (type 00, seq 028745, ts 2800199408, len 000160)
Got  RTP packet from    83.218.94.20:26384 (type 00, seq 058723, ts 2818384357, len 000160)
Sent RTP packet to      83.218.94.20:26458 (type **00**, seq 030170, ts 2818384352, len 000160)

dkgroot, 2018-03-08:
Thanks Niklas, will merge this into the develop repo.
First step will be to fix the conference with more that 3 participants issue. And then see if we can fix the codec selection issue.
Thanks a lot for helping out and your patience !

dkgroot, 2018-03-15:
@pnlarsson 
I reopened the 'issues/420' branch, and pushed another commit (reinstating most of the previously faulty c461215 changes. I found the actual cause where i accidentally used '1' instead of 'i' in the setNativeAudioFormats function.
Would you mind testing/checking this again, before i push it ?

noppel, 2018-03-15:
It works for me now 👍

pnlarsson, 2018-03-15:
Nope, same as before: 

uc01*CLI> sccp show version 
Skinny Client Control Protocol (SCCP). Release: 4.3.1 issues/420 - f92fa5dM (built by 'root' on '2018-03-15 13:15:59 UTC')

uc01*CLI> core show version 
Asterisk 13.19.1 built by root @ uc01.nguc.weblink.se on a x86_64 running Linux on 2018-02-16 09:09:48 UTC

Log with sccp set debug rtp,codec,core:

   -- SEPC8F9F9688E82: (sccp_channel_answer) Answer Channel SCCP/wid23026-00000004
    -- SEPC8F9F9688E82: Answer channel SCCP/wid23026-00000004
    -- SEPC8F9F9688E82: (sccp_channel_answer) current preferredCodec=2
    -- SEPC8F9F9688E82: (sccp_channel_answer) Update Channel Capability
    -- SEPC8F9F9688E82: recalculateReadformat
    -- no more preferences at position 3
    -- did not find joint capability with remote device, using first joint capability 2(alaw/64k)
    -- SCCP/wid23026-00000004: (recalculateReadformat) 
	capabilities: [slin16, ulaw/64k, alaw/64k, ilbc, g729b, g729ab, g729, g729a]
	preferences: [alaw/64k, alaw/64k, alaw/56k]
	remote caps: [ulaw/64k]
	READ:[alaw/64k]
    -- SEPC8F9F9688E82: recalculateWriteformat
    -- no more preferences at position 3
    -- did not find joint capability with remote device, using first joint capability 2(alaw/64k)
    -- SCCP/wid23026-00000004: (recalculateWriteformat) 
	capabilities: [slin16, ulaw/64k, alaw/64k, ilbc, g729b, g729ab, g729, g729a]
	preferences: [alaw/64k, alaw/64k, alaw/56k]
	remote caps: [ulaw/64k]
	Write:[alaw/64k]
    -- SEPC8F9F9688E82: (sccp_channel_answer) Set Called Party
    -- SEPC8F9F9688E82: Answering channel with state 'Ringing' (5)
    -- SEPC8F9F9688E82: channel SCCP/wid23026-00000004 payloadType 2
    -- SEPC8F9F9688E82: Ask the device to open a RTP port on channel SCCP/wid23026-00000004. Codec: G.711 A-law 64k, echocancel: ON
    -- SCCP/wid23026-00000004 connected line has changed. Saving it until answer for SCCP/wid23025-00000003
    -- SCCP/wid23026-00000004: rtp server instance created at 0.0.0.0:25302
  == Using SCCP RTP TOS bits 184
  == Using SCCP RTP CoS mark 6
    -- SCCP/wid23026-00000004: (createRTPServer) updated phone Audio RTP destination to : A.A.A.246:25302, family:IPv4, mapped: False
    -- SEPC8F9F9688E82: Started RTP on channel SCCP/wid23026-00000004
    -- SEPC8F9F9688E82: Open receive channel with format G.711 A-law 64k[2], payload 2, echocancel: YES, passthrupartyid: 4294967291, callid: 4
    -- SEPC8F9F9688E82: checkNat us: A.A.A.246:35197, usFamily: IPv4
    -- SEPC8F9F9688E82: checkNat remote: A.A.A.246:25302, remoteFamily: IPv4
    -- SCCP: No externip/externhost set in sccp.conf.
When you are running your PBX on a seperate host behind a NAT-TING Firewall you need to set externip/externhost.
  == SEPC8F9F9688E82: no externip/externhost set, falling back to using incoming interface address:A.A.A.246:35197
    -- SEPC8F9F9688E82: (updateNatRemotePhone) new remote: A.A.A.246:25302, new remoteFamily: IPv4
    -- SEPC8F9F9688E82: Answered channel SCCP/wid23026-00000004 on line wid23026
    -- SEPC8F9F9688E82: Got OpenChannel ACK. Status:'Media Status: OK' (0), Remote RTP/UDP:'192.168.50.139:31282', Type:Indirect RTP, PassThruPartyId:4294967291, CallID:4
       > 0x7f6a5036ded0 -- Strict RTP learning after remote address set to: B.B.B.20:31282
    -- SCCP: (asterisk113_setPhoneRTPAddress) Update PBX to send RTP/UDP media to 'B.B.B.20:31282' (new remote) (NAT: yes)
    -- SEPC8F9F9688E82: Tell PBX   to send RTP/UDP media from A.A.A.246:25302 to B.B.B.20:31282 (NAT:192.168.50.139:31282)
    -- SEPC8F9F9688E82: Opened Receive Channel (State: CONNECTED[30])
    -- SEPC8F9F9688E82: Starting Phone RTP/UDP Transmission (State: CONNECTED[30])
    -- SEPC8F9F9688E82: checkNat us: A.A.A.246:35197, usFamily: IPv4
    -- SEPC8F9F9688E82: checkNat remote: A.A.A.246:25302, remoteFamily: IPv4
    -- SCCP: No externip/externhost set in sccp.conf.
When you are running your PBX on a seperate host behind a NAT-TING Firewall you need to set externip/externhost.
  == SEPC8F9F9688E82: no externip/externhost set, falling back to using incoming interface address:A.A.A.246:35197
    -- SEPC8F9F9688E82: (updateNatRemotePhone) new remote: A.A.A.246:25302, new remoteFamily: IPv4
    -- SEPC8F9F9688E82: Tell Phone to send RTP/UDP media from B.B.B.20:31282 to A.A.A.246:25302 (NAT: (Auto)On)
    -- SEPC8F9F9688E82: Using codec: G.711 A-law 64k(2), TOS 184, Silence Suppression: OFF for call with PassThruId: 4294967291 and CallID: 4
    -- SCCP/wid23026-00000004 answered SCCP/wid23025-00000003
  == Spawn extension (profile-dial-real, 4816, 5) exited non-zero on 'Local/0-25-Local*wid13694-MEX@profile-dial-0000012e;2'
    -- SEPC8F9F9688E82: Got startMediaTransmission ACK. Status:'Media Status: OK' (0), Remote RTP/UDP:'192.168.50.139:31282', Type:Indirect RTP, PassThruPartyId:4294967291, CallID:4, CallID1:4
    -- SEPC8F9F9688E82: Media Transmission Started (State: CONNECTED[30])
    -- SEP1C1D86C41453: (sccp_pbx_answer) Outgoing call SCCP/wid23025-00000003 has been answered by remote party
    -- SEP1C1D86C41453: recalculateReadformat
    -- no more preferences at position 1
    -- did not find joint capability with remote device, using first joint capability 4(ulaw/64k)
[... 102 lines cut ...]
    -- SCCP: destroying PBX rtp server on channel SCCP/wid23025-00000003
  == MixMonitor close filestream (mixed)
    -- QualityStats: MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;ICR=0.0000;CCR=0.0000;ICRmx=0.0000;CS=0;SCS=0;MLQKvr=0.95
    -- SEP1C1D86C41453: Call Statistics:
[... cut ...]
         Mean Statistics  : #Calls: 2 Packets sent: 143 rcvd: 133 lost: 0 jitter: 0 latency: 0
         Mean Quality     : MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;MLQKvr=0.95|ICR=0.0018;CCR=0.0018;ICRmx=0.0036|CS=0;SCS=0
       ]
  == End MixMonitor Recording SCCP/wid23025-00000003

uc01*CLI> sccp show version 
Skinny Client Control Protocol (SCCP). Release: 4.3.1 issues/420 - f92fa5dM (built by 'root' on '2018-03-15 13:15:59 UTC')

uc01*CLI> core show version 
Asterisk 13.19.1 built by root @ uc01.nguc.weblink.se on a x86_64 running Linux on 2018-02-16 09:09:48 UTC

dkgroot, 2018-03-15:
Based on @Noppel test i already merged it :-(

@pnlarsson 
I will quickly move back to asterisk 13 to see if i can reproduce the issue you are having (had some difficulty reproducing it before).
Would you be able to take the changes i made in this branch apart, to see what is causing the issue on your system.

TofuJoe, 2018-03-21:
Asterisk 11.21.2 with latest chan-sccp dev branch built today and have a same/similar issue. Extension recv audio but appears server is not expecting to transcode and dropping incoming frames.
Could not find the issues/420 to test with.

Attached cli log with sccp set debug rtp,codec,core
[log.txt](https://github.com/chan-sccp/chan-sccp/files/1834571/log.txt)

dkgroot, 2018-03-21:
Created a new branch 'issues/420_1'

- Reverted f92fa5d/c461215 (again)
- Slowly working toward a better solution for Local Channel codec selection (Small incremental improvements / commits).
- @pnlarsson: Can reprocuce the issue now (I overlooked that my sccp.conf contained earlyrtp=dialout, which is very different from earlyrtp=progress), this however did give me a potential solution.

@pnlarsson @TofuJoe @noppel 
Please try out the new 'issues/420_1' branch. I am making small incremental commits / improvements. If you do run into an issue, do have a look at the `git log` to see if an earlier commit fixes the issue (ie: i created a regression).

To try:
```
git pull
git fetch -a
git checkout 'issues/420_1'
make -j2 && make install & make reload
```

TofuJoe, 2018-03-23:
@dkgroot The joys of working with legacy systems. Thank you for looking into this, I will revisit once we transition to 13.

dkgroot, 2018-03-23:
@TofuJoe: Thanks for the update. 
In the second call I would have hoped that ulaw was choose => no transcoding required. I am still trying to figure out where this is failing.

TofuJoe, 2018-03-30:
@dkgroot
- Probably a side-effect of running A13 with chan_sip and not pjsip, will redo with pjsip, see below.
- Sorry if that was confusing, I simply meant that the call was between two extensions on the same server and not an outbound trunk.

Just to clarify - SCCP Codecs preference
```
[g722/64k, g722/56k, g722/48k, ulaw/64k, ulaw/56k, g729, g729a, g729b, g729ab, g729/annex/b]
```

SCCP > PJSIP
```
Name: PJSIP/mobile-00000054
  NativeFormats: (speex)
    WriteFormat: slin16
     ReadFormat: slin16
 WriteTranscode: Yes (slin@16000)->(slin@8000)->(speex@8000)
  ReadTranscode: Yes (speex@8000)->(slin@8000)->(slin@16000)

Name: SCCP/desk-0000003A
  NativeFormats: (g722)
    WriteFormat: slin16
     ReadFormat: slin16
 WriteTranscode: Yes (slin@16000)->(g722@16000)
  ReadTranscode: Yes (g722@16000)->(slin@16000)
```

SCCP > PJSIP Trunk
```
Name: SCCP/desk-0000003E
  NativeFormats: (g722)
    WriteFormat: ulaw
     ReadFormat: g722
 WriteTranscode: Yes (ulaw@8000)->(slin@8000)->(g722@16000)
  ReadTranscode: No
  
Name: PJSIP/trunk-00000056
  NativeFormats: (ulaw)
    WriteFormat: g722
     ReadFormat: ulaw
 WriteTranscode: Yes (g722@16000)->(slin@8000)->(ulaw@8000)
  ReadTranscode: No
```

mpaland, 2018-05-02:
I can confirm, that the issues/420_1 branch fixes my described problem.
Very good work!

dkgroot, 2018-12-23:
Sorry it took so long to get this merged.

closed 2018-12-23 by dkgroot

mpaland, 2019-01-11:
Hi @dkgroot ,
is/was this 420_1 branch already/really merged?
I'm using the develop version from Jan 07 and got one way audio problems again. 😞

dkgroot, 2019-01-11:
@mpaland Actually 420_2 was merged (in revision d0708a6) on jan 1 + some corrections afterward.
I have been doing quite a number of inbound/outbound calls both sccp-sccp and sip-sccp, and have not noticed one way audio.
If you are on the current revision and you still see the the issue, it might be a good idea to start a new issue (and link back to this one by mentioning it). That way we get a fresh start. It might be something unrelated to the issue above.

commit 9f56c5b0 (2018-12-23): Re-enable setting tos/cos on rtp stream

fixes #420
references #437
references #308