chan-sccp/chan-sccp #421: bridge_p2p_rtp_write: Unsupported payload type received, on AMI Originated call (ast15)

alessandrocarminati, 2018-01-03:
I'm an old user of chan SCPP. Recently I updated my system, a big update actually, I switched from asterisk 11 to the asterisk 15.
The service once used to work, webcall,  now does not work anymore.
The service were pretty simple, thorough AMI I generated a call which were later joined with another also generated.
Now after the 2 calls are joined, no audio is audible anymore, and in its place I can see on the asterisk main console:
res_rtp_asterisk.c:5370 bridge_p2p_rtp_write: Unsupported payload type received
hundreds of lines like 
[Jan  3 12:22:48] DEBUG[29960][C-00000007]: res_rtp_asterisk.c:5370 bridge_p2p_rtp_write: Unsupported payload type received
Got  RTP packet from    10.1.5.57:31786 (type 08, seq 007294, ts 1113872, len 000160)
Sent RTP packet to      10.1.5.56:18678 (type 10, seq 011414, ts 1113872, len 000320)
[Jan  3 12:22:48] DEBUG[29959][C-00000006]: res_rtp_asterisk.c:5370 bridge_p2p_rtp_write: Unsupported payload type received
Got  RTP packet from    10.1.5.56:18678 (type 08, seq 011937, ts 180222800, len 000160)
Sent RTP packet to      10.1.5.57:31786 (type 10, seq 024979, ts 180222800, len 000320)
[Jan  3 12:22:48] DEBUG[29960][C-00000007]: res_rtp_asterisk.c:5370 bridge_p2p_rtp_write: Unsupported payload type received
Got  RTP packet from    10.1.5.57:31786 (type 08, seq 007295, ts 1114032, len 000160)
Sent RTP packet to      10.1.5.56:18678 (type 10, seq 011415, ts 1114032, len 000320)
[Jan  3 12:22:48] DEBUG[29959][C-00000006]: res_rtp_asterisk.c:5370 bridge_p2p_rtp_write: Unsupported payload type received
Got  RTP packet from    10.1.5.56:18678 (type 08, seq 011938, ts 180222960, len 000160)
Sent RTP packet to      10.1.5.57:31786 (type 10, seq 024980, ts 180222960, len 000320)
[Jan  3 12:22:48] DEBUG[29960][C-00000007]: res_rtp_asterisk.c:5370 bridge_p2p_rtp_write: Unsupported payload type received
Got  RTP packet from    10.1.5.57:31786 (type 08, seq 007296, ts 1114192, len 000160)
Sent RTP packet to      10.1.5.56:18678 (type 10, seq 011416, ts 1114192, len 000320)

alessandrocarminati, 2018-01-03:
Don't know if this issue depends on chan sccp, or something new in the asterisk 15. For the moment I resolved my problem specifing the codec in the AMI request. It seems in fact that if not specified, asterisk starts the call using the slin codec. While previous in version of the asterisk, the core performed transcode, in this late version it seems it doesn't. The previous code generated a call where a party used ulaw codec, while the other used slin. It is interesting to know if the previous was the intended behaviour, or this last is more coherent. My problem is solved.

closed 2018-01-03 by alessandrocarminati

dkgroot, 2018-01-04:
Sorry i was a little too late to help out
I do think that automatic translation should have happened, and the issue you raised might be a good example to help out with two of the other open issue reports, which might be related to this same issue.

alessandrocarminati, 2018-01-06:
thank you for answered me. Now I just hopemy observations might be helpful somehow. Just to let you know, even if it's not related in anyway with chan_sccp. After I moved from ast11 to ast15, even without change anithing in the chan_dahdi an th low level driver sangoma's wanpipe, I lost the ability to collect digit from a dahdi device which is signaling using analog pulse dial.

dkgroot, 2018-01-06:
@alessandrocarminati I will check the asterisk-15 changelog and issue reports, just to verify. Maybe there is some upstream issue i have missed. Thanks for that extra information, which may save a lot of time !