chan-sccp/chan-sccp #423: All calls on Cisco 7920 shown as "Private Call"

evs38, 2018-03-01:
Hello!

Can't fix this issue: all calls, both inbound and outbound are shown as "Private Call" on Cisco 7920 phones. Caller ID is shown only in "Detail" option.

**Reproduction Steps:**

1. Dial Cisco phone
2. See the call with "Private Call" 

**Expected behavior:**

See the right caller ID

**Screenshots**

https://imgur.com/gallery/uIsc3

**chan-sccp version:** 

Skinny Client Control Protocol (SCCP). Release: 4.3.0 master - 337b104M (built by 'root' on '2018-03-01 07:20:58 UTC')

**asterisk version:** (include revision and/or branch if relevant)

Asterisk 13.19.0 / FreePBX 14.0.1.36

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [No]
- Problem can be reliably reproduced, doesn't happen randomly: [Yes]
- Platform: Raspberry Pi
- Direct Contact information: [Telegram: https://t.me/RN4HGW]

dkgroot, 2018-03-01:
@evs38 Hi Eugene

I don't have one of these 7920 phones, so it will be a little difficult to re-create the issue locally.
What happens when you press the 'private' softkey before making the call (maybe the handling is inverted ?).

evs38, 2018-03-01:
@dkgroot Yes, it is. Private calls are placed "normally" :)

dkgroot, 2018-03-01:
I was afraid of that (I think someone else told me about it some time ago).
Can you attach a tcpdump of two calls (with and without private pressed), so i can have a look what to fix (flip).

evs38, 2018-03-02:
@dkgroot no problems:
[7920_calls.tar.gz](https://github.com/chan-sccp/chan-sccp/files/1773244/7920_calls.tar.gz)

dkgroot, 2018-03-02:
Hi Eugene,
The issue your reported gave me the feeling i had heard it before. And looking at the tcpdump and the sourcecode, it looks like it was patched specifically for the 7920 (already). As far as i know the 7920 firmware is pretty ancient (and not updated recently).

You can reverse the previous patch i made by applying this patch (7920.patch):
```diff
diff --git a/src/sccp_protocol.c b/src/sccp_protocol.c
index c4a0436e..0e40e6f0 100644
--- a/src/sccp_protocol.c
+++ b/src/sccp_protocol.c
@@ -53,12 +53,12 @@ static void sccp_protocol_sendCallInfoV3 (const sccp_callinfo_t * const ci, cons
                SCCP_CALLINFO_PRESENTATION, &presentation,
                SCCP_CALLINFO_KEY_SENTINEL);
 
-       // 7920's exception. They don't seem to reverse the interpretation of the presentation flag
-       if (device->skinny_type == SKINNY_DEVICETYPE_CISCO7920) {
-               msg->data.CallInfoMessage.partyPIRestrictionBits = presentation ? 0x0 : 0xf;
-       } else {
+       // 7920's exception. They seem to reverse the interpretation of the presentation flag
+       //if (device->skinny_type == SKINNY_DEVICETYPE_CISCO7920) {
+       //      msg->data.CallInfoMessage.partyPIRestrictionBits = presentation ? 0x0 : 0xf;
+       //} else {
                msg->data.CallInfoMessage.partyPIRestrictionBits = presentation ? 0xf : 0x0;
-       }
+       //}
        msg->data.CallInfoMessage.lel_lineInstance = htolel(lineInstance);
        msg->data.CallInfoMessage.lel_callReference = htolel(callid);
        msg->data.CallInfoMessage.lel_callType = htolel(calltype);
```
git apply --reject 7920.patch

Reason for the previous patch:
Someone else reported this issue, and i changed the source code to adapt to chan-sccp, but apparently i shouldn't have. Logically there must be a setting in the phone's firmware/config that would enable/disable privacy/private by default. The other guy report the issue must have had that enabled (or disabled) and you have the reverse setting.

I do not know what the default is or should be.
https://www.manualslib.com/manual/882240/Cisco-7920.html?page=46&term=privacy
Not sure how to interpret this. Cannot find any hard settings in the manual concerning privacy/private.

I am not sure if i should revert the change in the source at this moment, as i do not know what the default is. It does seem odd that this 'reversal' would be necessary, so i guess the 'setting' you have should be the correct one. Not sure how to confirm this though, nor do i know what the other users should change in their settings, to make it work with this patch.

dkgroot, 2018-03-15:
@evs38 Any update ?

bknv, 2021-01-10:
@dkgroot Hello.
I have the same problem, but on phone 7940
I can help you?

Skinny Client Control Protocol (SCCP). Release: 4.3.2 develop - 76000de

dkgroot, 2021-01-10:
@bknv I never heard back from the previous person who reported this issue. 
- "Private Call" in the caller's phone or the destination ?
- Running FreePBX as well ?
- is "private" enabled and or used on the phone
Can you collect some sccp debug logging with "sccp debug core, channel, pbx, indicate, callinfo" that will show the issue ?

bknv, 2021-01-10:
@dkgroot 
- "Private Call" show in incoming and outgoing call. Send you photo
[image]
- Yes, FPBX 14.0.13.40
- I use sccp_manager, and switch "Private Calling" to Off. Right? In /etc/asterisk/sccp.conf also private=off
[sccpdebug.txt](https://github.com/chan-sccp/chan-sccp/files/5792946/sccpdebug.txt)

dkgroot, 2021-01-11:
It's interesting that this only happens to a small amount of people. Maybe it's firmware related or something in the SEPxxxx.cnf.xml file. I am not sure where to start to figure this out. You sccpdebug contains:
`presentation: CallerId Presentation Allowed` in every block of callinfo. So it's telling the phone to show the callerid information. I would assume that when you look in the Phonebook History it will have the callerid information as well. Can you attach the sep file user for this phone and tell use the phones' firmware version ?

bknv, 2021-01-12:
Firmware 8.1(2)SR2, name P0030801SR02
[SEP000750AC6141.cnf.xml.zip](https://github.com/chan-sccp/chan-sccp/files/5800076/SEP000750AC6141.cnf.xml.zip)

mailanetworks, 2021-01-12:
You can update your firmware and it'll fix it.

Att,

bknv, 2021-01-12:
@mailanetworks This is latest version firmware
You think problem in devices?

It device work with Cisco Unified Communications Manager Express version 10.5 normaly

dkgroot, 2021-01-12:
I don't know to be honest. Can you make a packet-capture/tcpdump/wireshark for this device when it is connected to a CME so we can see what it might be doing differently ?

bknv, 2021-01-13:
@dkgroot 
[G.zip](https://github.com/chan-sccp/chan-sccp/files/5806971/G.zip)

dkgroot, 2021-01-14:
@bknv:

Thanks for the tcpdump, that makes live a lot easier.
```
Cisco CallInfo Packet:
Skinny Client Control Protocol
    Data length: 388
    Header version: Basic (0x00000000)
    Message ID: CallInfo (143)
    callingPartyName: 1154
    callingParty: 1154
    calledPartyName: Test Bu
    calledParty: 2360
    lineInstance: 1
    callReference: 80376
    callType: InBoundCall (0x00000001)
    originalCalledPartyName: Test Bu
    originalCalledParty: 2360
    lastRedirectingPartyName: 
    lastRedirectingParty: 
    originalCdpnRedirectReason: 0
    lastRedirectingReason: 0
    cgpnVoiceMailbox: 1154
    cdpnVoiceMailbox: 2360
    originalCdpnVoiceMailbox: 2360
    lastRedirectingVoiceMailbox: 
    callInstance: 1
    callSecurityStatus: NotAuthenticated (0x00000001)
    partyPIRestrictionBits (00 000 00 00)
        .... .... .... .... .... .... .... ...0 = CallingPartyName: No
        .... .... .... .... .... .... .... ..0. = CallingPartyNumber: No
        .... .... .... .... .... .... .... ..00 = CallingParty: No
        .... .... .... .... .... .... .... .0.. = CalledPartyName: No
        .... .... .... .... .... .... .... 0... = CalledPartyNumber: No
        .... .... .... .... .... .... .... 00.. = CalledParty: No
        .... .... .... .... .... .... ...0 .... = OriginalCalledPartyName: No
        .... .... .... .... .... .... ..0. .... = OriginalCalledPartyNumber: No
        .... .... .... .... .... .... ..00 .... = OriginalCalledParty: No
        .... .... .... .... .... .... .0.. .... = LastRedirectPartyName: No
        .... .... .... .... .... .... 0... .... = LastRedirectPartyNumber: No
        .... .... .... .... .... .... 00.. .... = LastRedirectParty: No
        0000 0000 0000 0000 0000 0000 .... .... = BitsReserved: No
```
Asterisk CallInfo Packet:
```
Skinny Client Control Protocol
    Data length: 388
    Header version: Basic (0x00000000)
    Message ID: CallInfo (143)
    callingPartyName: Bukanov Aleksandr
    callingParty: 2327
    calledPartyName: test_bu
    calledParty: 5578
    lineInstance: 1
    callReference: 2215
    callType: InBoundCall (0x00000001)
    originalCalledPartyName: 
    originalCalledParty: 
    lastRedirectingPartyName: 
    lastRedirectingParty: 
    originalCdpnRedirectReason: 0
    lastRedirectingReason: 0
    cgpnVoiceMailbox: 
    cdpnVoiceMailbox: 
    originalCdpnVoiceMailbox: 
    lastRedirectingVoiceMailbox: 
    callInstance: 1
    callSecurityStatus: NotAuthenticated (0x00000001)
    partyPIRestrictionBits (0f 00 00 00);
        .... .... .... .... .... .... .... ...1 = CallingPartyName: Yes
        .... .... .... .... .... .... .... ..1. = CallingPartyNumber: Yes
        .... .... .... .... .... .... .... ..11 = CallingParty: Yes
        .... .... .... .... .... .... .... .1.. = CalledPartyName: Yes
        .... .... .... .... .... .... .... 1... = CalledPartyNumber: Yes
        .... .... .... .... .... .... .... 11.. = CalledParty: Yes
        .... .... .... .... .... .... ...0 .... = OriginalCalledPartyName: No
        .... .... .... .... .... .... ..0. .... = OriginalCalledPartyNumber: No
        .... .... .... .... .... .... ..00 .... = OriginalCalledParty: No
        .... .... .... .... .... .... .0.. .... = LastRedirectPartyName: No
        .... .... .... .... .... .... 0... .... = LastRedirectPartyNumber: No
        .... .... .... .... .... .... 00.. .... = LastRedirectParty: No
        0000 0000 0000 0000 0000 0000 .... .... = BitsReserved: No
```
So "partyPIRestrictionBits" field is the cause, and that makes sense, that what these fields are for. Somehow we are converting "CallerId Presentation Allowed" to "F00000000" instead of "00000000". I already looked for that yesterday, but apparently i missed something. If i cannot find it, i will have to make a little branch with extended logging so we can track it down. At least we know the source of the problem now.

dkgroot, 2021-01-14:
I have reverted a previous patch which logically solves the issue. Based on the data you provided this should just work as it was originally. I left the previous code in place in case there are issues with some 7920's in the future for which the original change was made.

commit 8ab6d72b (2021-01-14): Fix: CallInfo:Presentation to CallInfo:partyPIRestrictionBits for ProtocolLevel:3 messages

Fixes: #423
commit 9af9bd9a (2021-01-14): Fixup: 8ab6d72
Fixes: #423
commit 87900f74 (2021-01-14): Fix: CallInfo:Presentation to CallInfo:partyPIRestrictionBits for ProtocolLevel:3 messages

Fixes: #423
closed 2021-01-14 by dkgroot