chan-sccp/chan-sccp #424: SCCP resetting 7911 phones after update

uha666, 2018-03-01:
After updating chan_sccp to latest version 

- Updated chan_sccp to latest revision.
- Updated asterisk using "yum update" and rebooted.

sccp_class | 11.3 | Sccp mysql class Base ver: 11.3, Sccp ver: 431
Core_sccp | 4.3.1 | Sccp ver.4.3.1 r432 Revision :10451 Hash :452b794
Asterisk | 13.19.1 | Asterisk.
Сompatible | 431

**Observed behavior:**
`[2018-03-01 06:55:32] VERBOSE[3531] : -- SEP00215503A8FB: nat=On set manually in config file
[2018-03-01 06:55:32] VERBOSE[3531] : -- SEP00215503A8FB: Sending ConfigStatMessage, lines 1, speeddials 0
[2018-03-01 06:55:32] VERBOSE[3531] : -- SEP00215503A8FB: Device has 12 Capabilities
[2018-03-01 06:55:32] VERBOSE[3531] : -- SEP00215503A8FB: Got Forward Status Request. Line: 1
[2018-03-01 06:55:32] VERBOSE[3531] : -- SEP00215503A8FB: Device registered; performing post registration tasks...
[2018-03-01 06:55:32] VERBOSE[2430] : -- SEP00215503A8FB: (sccp_devstate_deviceRegisterListener) device registered
[2018-03-01 06:55:32] VERBOSE[3531] : -- SCCP: Alarm Message: Severity: Critical (0), 32: Name=SEP00215503A8FB Load= term06.default : Invalid SCCP message! : ID :87: [0/0]
[2018-03-01 06:55:39] VERBOSE[3462] : SCCP: Clean Device SEP001BD4602EF8, remove from global:No, restart_device:No
[2018-03-01 06:55:39] VERBOSE[3462] : == SCCP: Remove Line 203 from device SEP001BD4602EF8
[2018-03-01 06:55:39] VERBOSE[3462] : == SCCP: Unregister Device SEP001BD4602EF8
[2018-03-01 06:55:39] VERBOSE[2430] : -- SEP001BD4602EF8: (sccp_devstate_deviceRegisterListener) device unregistered
[2018-03-01 06:59:02] VERBOSE[2440] : -- SCCP: Accepted Client Connection from 192.168.56.138:15328
[2018-03-01 06:59:02] VERBOSE[2440] : -- SCCP: Connected on server via 192.168.56.245:35059
[2018-03-01 06:59:02] VERBOSE[3692] : -- SCCP: Alarm Message: Severity: Informational (2), 10: Name=SEP001BD4602EF8 Load= term11.default Last=TCP-timeout [0/0]
[2018-03-01 06:59:02] VERBOSE[3692] : -- SCCP: Alarm Message: Severity: Informational (2), 17: Name=SEP001BD4602EF8 Load= term11.default Last=KeepaliveTO [0/0]
[2018-03-01 06:59:02] VERBOSE[3692] : -- SEP001BD4602EF8: nat=On set manually in config file
[2018-03-01 06:59:02] VERBOSE[3692] : -- SEP001BD4602EF8: Sending ConfigStatMessage, lines 1, speeddials 0
[2018-03-01 06:59:02] VERBOSE[3692] : -- SEP001BD4602EF8: Device has 12 Capabilities
[2018-03-01 06:59:03] VERBOSE[3692] : -- SEP001BD4602EF8: Got Forward Status Request. Line: 1
[2018-03-01 06:59:03] VERBOSE[3692] : -- SEP001BD4602EF8: Device registered; performing post registration tasks...
[2018-03-01 06:59:03] VERBOSE[2430] : -- SEP001BD4602EF8: (sccp_devstate_deviceRegisterListener) device registered
[2018-03-01 06:59:03] VERBOSE[3692] : -- SCCP: Alarm Message: Severity: Critical (0), 32: Name=SEP001BD4602EF8 Load= term11.default : Invalid SCCP message! : ID :87: [0/0]
[2018-03-01 07:05:13] VERBOSE[3343] : SCCP: Clean Device SEP0024973409FC, remove from global:No, restart_device:No
[2018-03-01 07:05:13] VERBOSE[3343] : == SCCP: Remove Line 204 from device SEP0024973409FC
[2018-03-01 07:05:13] VERBOSE[3343] : == SCCP: Unregister Device SEP0024973409FC
[2018-03-01 07:05:13] VERBOSE[2430] : -- SEP0024973409FC: (sccp_devstate_deviceRegisterListener) device unregistered
[2018-03-01 07:05:44] VERBOSE[3692] : SCCP: Clean Device SEP001BD4602EF8, remove from global:No, restart_device:No
[2018-03-01 07:05:44] VERBOSE[3692] : == SCCP: Remove Line 203 from device SEP001BD4602EF8
[2018-03-01 07:05:44] VERBOSE[3692] : == SCCP: Unregister Device SEP001BD4602EF8
[2018-03-01 07:05:44] VERBOSE[2430] : -- SEP001BD4602EF8: (sccp_devstate_deviceRegisterListener) device unregistered
[2018-03-01 07:07:22] VERBOSE[3531] : SCCP: Clean Device SEP00215503A8FB, remove from global:No, restart_device:No
[2018-03-01 07:07:22] VERBOSE[3531] : == SCCP: Remove Line 202 from device SEP00215503A8FB
[2018-03-01 07:07:22] VERBOSE[3531] : == SCCP: Unregister Device SEP00215503A8FB
[2018-03-01 07:07:22] VERBOSE[2429] : -- SEP00215503A8FB: (sccp_devstate_deviceRegisterListener) device unregistered
[2018-03-01 07:08:40] VERBOSE[2440] : -- SCCP: Accepted Client Connection from 192.168.56.137:34328
[2018-03-01 07:08:40] VERBOSE[2440] : -- SCCP: Connected on server via 192.168.56.245:40897
[2018-03-01 07:08:40] VERBOSE[4142] : -- SCCP: Alarm Message: Severity: Informational (2), 10: Name=SEP0024973409FC Load= SCCP11.8-4-2S Last=TCP-timeout [0/0]
[2018-03-01 07:08:40] VERBOSE[4142] : -- SCCP: Alarm Message: Severity: Informational (2), 17: Name=SEP0024973409FC Load= SCCP11.8-4-2S Last=KeepaliveTO [0/0]
[2018-03-01 07:08:41] VERBOSE[4142] : -- SEP0024973409FC: nat=On set manually in config file
[2018-03-01 07:08:41] VERBOSE[4142] : -- SEP0024973409FC: Sending ConfigStatMessage, lines 1, speeddials 0
[2018-03-01 07:08:41] VERBOSE[4142] : -- SEP0024973409FC: Device has 12 Capabilities
[2018-03-01 07:08:41] VERBOSE[4142] : -- SEP0024973409FC: Got Forward Status Request. Line: 1
[2018-03-01 07:08:41] VERBOSE[4142] : -- SEP0024973409FC: Device registered; performing post registration tasks...
[2018-03-01 07:08:41] VERBOSE[2429] : -- SEP0024973409FC: (sccp_devstate_deviceRegisterListener) device registered
[2018-03-01 07:08:41] VERBOSE[4142] : -- SCCP: Alarm Message: Severity: Critical (0), 32: Name=SEP0024973409FC Load= SCCP11.8-4-2S : Invalid SCCP message! : ID :87: M [0/0]
[2018-03-01 07:09:08] VERBOSE[2440] : -- SCCP: Accepted Client Connection from 192.168.56.138:26957
[2018-03-01 07:09:08] VERBOSE[2440] : -- SCCP: Connected on server via 192.168.56.245:56531
[2018-03-01 07:09:08] VERBOSE[4174] : -- SCCP: Alarm Message: Severity: Informational (2), 10: Name=SEP001BD4602EF8 Load= term11.default Last=TCP-timeout [0/0]
[2018-03-01 07:09:08] VERBOSE[4174] : -- SCCP: Alarm Message: Severity: Informational (2), 17: Name=SEP001BD4602EF8 Load= term11.default Last=KeepaliveTO [0/0]
[2018-03-01 07:09:08] VERBOSE[4174] : -- SEP001BD4602EF8: nat=On set manually in config file
[2018-03-01 07:09:08] VERBOSE[4174] : -- SEP001BD4602EF8: Sending ConfigStatMessage, lines 1, speeddials 0
[2018-03-01 07:09:08] VERBOSE[4174] : -- SEP001BD4602EF8: Device has 12 Capabilities
[2018-03-01 07:09:08] VERBOSE[4174] : -- SEP001BD4602EF8: Got Forward Status Request. Line: 1
[2018-03-01 07:09:08] VERBOSE[4174] : -- SEP001BD4602EF8: Device registered; performing post registration tasks...
[2018-03-01 07:09:08] VERBOSE[2430] : -- SEP001BD4602EF8: (sccp_devstate_deviceRegisterListener) device registered
[2018-03-01 07:09:09] VERBOSE[4174] : -- SCCP: Alarm Message: Severity: Critical (0), 32: Name=SEP001BD4602EF8 Load= term11.default : Invalid SCCP message! : ID :87: [0/0]
[2018-03-01 07:10:59] VERBOSE[2440] : -- SCCP: Accepted Client Connection from 192.168.56.139:13157
[2018-03-01 07:10:59] VERBOSE[2440] : -- SCCP: Connected on server via 192.168.56.245:41691
[2018-03-01 07:10:59] VERBOSE[4245] : -- SCCP: Alarm Message: Severity: Informational (2), 10: Name=SEP00215503A8FB Load= term06.default Last=TCP-timeout [0/0]
[... 428 lines cut ...]
[2018-03-01 08:53:34] VERBOSE[8351] : == SCCP: Unregister Device SEP00215503A8FB
[2018-03-01 08:53:34] VERBOSE[2429] : -- SEP00215503A8FB: (sccp_devstate_deviceRegisterListener) device unregistered
[2018-03-01 08:57:11] VERBOSE[2440] : -- SCCP: Accepted Client Connection from 192.168.56.139:19638
[2018-03-01 08:57:11] VERBOSE[2440] : -- SCCP: Connected on server via 192.168.56.245:38079
[2018-03-01 08:57:11] VERBOSE[9066] : -- SCCP: Alarm Message: Severity: Informational (2), 10: Name=SEP00215503A8FB Load= term06.default Last=TCP-timeout [0/0]
[2018-03-01 08:57:11] VERBOSE[9066] : -- SCCP: Alarm Message: Severity: Informational (2), 17: Name=SEP00215503A8FB Load= term06.default Last=KeepaliveTO [0/0]
[2018-03-01 08:57:11] VERBOSE[9066] : -- SEP00215503A8FB: nat=On set manually in config file
[2018-03-01 08:57:11] VERBOSE[9066] : -- SEP00215503A8FB: Sending ConfigStatMessage, lines 1, speeddials 0
[2018-03-01 08:57:11] VERBOSE[9066] : -- SEP00215503A8FB: Device has 12 Capabilities
[2018-03-01 08:57:11] VERBOSE[9066] : -- SEP00215503A8FB: Got Forward Status Request. Line: 1
[2018-03-01 08:57:11] VERBOSE[9066] : -- SEP00215503A8FB: Device registered; performing post registration tasks...
[2018-03-01 08:57:11] VERBOSE[2429] : -- SEP00215503A8FB: (sccp_devstate_deviceRegisterListener) device registered
[2018-03-01 08:57:11] VERBOSE[9066] : -- SCCP: Alarm Message: Severity: Critical (0), 32: Name=SEP00215503A8FB Load= term06.default : Invalid SCCP message! : ID :87: [0/0]`

As far as i can tell it only affects our 7911 phones.

Our current network is as follows:

2x 7975 phones
4x 7911 phones
2x CIPC app's

FreePBX 13.0.194.2	'VoIP Server'
Current Asterisk Version: 13.19.1
CentOS Linux release 7.1.1503 (Core)

dkgroot, 2018-03-01:
@uha666 Looks lilke we have an issue in sccp protocol message handling. The 87 (ie:0x57) does not exist, causing the device to reset. Logically we are sending something to the 7911 device it does not know how to handle.
It would help to have the debug log from the phone itself. You should be able to grab it from the phones webpage.
BTW: Which firmware version are you using on these 7911 phones ?

uha666, 2018-03-02:
All 7911G phones are running firmware SCCP11.8-4-2S.loads.

I will try and get the debug logs from the phone and get back to you.

Thanks.

uha666, 2018-03-02:
Logs attached.

Regards.

[7911G_logs.zip](https://github.com/chan-sccp/chan-sccp/files/1774479/7911G_logs.zip)

dkgroot, 2018-03-02:
@uha666 I looked at the phones debug log and i see this repeating pattern:
```
ERR 08:19:37.141661 JVM: Startup Module Loader|cip.sccp.f:? - Button not allowed on this phone model, button=3 inst=0
CRT 08:19:37.143137 JVM: Startup Module Loader|cip.sccp.ae@997834:createMessage - No entry in hashtable for ID=0x87
WRN 08:19:37.145181 JVM: Startup Module Loader|cip.sccp.ck:? - Invalid SCCP message! : ID :87: MessageFactory.createMessage failed, length = 0 - close connection and alarm in future
```
It seems that somthing in the devices button definition does not get parsed nicely (This is a one line/button phone, so having multiple buttons assigned to it might not help). The nasty thing is, that while/when it stumbles over the  the button=3 message, it then does not finish the skinny message, and actually falls over some moments later, when it tries to read the next message.

**Explanation:**
Let's say these are to messages on the tcp stream
```[ 08 02 13 04 05 04 02 87 ] [ 05 02 11 05 03 ]```
The first number denotes the length, the second the protocol version, the third the message ID
So:
```
message 1 reads: length=8, version=2, messageID=13, data = [04 05 04 02 87]
message 2 reads: length=5, version=2, messageID=11, data = [05, 03]
```
What seems to happen, when reading message 1 it encounters an error in the button definition, after heaving read the first 5 values from the tcp stream, it stops and reports the error, but it does not reset or fault.
It then picks up reading from the stream, but instead of skipping the whole number 1 message (ie skip 3), it starts to read at position 06 in the stream, so that it sees: 
lenght=4, version=2, messageID=87, data= [05]
Reports that it does not know messageID87 and it finally gives up an resets.

Can you please send the device definition for this device in you sccp.conf ? You should be able to fix the issue, by reducing the number of buttons assigned to this device, or maybe fix the layout of the button definition.
I would like to see the sccp.conf part for this device, to see if there is anything we could have done to prevent this error in the future.

dkgroot, 2018-03-02:
BTW: There is newer firmware for these phones 9-4-2sr3 as far as i know. The message handling issue might have been fixed by cisco between these revisions.

uha666, 2018-03-02:
Sccp config attached.

Regards.

dkgroot, 2018-03-02:
Looked at your sccp.conf and nothing really stands out, to be honest. I was expecting some 7911 device with a lot of buttons or complex button definitions or something.
Would you be able to make a very simple/minimal sccp.conf with only one device/line combination, just to see if that works ? 
Have you tried newer firmware on one of the 7911 phones to see if that works ?

The latest i have is : [7906_7911_9.2.1](https://github.com/dkgroot/provision_sccp/tree/master/tftpboot/firmware/7906_7911)

But on the cisco site you can find 9.4.SR3 for this phone.

warst, 2018-03-02:
In case you can’t get hold of them easily, I’ve put them on a wetransfer.com link - https://we.tl/gtoXY93oil

dkgroot, 2018-03-02:
@warst Thanks Warren
I updated my github tftp / provisioning / firmware repo, with the new files !

warst, 2018-03-04:
No probs. Let me know if you require any others down the line, chances I have them or can get them!

dkgroot, 2018-03-04:
@uha666 Following is Unrelated to your/this PR:

@warst Would you mind checking https://github.com/dkgroot/provision_sccp/ To see:
- if i am missing any
- if you have updated once of any of these

If you like you can send them via mail / or submit a PR against the gh repo.
Little warning: the provision_sccp project uses regex rewrite rules, to redirect incoming traffic to their destination (nginx and tftp-hpa). This was done to be able to have a clean/structured tftpboot directory without everything jumbled together.

uha666, 2018-03-05:
@dkgroot can you explain how can i update the device? They came this way and i never performed an update on them.

warst, 2018-03-05:
@uha666 it's straightforward if you already have chan-sccp setup you will already have a suitable TFTP server running - have a look here - https://supportforums.cisco.com/t5/collaboration-voice-and-video/upgrade-firmware-with-3rd-party-tftp-server-tfpd32/ta-p/3130188

dkgroot, 2018-03-05:
@uha666 You download the firmware file, put them in your tftpboot directory
Then also download the [XMLDefault.cnf.xml](https://github.com/dkgroot/provision_sccp/blob/master/tftpboot/settings/XMLDefault.cnf.xml) and update the IP settings  in it. The loadInformation entries is what tells the phone which files to download.
If you do not want to use XMLDefault.cnf.xml you can copy the loadInformation entry into your SEP.....cnf.xml. Or you can give the phone a factory reset (They will retrieve the xxxxx.loads file upon a factory reset, and download whatever it points to).

dkgroot, 2018-03-05:
@warst You beat me to it, although that is a very windows oriented solution :-)

dkgroot, 2018-03-05:
FYI: 
For questions like how to setup tftp etc, always check the [wiki](https://github.com/chan-sccp/chan-sccp/wiki/) first, and you are welcome to our [chat](https://gitter.im/chan-sccp/chan-sccp?utm_source=badge&utm_medium=badge&utm_campaign=pr-badge), where there should be someone who can help as well.

uha666, 2018-03-05:
Hi,

Thanks i managed to upgrade all my phones using your provisioning files thank!

However the problem remained, but i figured out what the problem was after doing some testing.

I noticed that after the upgrade only 3 of the 4 7911G's were rebooting, wich didn't make any sense since they were now on the same firmware.

I did some digging and found out that there was an ip conflict in my network because of newly created vm's with fixed ip's, that had the same ip's as 3 of our phones, after correting that issue the phones stopped rebooting. Shouldn't there be a way for the server or phone to detect this issue and report it besides just reseting?

Thanks for all your help. 

Regards

uha666, 2018-03-05:
While i'm here can @dkgroot or @warst explain how do i update the localization? I'm trying to use Portuguese_Portugal but can't seem to make the phones download i always get an "Error Updating Locale" in the phone.

Regards

dkgroot, 2018-03-05:
@uha666 
Rebooting because of duplicate ip's after having been able to connect to the server, is a bit strange. Then again we do not have insight into the phones firmware. For all we know they might be trying to calculate what the question was that has 42 as it's answer :-) The behaviour is curious, but that is about all i can say.
Advise: when you want to assign a static ip-address to a particular server, make a reservation for this number on your dhcp server, and have the server just obtain the 'static' ip-address from the dhcp server. Real static ip-address always bite you at some time.
Closing the issue report.
It might be a good idea to add an entry in our wiki Q&A about this phenomenon, would you feel up to it ?

dkgroot, 2018-03-05:
@uha666 Regarding Localization:
You normally add 2 locale entries into your SEP-file, like so:
```xml
  <!-- Begin Locale Information -->
  <userLocale> 
    <name>English_United_States</name>
    <uid>39</uid>
    <langCode>en_US</langCode>
    <version>10.5(2.9902)US</version>
    <winCharSet>utf-8</winCharSet>
  </userLocale> 
  <networkLocale>United_States</networkLocale> 
  <networkLocaleInfo>
    <name>United_States</name>
    <uid>64</uid>
    <version>10.5(1)</version>
  </networkLocaleInfo>
  <!-- End Locale Information -->
```
The "name" entry equals the directory name in the tftpboot root dir. So in this case i have:
```
/tftpboot/United_States          <-- Country information (ie: network locale)
/tftpboot/English_United_States  <-- Language information pertaining to this country (ie: user locale)
```
Having a tftp server that generates a little bit of debug normally helps to figure out what the the client/phone is requesting/expecting to find and where it is expecting these files to be.

uha666, 2018-03-05:
How do i know what is the configuration and files for portugal? is there a tutorial i can check?

Regards

dkgroot, 2018-03-05:
Just look in https://github.com/dkgroot/provision_sccp/tree/master/tftpboot/locales

dkgroot, 2018-03-05:
@uha666 Added some initial instructions to the provision_sccp page on how it can/could be used.

ojaksch, 2018-03-07:
@uha666 Any news on that? I prepared a ready-to-run package in the meantime, which _should_ do it's job at your site.
Please have a closer look at SEP00215503A8FB.cnf.xml in this package (an example and one of your phones) to the **timeZone**, **processNodeName** and **ipAddr1** parameters and change it to the corresponding values.
Then test and report ;)

[uha666.zip](https://github.com/chan-sccp/chan-sccp/files/1788176/uha666.zip)

uha666, 2018-03-09:
Hi @ojaksch, big fan of leisure suit larry btw =P

I tried the package but had to move everything to root folder (language files included) it couldn't find the folders for some reason. 

Thanks!

dkgroot, 2018-03-12:
@uha666 Congrats on getting everything working. Check the tftp-servers logging output (syslog), to see if the phones requested the files via any subdirectories first.

@ojaksch Thanks for helping out !

Closing this issue as the main issue request has been resolved.

closed 2018-03-12 by dkgroot