chan-sccp/chan-sccp #426: Setting callingparty/calledparty via Set(CHANNEL) is being ignored/overwritten

marek-cerny, 2018-03-19:
As the title says, when trying to set CallingParty and/or CalledParty using `Set(CHANNEL(%a)=%b)` dialplan function, nothing happes. Setting CallingParty via classic `Set(CALLERID(all)=%b)` works, setting CalledParty via now deprecated `SCCPSetCalledParty(%b)` doesn't.

**Reproduction Steps:**

1. Debian 8 or 9, clean install
2. Install Asterisk 11 or 13 respectively from repos
3. Build chan-sccp (I've tested v4.3.1-beta.1, v4.2.3 (STABLE) and v4.2.1.1, all the same)
4. Stock Asterisk config, basic dialplan and sccp config with 2 Cisco phones
5. Call one phone from another

**Expected behavior:**

Friendly CallingParty and CalledParty names being displayed on the phones.

**Observed behavior:**

Only numbers being displayed on the phones.

**chan-sccp version:** v4.3.1-beta.1, v4.2.3 (STABLE), v4.2.1.1

**asterisk version:** 13.14.1, 11.13.1

**Addition / Optional Information:**
- I do remember using function `SCCPSetCalledParty()` successfully in the past - unfortunately I do not have access to that server any more and I don't remember specifics or anything important that would explain why it doesn't work with current stable any more. I belive it was built-from-source Asterisk on Debian Wheezy with chan_sccp v4.2.3, but I am not sure.
- Problem can be reliably reproduced, doesn't happen randomly: Yes

When observing the call flow with `sccp debug core, channel, callinfo`, one can see that the variables get properly updated with their corresponding `Set(CHANNEL())` calls, but once the call gets dialed, the info is lost.

sccp.conf: [pastbin](https://pastebin.com/zP1qEq4N)
extensions.conf: [pastbin](https://pastebin.com/b9naJcbR)
debug: [pastbin](https://pastebin.com/UjJkRKn7)

dkgroot, 2018-03-19:
@marek-cerny 
Thanks for gathering the info and being explicit in the versions you use, that is always very helpfull.

I need to figure out a way to solve this. Current thinking is just to add a simple parameter to the callinfo handler, to disallow it from altering the state, after it was set manually (through SCCPSetCalledParty or Channel(calledparty). That should be somewhat easy to add and hopefully prevent major regressions.

I am guess you came to your current version from a rather old one (<V4.2) ?

I am not sure what you are using SCCPSetCalledParty for specifically. I do remember it was used back then because the options to set values like this in older asterisk version was a little limited.
Current callinfo handling in normal call-flow (without any aid) has changed/improved quite a bit. So you might even be able to do without. If not you will have to wait a little bit until i figure out a fix.

marek-cerny, 2018-03-20:
> I am guess you came to your current version from a rather old one (<V4.2) ?

I belive that chan_sccp wasn't that old (it already supported conferences), but the Asterisk itself might have been rather old, yes. But as I don't have access to that server any more, I really can't tell.

> I am not sure what you are using SCCPSetCalledParty for specifically. I do remember it was used back then because the options to set values like this in older asterisk version was a little limited.
Current callinfo handling in normal call-flow (without any aid) has changed/improved quite a bit. So you might even be able to do without. If not you will have to wait a little bit until i figure out a fix.

Not sure if I fully understand. In the `extension.conf` above, I've used both methods (`Set(CHANNEL)` and `SCCPSetCalledParty()`) just to test them out and compare to each other. But at the moment, I don't really mind which of them will be used as long as I can achieve the expected behavior - i.e. user dials a number, and if that number is known in the central address book on the server, calledParty name gets displayed on his phone. At the same time, it's not a critical funcionality, it only adds something to the user experience but does not affect core VoIP functionality.

Are there any other methods to set calledParty besides `Set(CHANNEL)` and `SCCPSetCalledParty()`? I am not entirely sure what are you referring to when talking about "current callinfo handling in normal call-flow".

dkgroot, 2018-03-25:
Between asterisk-1.8 and asterisk-11 all callerid handling changed (a lot). It changed again when asterisk-13 came about. You can find more about how callerid can be influence on [this page](https://wiki.asterisk.org/wiki/display/AST/Manipulating+Party+ID+Information). Chan-sccp has adapted to this new handling and also underwent some heavy remodeling to improve callinfo representation on the phone.

I don't use extension.conf instead i moved to using ael (extension.ael) a long time ago and 'lost' the ability to write/read/interpret extension.conf. So most of my examples would use ael syntax (this can however be easily converted back to conf format using the 'ael2conf' util).

I mostly focus on normal call handling (simple SCCP->SIP  / SIP->SCCP calls) to try and make sure the presented callinfo is correct. Intervention of the callinfo using a dialplanfunction like SCCPSetCalledParty has not received a lot of attention as of late. As soon as i am done with issue #420 (which is taking way more time than i had hoped), i will move on to this issue.

marek-cerny, 2018-05-17:
Just to add a little bit more info:

I've described that "calledParty" variable is properly set by the "SCCPSetCalledParty()" call, but the content gets lost once the call gets dialed.

Today I've noticed that when the call doesn't go thru (wrong number, subscriber absent), the variable doesn't get truncated and calledparty is displayed on the phone as expected.

Debug of a proper call that does go thru differs at this line:
"SCCP: callinfo constructor: 0x7f7e38019df0"
After this, "calledParty" content gets truncated.

So this is the place where I'd probably start to look for the issue. I've tried to go thru the source files myself but with no luck so far.

dkgroot, 2019-09-15:
@marek-cerny Would you mind checking the current latest version. I have done some work on caller id handling in the last couple of weeks.

What should work.
Outgoing call:
 - [X] Set(CHANNEL(CalledParty)=XXXX)
 - [X] SCCPSetCalledParty(XXXX)
 - [X] Set(CALLERID(all)=XXXX)

Incoming call (Before Dial()):
 - [ ] Set(CHANNEL(CalledParty)=XXXX)
 - [ ] SCCPSetCalledParty(XXXX)
 - [X] Set(CALLERID(all)=XXXX)

Incoming call (using Dial() Macro):
 - [X] Set(CHANNEL(CalledParty)=XXXX)
 - [X] SCCPSetCalledParty(XXXX)
 - [ ] Set(CALLERID(all)=XXXX)

Note: If this issue was already remedied / not of interest any more. Then please just close it :-)

dkgroot, 2020-01-17:
in 9410ed89 a crash issue was fixed. Also SCCPSetCalledParty was fixed and should work like before. As there has not been any further feed back, closing this issue for now.

closed 2020-01-17 by dkgroot