chan-sccp/chan-sccp #428: Double ringback tone when calling via SIP provider

marek-cerny, 2018-05-28:
Our Asterisk PBX is connected to a VoIP provider via SIP trunk. When making calls that go thru this trunk, we hear a double ringback tone (i.e. the tone that hears the calling party from the earpiece while the remote destination is ringing) from the phones. This does not happen when local SCCP phones call each other, or when we dial a secondary Asterisk PBX connected via IAX2 trunk.

As far as I understood the situation, when SCCP phones call each other and there is just one proper ringback tone, it's actually generated by the phone itself (I see no RTP packets coming from/to the PBX).

During the problematic behavior, one of the ringback tones is the regular one from the phone, while the other comes as an RTP stream - I've found out that it's generated by our own local PBX, it does not come from the VoIP provider.

When testing against a different VoIP provider that actually does send the ringback tone via the SIP trunk, the double ringback problem does not happen - in this case the phone somehow knows that there is an incoming RTP ringback and does not generate it's own.

Setting `earlyrtp=None` does get rid of the secondary ringback tone, but also kills all other RTP announcements ("number is not available", etc.) which is not an option.

**chan-sccp version:** 4.3.1-beta.1

**asterisk version:** 13.14.1

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: Yes
The problem **does not** happen in 4.2.3. It does persist in current master (69fdf95).
- Problem can be reliably reproduced, doesn't happen randomly: Yes

I'd like to provide additional logs and so on (I beliveve that especially a comparison towards 4.2.3 could be helpful), but I'd like to know which debug flags should I use to produce the logs with desired scope.

Thanks!

stoyanmarinov, 2018-08-02:
I have the same issue. @marek-cerny did you find a solution for it?

ger-pakon, 2018-10-06:
Hey, I've got the same Issue. Two Asterisks a trunked via IAX2. Calls from the secondary Asterisk to its local phones (via SIP) have ever double ringbacks. Calls from the secondary Asterisk to other SIP providers have never  double ringbacks.

Thanks,
Pakon

dkgroot, 2018-12-23:
@marek-cerny We should be able to figure this out together. Either by doing a "git bisect" (which is pretty simple and fast). Or finding the causing indication in sccp_indications.c. There are a couple of points where we generate ringing tones (using sccp_dev_starttone), so should be pretty easy to figure it out.
Can you give me a hand with this ?

ger-pakon, 2018-12-23:
Hey @dkgroot,

I may also can give you further information.

Best,
Pakon

dkgroot, 2018-12-23:
BTW: Also logged in on [gitter](https://gitter.im/chan-sccp/chan-sccp), if you have any questions !

dkgroot, 2018-12-24:
@marek-cerny @ger-pakon @stoyanmarinov 
Can you do a test with the current develop branch, just to make sure the issue is not already resolved using the latest changes (issue #420, #308, #410 and #438 were fixed and merged). Before we start on a quest to solve it.

marek-cerny, 2018-12-24:
Hey @dkgroot. It's great to have you back!

I am not at the office these days to do some thorough testing (as this particular issue requires physical access to the phones). But a very quick test at my home mini-pbx shows that the issue indeed does not happen any more with current develop branch (dbcb927).

For now, you can either close the issue, or keep it on hold until further confirmation. I'll be back at the office next week.

Thanks!

ger-pakon, 2018-12-25:
Hey, 

i've just tried the newest branch. *Should* be fixed, now, but i've hadn't enough time to test it very well.

Maybe someone else has a little bit more time to test it :-)

greetings,
Pakon

dkgroot, 2018-12-25:
@ger-pakon
I also ran into the issue on one of my systems, which made it a little easier for me to fix it and test it. So it should be ok. Let's close the issue for now.
Thanks all of you for reporting/testing and being quite patient !
Merry Christmas !

ger-pakon, 2018-12-25:
@dkgroot

Thank very much you for your work and merry christmas.

Greetings, 
Pakon

dkgroot, 2018-12-25:
@stoyanmarinov 
Same to you :-)
I noticed some of the double progress indications as well. I am currently working on #438 which deals heavily with Progress/Proceed indications, so i am sure i will get around to it.

dkgroot, 2019-03-13:
@marek-cerny  Reopened / Issue is back

marek-cerny, 2019-03-13:
Based on our gitter conversation, I'd like to report that the issue is back with "v4.3.2-gamma" release, also with the current "cae1a5f" head and even with "issues/453_1" branch.

When I go back to the "dbcb927", the problem does not happen.

dkgroot, 2019-03-13:
Test using: "same => n,Dial(PJSIP/${EXTEN}@provider-trunk,55,t)"

dkgroot, 2019-03-13:
@marek-cerny 
FYI: You might want to check the meaning of the 't' parameter.. It means that the called party can transfer calls using DTMF. I think you want to set 'T' when dialing outside, allowing the caller to transfer using DTMF. Using 'T' and or 't' is inherently dangerous, and maybe neither should be used (ie: not necessary with sccp anyway).

marek-cerny, 2019-03-14:
@dkgroot 
Hey Diederik.

I've run `git bisect` in an attempt to narrow it down. I've never used that tool before - hopefully I understood the process. Here is the result:
```
dc3b2fe17e5766b87d06d044fa7e08e172162ea5 is the first bad commit
commit dc3b2fe17e5766b87d06d044fa7e08e172162ea5
Author: Diederik de Groot <dkgroot@talon.nl>
Date:   Thu Jan 3 23:45:42 2019 +0100

    Minimal fix for earlyrtp issues reported in issue #438

    references #438

:040000 040000 d50e4b5c52706f421176bcf320c196afdc07877d 0ad13b586a3897a9e00f626b76cd3163740c4045 M      src
```

Also, thanks for the hint regarding Dial() parameters.

dkgroot, 2019-03-14:
@marek-cerny 
It would be easy to verify if you did it correctly, the commit before the one your found (353be6d) should be working all-right, and then moving forward to it (dc3b2fe), should make the issue appear again. Can you do that verification ?

You are right that the change is really small which is very positive. Looking at the content of the patch, it's not easily apparent to me how it would be the cause, but there might be a subtle detail i am overlooking.
It would also make it easy to create reverse patch that we can apply the latest (current) code.

I created a new branch that does just that, called "issues/428_1". That would be the next step for you to confirm the reversal of the patch is enough to get rid of the double ringback. We can explorer why it works later on.

```
git pull
git fetch -a
git branch
git checkout issues/428_1
./configure [....]
make -j4 && make install && make reload
```

marek-cerny, 2019-03-14:
@dkgroot 

Yep, I can confirm that.

Proper single ringback with [353be6d](https://github.com/chan-sccp/chan-sccp/commit/353be6d7cb0cd7e6c63b9472a8c6a07b84afc054).

Double ringback with [dc3b2fe](https://github.com/chan-sccp/chan-sccp/commit/dc3b2fe17e5766b87d06d044fa7e08e172162ea5).

And once again single ringback with "issues/428_1 branch".

dkgroot, 2019-03-14:
@marek-cerny I don't understand how it fixes it (yet) it will require a little more indepth research... but at least we can make it work... and you can use the "issues/428_1 branch" for now :-)

dkgroot, 2019-03-14:
@marek-cerny Can you check which part of the small change does have the actual impact (ie: the change in sccp_channel.c or the one in sccp_pbx.c) ?

dkgroot, 2019-09-15:
No response for some time ! We closed this issue for now.
If/when necessary, do reopen the issue please.

closed 2019-09-15 by dkgroot