chan-sccp/chan-sccp #429: 7925 keeps reconnecting every 35/50 secs

Neo2003, 2018-06-03:
Hi

I use chan_sccp with asterisk for years.
Lately, I had one phone dead (a 7921) which I replaced with a refurbished 7925.

My setup is very small to cover a home: 2 * 7962G + 1 * 7975G + 1 * 7925

While the 7925 has no apparent problem I see lots of disconnection events in logs and it seams to consume the 7925 battery fast.

**Reproduction Steps:**
My sccp.conf (with lines removed) : https://pastebin.com/MDuJXi4a
7925 device is 7CAD74424393, its cnf.xml file: https://pastebin.com/BYU5YQgY

What else can I say? the phone is connected to the network using a Cisco Aironet 2700 series and the wifi connection is WPA2 5Ghz. with a cipher TKIP/AES CCM.

**Expected behavior:**
No reconnection every 30-50 secs, longer battery life.

**Observed behavior:**

The phone reconnects using the next port every 30 to 50 secs.

**Screenshots**

**chan-sccp version:** (include revision and/or branch if relevant)
Skinny Client Control Protocol (SCCP). Release: 4.3.1 (HEAD - 69fdf95M (built by 'root' on '2018-06-03 11:54:54 UTC')
Branch "develop"

**asterisk version:** (include revision and/or branch if relevant)
Asterisk 13.19.0

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Yes/No] Problem since I have this 7925 1 month ago.
- Problem can be reliably reproduced, doesn't happen randomly: [Yes/No] Yes
- Platform:
  - [CPU] Intel(R) Core(TM) i5-3470 CPU @ 3.20GHz
  - [32/64 Bit] 32 bits
  - [Big/Small Endian] 
  - [Operating System] Gentoo linux
  - [Operating System Version] Linux version 4.9.95-gentoo (gcc version 6.4.0 (Gentoo 6.4.0-r1 p1.3) )
- Direct Contact information: [You can contact me here]
- Log Files (attach .txt / [pastbin](http://pastebin.com/) / [gist](http://gist.github.com/)): [url]

    -- SCCP: Accepted Client Connection from 192.168.10.87:1229
    -- SCCP: Connected on server via 192.168.10.254:2000
    -- SCCP: Accepted Client Connection from 192.168.10.87:1230
    -- SCCP: Connected on server via 192.168.10.254:2000
    -- SCCP: Alarm Message: Severity: Informational (2), Name=SEP7CAD74424393 Load=CP7925G-1.4.7.3 Last=CM-aborted-TCP [67584/1460316352]
    -- SEP7CAD74424393: Device has 12 Capabilities
    -- Added extension '107' priority 1 to sccpregistration
    -- SEP7CAD74424393: Device registered; performing post registration tasks...
    -- SEP7CAD74424393: (sccp_devstate_deviceRegisterListener) device registered
    -- SEP7CAD74424393: Accessory 'Headset' is 'On Hook' (0)
    -- SCCP: Accepted Client Connection from 192.168.10.87:1231
    -- SCCP: Connected on server via 192.168.10.254:2000
[Jun  3 14:16:35] NOTICE[26699]: sccp_session.c:648 sccp_session_device_thread: SCCP: Closing session because connection timed out after 31 seconds (ip-address: 192.168.10.254:2000).
    -- SCCP: Accepted Client Connection from 192.168.10.87:1232
    -- SCCP: Connected on server via 192.168.10.254:2000
[Jun  3 14:17:06] NOTICE[26700]: sccp_session.c:648 sccp_session_device_thread: SCCP: Closing session because connection timed out after 31 seconds (ip-address: 192.168.10.254:2000).
    -- SCCP: Accepted Client Connection from 192.168.10.87:1233
    -- SCCP: Connected on server via 192.168.10.254:2000
[Jun  3 14:17:39] NOTICE[26701]: sccp_session.c:648 sccp_session_device_thread: SCCP: Closing session because connection timed out after 31 seconds (ip-address: 192.168.10.254:2000).
    -- SCCP: Accepted Client Connection from 192.168.10.87:1234
    -- SCCP: Connected on server via 192.168.10.254:2000

and it goes up every 30 to 50 secs, port 1235 then 1236 then 1237... after 30 minutes: port 1275 and it goes on like this until the phone runs out of battery.

- Comments: [Other Information]
This does not prevent the phone from ringing and making calls. It works just fine. I had a 30 minutes call this morning without problem.

Neo2003, 2018-06-05:
2 days laters, ports are at 4880+ and still increasing:

[Jun  5 18:47:37] NOTICE[9507]: sccp_session.c:648 sccp_session_device_thread: SCCP: Closing session because connection timed out after 31 seconds (ip-address: 192.168.10.254:2000).
    -- SCCP: Accepted Client Connection from 192.168.10.87:4883
    -- SCCP: Connected on server via 192.168.10.254:2000
  == Spawn extension (maison, 100, 4) exited non-zero on 'SIP/knet-00000006'
[Jun  5 18:48:08] NOTICE[9514]: sccp_session.c:648 sccp_session_device_thread: SCCP: Closing session because connection timed out after 31 seconds (ip-address: 192.168.10.254:2000).
    -- SCCP: Accepted Client Connection from 192.168.10.87:4884
    -- SCCP: Connected on server via 192.168.10.254:2000
[Jun  5 18:49:09] NOTICE[9516]: sccp_session.c:648 sccp_session_device_thread: SCCP: Closing session because connection timed out after 31 seconds (ip-address: 192.168.10.254:2000).
    -- SCCP: Accepted Client Connection from 192.168.10.87:4885
    -- SCCP: Connected on server via 192.168.10.254:2000
[Jun  5 18:49:40] NOTICE[9517]: sccp_session.c:648 sccp_session_device_thread: SCCP: Closing session because connection timed out after 31 seconds (ip-address: 192.168.10.254:2000).
    -- SCCP: Accepted Client Connection from 192.168.10.87:4886
    -- SCCP: Connected on server via 192.168.10.254:2000

DarrylHon, 2018-06-07:
@Neo2003 I wonder if this could be an Cisco AP config issue. On the 7925, in the WLAN Config settings, try setting "Call Power Save Mode" to NONE. (defaults to U-APSD/PS-Poll)

Neo2003, 2018-06-11:
Thanks for the suggestion @DarrylHon, but it's still the same behavior after changing this setting.
Problem is that after days, it seams to cause asterick to hang.
In the meantime, do you know a way to force a phone reboot every 24H ?

DarrylHon, 2018-06-12:
@Neo2003 Oh too bad, I remember observing similar behavior when the handset connected to any AP without U-APSD WMM during our early testing.
Is your handset running the latest firmware? If so, then maybe the issue is outside of asterisk/chan-sccp.
We have a lone eval unit here and it's fairly stable.

PhantomVl, 2018-06-13:
You should pay attention to the parameter in sccp.conf
[general]
keepalive = 60 (sec)

You may also need to add the following parameters to the configuration file :
<cnf.xml>
<connectionMonitorDuration>[=keepalive*0.75] </connectionMonitorDuration>
 <srstInfo>
      <name>OFF</name>
      <srstOption>disable</srstOption>
      <userModifiable>false</userModifiable>
      <ipAddr1></ipAddr1>
      <port1></port1>
      <ipAddr2></ipAddr2>
      <port2></port2>
      <ipAddr3></ipAddr3>
      <port3></port3>
    <isSecure>false</isSecure>
</srstInfo>

Neo2003, 2018-06-16:
Many thanks @PhantomVl it works !!
Now the phone is not connecting on 2 ports, but only 1 connection and no more timeout of the 2nd not managed one.
Here is the final cnf.xml file which works properly with a 7925G
https://pastebin.com/tnc0vU2V

For information, can you elaborate a bit on what the section <srstInfo> is?

PhantomVl, 2018-06-18:
The possibility that your airnet will intercept connections from 7925. It is a functional from Cisco that provides operation of remote systems. Disabling this feature in the configuration will save you from unexpected problems.
If you use freepbx you should be interested in web add-on : https://github.com/PhantomVl/sccp_manager