chan-sccp/chan-sccp #430: Conference start from AMI Action or CLI

hfroliveira, 2018-06-04:
Hello,

After reading the documentation about [conferences](https://github.com/chan-sccp/chan-sccp/wiki/Conferencing) we have been able to setup conferences correctly using Cisco phones.
I would like to know if is possible to send an AMI action or a CLI command to start a conference call, something like the "Conf" key on the phone does.

By inspecting the available CLI conference related commands they all seem to operate on already existing conferences:
```bash
sccp conference                -- SCCP Conference Commands.
sccp show conferences          -- Show running SCCP Conferences.
sccp show conference           -- Show SCCP Conference Info.
```
**chan-sccp version:**  4.3.1 develop - 69fdf95M
**asterisk version:** 15.3.0 
**Addition / Optional Information:**
- Ubuntu Server 16.04 64 Bit

Thanks.

dkgroot, 2018-11-26:
Currently there is no buildin command start an sccp-conference (neither via CLI or AMI). You could of course send the appropriate SoftKey using an xml-push, but that's not very nice (but it might help you for now).
Can you tell me a little more about your intended use-case for this. If you want to start and 'empty' conference, with only one moderator device in it, that may be doable. Scripting the conference to pull in multiple participants may be a lot more difficult. If you intend to do a number of "sccp call xxxx xxxx" before hand and then start the conference on this device, then adding a command to start the conference would be pretty simple (somewhat similar to sending the softkey via xml-push).

Regards

dkgroot, 2019-08-31:
No response

closed 2019-08-31 by dkgroot