chan-sccp/chan-sccp #431: I do not have video calls between SCCP phones.

S199pWa1k9r, 2018-07-10:
[Hi All!
 
I have three phones Cisco CP-8945
Two in the SCCP version and one in the SIP.
I use ASTERISK 13.21 with fixes from http://usecallmanager.nz/patching-asterisk.html
**asterisk version:**
ast*CLI> core show version
Asterisk 13.21.0 built by root @ ast.kubsu.ru on a amd64 running FreeBSD on 2018-07-09 11:07:02 UTC
ast*CLI> sccp show version
Skinny Client Control Protocol (SCCP). Release: 4.2.3 STABLE - 6728M (built by 'wheel' on '2018-07-07 08:09:11 UTC')
**Phone firmware:**
SCCP894x.9-4-2SR3-1
SIP894x.9-4-2SR3-1
**Platform:**
FreeBSD ast.kubsu.ru 11.1-RELEASE FreeBSD 11.1-RELEASE #0: Tue May 22 10:46:30 MSK 2018
 
I do not have video calls between SCCP phones.
The call from the SIP to the SCCP works perfectly. Resolution 640x380.
Call from SCCP to SCCP displays the resolution 176x144.
 
What could be the problem?
 
- Direct Contact information: [s199p.wa1k9r@gmail.com]
 
Thank you.

xrioann, 2018-07-16:
I have the same problem (also described here: https://sourceforge.net/p/chan-sccp-b/mailman/message/36351368/).

xrioann, 2018-12-15:
I was testing a few versions. Seems that video functionality was working until version 4.1.0 and broken with 4.2.0

mpaland, 2019-01-08:
http://usecallmanager.nz is a very cool site with lots of 79xx config information I didn't know.
Thanks for sharing this!

dkgroot, 2019-01-08:
@S199pWa1k9r @xrioann @davidded 
Got inbound and outbound video working again in branch issues/431 :-)
Also updated the default resolution to the biggest possible (non-select-able for now)

dkgroot, 2019-01-08:
> http://usecallmanager.nz is a very cool site with lots of 79xx config information I didn't know.
> Thanks for sharing this!

Doesn't mean you should switch to chan_sip+usecallmanager and never look back (I hope) :-)

xrioann, 2019-01-08:
> @S199pWa1k9r @xrioann
> 
> Sorry i took such a long time, for us to look into this.
> We created a new branch "issues/431" and committed some fixes to it. Currently we only support H264 and incoming calls seem to work again. There is still an issue with outgoing calls, which we are working on.
> You can give this branch a try and report back:
> 
> ```
> git pull
> git fetch -a
> git checkout issues/431
> make && make reload
> ```
> The first step was to get video working again. The image size issue is a known issue, and we will work on that as well (already figured out where to fix this AFAICT). We also see if we can add support for h263 / h263P (and maybe even h261).
> Quite a bit of work to get everything ship shape again.

Thank you Diederik, no worries about it taking time. Your work with sccp has helped me a lot with or without video calls. Will try it tonight hopefully!

dkgroot, 2019-01-09:
Major fix was required for sccp_config.c to be able to parse video codecs correctly. It previously caused a segfault when setting a video codec in global allow/disallow. That one has been resolved in the last commit.
We should be getting back on track regarding video :-)

dkgroot, 2019-01-25:
@xrioann Ok it's clear what's happening here. I older versions we only provided some video support when using the h264 codec. Only in this 431 branch i added h263 as a potential option, but i could not really test it, because the 8945 only supports h264.
Do you have `allow=h263` in your sccp.conf for this device ? If so, it's a little strange why it would not show up in the video preferences.

BTW: --enable-conferencing and --enable-video-layer do not exists (anymore) in the configure script. Just use: `./configure --enable-video --enable-conference`. It's not going to help solve the video issue, but it would enable the conference feature correctly.

dkgroot, 2019-01-27:
I will have to ask for @davidded assistance on the h263 codec,  because i think he has one of these tandberg/7985's as well.

mrdoger, 2019-02-01:
I tried to compile with issues/431. But I got a build error
        git clone https://github.com/chan-sccp/chan-sccp chan-sccp
        cd chan-sccp
        git pull
        git fetch -a
         git checkout issues/431
         ./configure --enable-video --enable-conferencing
         make
`
 CC       libsccp_la-sccp_mwi.lo
sccp_mwi.c: In function 'sccp_mwi_addMailboxSubscription':
sccp_mwi.c:545:4: warning: implicit declaration of function 'stasis_subscription_accept_message_type' [-Wimplicit-function-declaration]
    stasis_subscription_accept_message_type(subscription->event_sub, ast_mwi_state_type());
    ^
sccp_mwi.c:545:4: warning: nested extern declaration of 'stasis_subscription_accept_message_type' [-Wnested-externs]
sccp_mwi.c:547:4: warning: implicit declaration of function 'stasis_subscription_set_filter' [-Wimplicit-function-declaration]
    stasis_subscription_set_filter(subscription->event_sub, STASIS_SUBSCRIPTION_FILTER_SELECTIVE);
    ^
sccp_mwi.c:547:4: warning: nested extern declaration of 'stasis_subscription_set_filter' [-Wnested-externs]
sccp_mwi.c:547:60: error: 'STASIS_SUBSCRIPTION_FILTER_SELECTIVE' undeclared (first use in this function)
    stasis_subscription_set_filter(subscription->event_sub, STASIS_SUBSCRIPTION_FILTER_SELECTIVE);
                                                            ^
sccp_mwi.c:547:60: note: each undeclared identifier is reported only once for each function it appears in
sccp_mwi.c: At top level:
cc1: warning: unrecognized command line option "-Wno-format-truncation" [enabled by default]
make[3]: *** [libsccp_la-sccp_mwi.lo] Error 1
make[3]: Leaving directory `/root/chan-sccp/chan-sccp/src'
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory `/root/chan-sccp/chan-sccp/src'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/root/chan-sccp/chan-sccp/src'
make: *** [all-recursive] Error 1

`

S199pWa1k9r, 2019-03-08:
Thank you Diederik, issues/431 works great!
But the current issues/445 does not work at all.

dkgroot, 2019-03-08:
@S199pWa1k9r I will see how to merge 431 into the current develop tree... I will require a little bit more work. Thanks a lot for checking / testing it !
Much appreciated

dkgroot, 2019-03-08:
@S199pWa1k9r 
Step one... rebasing issues/432 branch ontop of the develop branch changes...
I had to do some manual merging... Everything still compiles AFAIK.
Can you give it another go, to see if everything is still compatible and works correctly ? Including video ?

dkgroot, 2019-03-16:
Created a fresh new branch called "issues/431_1". 
And ran a number of tests:
SIP->SCCP
SCCP->SIP
SCCP->SCCP
And they all seem fine.

The debug logging for video handling has been extended quite a bit, to show exactly how/where we are handling video.
There is still a little more to.. But the major part is already working.

Please give it a try:
```
git pull
git fetch -a
git branch
git checkout issues/431_1
./configure [...]
make -j4 && make install && make reload
```
Please use "sccp debug core, pbx, indicate, codec, rtp, device, line, channel" then debugging.

dkgroot, 2019-03-16:
Tests:
\-  unknown/needs checking
n/a not applicable
ok = ok / works
x = not ok
ast = cause by asterisk

test                       | single line | shared line | comment
----------------------- | --------- | --------- | -----------
videomode softkey | -  | - | 
hold/resume          | ok | - | 
leave voicemail      | ok | - | 
playback voicemail | ast | - | see below (*)
callforward            | -   | - | 
transfer            | -       | - | 
pickup            |  -      | - | 
gpickup            | -      | - | 
intercept           | n/a     | - | 
conference       | -     | - | 
AMI/CLI output | -     | - | 
CHANNEL(video) | -   | -  | 

> (*) Slow voicemail playback is an (open) digium/asterisk issue: https://issues.asterisk.org/jira/browse/ASTERISK-27125

dkgroot, 2019-03-25:
Issued a new branch called "issues/457_431" which contains the changes from #457 and the changes from #431 (on top).

dkgroot, 2019-04-11:
I have merged "issues/457" into develop
I have rebased "issues/457_431_1" on top of the new develop branch, and renamed it to "issues/431_2"

So if you are interested in having video support, please jump to "issues/431_2" and give it a go !

dkgroot, 2019-04-23:
The Branch "issues/431_2" was just merged into develop. We should now have video support back in the main tree.
Thanks @darkponay for testing the last branch extensively.
Thanks you all for test/checking and your patience.

commit e377f559 (2019-04-23): Fixes related to Video handling
 - Update/Modernize iPbx.set_nativeAudioFormats and iPbx.set_nativeVideFormats, allowing multiple capabilities
 - Updated/Extended OpenMultiMediaChannel skinny protocol message
   - Todo: More research needed to figure out how to use H261/H263 and H263P
 - Add generic handler: __find_joint_capabilities, called from pbx_retrieve_remote_capabilities
 - Add sccp_channel_recalculateVideoCodecFormat to support H264 and other video codecs
 - Group DisplayPromptMessages together
 - Update DND handling

 - Active incoming video when available

 - Add sccp.conf line para [... cut]
closed 2019-04-23 by dkgroot