chan-sccp/chan-sccp #433: Question how make load = chan_sccp.so to be persistent in fpbx modules.conf

ostridge, 2018-08-18:
[Short description of problem here]
In Asterisk 13.22.0  + freePBX GUI ver. 13.0.120.10 The /etc/asterisk/modules.conf is autogenerated - so Where do I add the line ' load = chan_sccp.so ' in order for it to be persistent.
**Reproduction Steps:** https://github.com/chan-sccp/chan-sccp/wiki/FreePBX_Installation
Step 4 
1. [First Step] worked thro /chan-sccp/wiki/FreePBX_Installation to step 3
2. [Second Step] Did git clone  https://github.com/chan-sccp/chan-sccp chan-sccp
3. [Other Steps...] executed as step3 and did  make install OK 
4.  edit the /etc/asterisk/modules.conf file. Add: load = chan_sccp.so
( noload = chan_skinny.so ; already listed in autogenerated file )
there are no #include lines in /etc/asterisk/modules.conf

??? do I need  to run menu select as previous depreciated instructions ?

**Expected behavior:**

[Describe expected behavior here]

**Observed behavior:**

[Describe observed behavior here]

**Screenshots**

(Screenshots which show reproduction steps and/or demonstrate the problem)
[url]

**chan-sccp version:** (include revision and/or branch if relevant)

**asterisk version:** (include revision and/or branch if relevant)

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Yes/No]
- Problem can be reliably reproduced, doesn't happen randomly: [Yes/No]
- Platform:
  - [CPU]
  - [32/64 Bit]
  - [Big/Small Endian]
  - [Operating System]
  - [Operating System Version]
- Direct Contact information: [You can contact me here]
- Log Files (attach .txt / [pastbin](http://pastebin.com/) / [gist](http://gist.github.com/)): [url]
- Comments: [Other Information]

beta2k1, 2019-02-04:
You need to make the changes through the Asterisk Modules menu under Admin.  Add chan_skinny.so to the exclusions list and add chan_scco,so to the manually loaded list.

dkgroot, 2019-02-04:
@ostridge I am assuming that that answers your question. Closing this issue.
@beta2k1 thanks a lot for helping with this one !!

closed 2019-02-04 by dkgroot