chan-sccp/chan-sccp #434: Call Forward and Transfer buttons are missing

Fedor114, 2018-09-17:
Hi!
We are unable to add cfwdall and transfer buttons.
cfwdall and transfer byttons are enabled in keyset but there is an empty place insted of buttons on phones. We use ip phones CISCO 7965, 7975, 7942 - we have the same problem on any device of this model.
Can anybody healp please.

Thanks in advance.

Fedor

dkgroot, 2018-12-31:
@Fedor114 
Do the empty buttons have any effect when pressed ?
That is quite peculiar. It would require quite a bit of debugging to figure out what is going on in this case. Are you up for that ?
FYI: have you tried a clean install of chan-sccp and a factory reset of one of these devices ?

dkgroot, 2019-09-15:
No response for some time ! We closed this issue for now.
If/when necessary, do reopen the issue please.

closed 2019-09-15 by dkgroot